esp-rs/esp-idf-template #2: Stack overflow

hongquan, 2021-10-30:
I built and flashed the fw from this project, but got this error when monitoring the board:

```
Hello, world!
***ERROR*** A stack overflow in task main has been detected.
Backtrace:0x40085112:0x3ffb5f50 0x40085915:0x3ffb5f70 0x40088ce1:0x3ffb5f90 0x400875e5:0x3ffb6010 0x40085a0c:0x3ffb6030 0x400859be:0x3f400120 |<-CORRUPTED
ELF file SHA256: 0000000000000000
Rebooting...
```

What do I need to do to fix it?

hongquan, 2021-10-30:
Full log:

```
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
I (180) cpu_start: Pro cpu up.
I (180) cpu_start: Starting app cpu, entry point is 0x4008115c
I (165) cpu_start: App cpu up.
I (194) cpu_start: Pro cpu start user code
I (194) cpu_start: cpu freq: 160000000
I (194) cpu_start: Application information:
I (199) cpu_start: Project name:     esp-idf
I (203) cpu_start: App version:      5fe3a5d
I (208) cpu_start: Compile time:     Oct 30 2021 12:08:25
I (215) cpu_start: ELF file SHA256:  0000000000000000...
I (221) cpu_start: ESP-IDF:          4.3.0
I (226) heap_init: Initializing. RAM available for dynamic allocation:
I (233) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (239) heap_init: At 3FFB3488 len 0002CB78 (178 KiB): DRAM
I (245) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (251) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (258) heap_init: At 4008C538 len 00013AC8 (78 KiB): IRAM
I (265) spi_flash: detected chip: generic
I (269) spi_flash: flash io: dio
I (274) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello, world!
***ERROR*** A stack overflow in task main has been detected.
Backtrace:0x40085112:0x3ffb5f50 0x40085915:0x3ffb5f70 0x40088ce1:0x3ffb5f90 0x400875e5:0x3ffb6010 0x40085a0c:0x3ffb6030 0x400859be:0x3f400120 |<-CORRUPTED
ELF file SHA256: 0000000000000000
Rebooting...
```

ivmarkov, 2021-10-30:
You can try to put
```
CONFIG_ESP_MAIN_TASK_STACK_SIZE=7000
```

... in the `sdkconfig.defaults` file of the project, then do `cargo clean; cargo build` and let me know whether you are still experiencing the issue. I have another report like that, but for me that issue is just not reproducible.

Also:
* What board are you using?
* Rust compiler version?

hongquan, 2021-10-30:
@ivmarkov I added `CONFIG_ESP_MAIN_TASK_STACK_SIZE=7000`, but the problem still remains.

- Rust version:
  ```
  ❯ ~/.rustup/toolchains/esp/bin/rustc --version
  rustc 1.56.0-dev
  ```
- Board: An old kit that I borrowed from friend, with "ESP32-WROOM-32D" chip.

hongquan, 2021-10-30:
[image]
[image]

hongquan, 2021-10-30:
Sorry, last time I didn't do `cargo clean`. Now I clean, rebuild and flash again, the board seems to be bricked:

```
❯ espmonitor /dev/ttyUSB0
ESPMonitor 0.5.2

Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

Opening /dev/ttyUSB0 with speed 115200
Resetting device... done
```

hongquan, 2021-10-30:
This repo https://github.com/espressif/rust-esp32-example is working and here is the info printed from that fw:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7048
load:0x40078000,len:14308
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (27) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (27) boot: compile time 13:01:18
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=06c04h ( 27652) map
I (119) esp_image: segment 1: paddr=00016c2c vaddr=3ffb0000 size=02908h ( 10504) load
I (124) esp_image: segment 2: paddr=0001953c vaddr=40080000 size=06adch ( 27356) load
I (137) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=13b48h ( 80712) map
I (167) esp_image: segment 4: paddr=00033b70 vaddr=40086adc size=04178h ( 16760) load
I (174) esp_image: segment 5: paddr=00037cf0 vaddr=50000000 size=00010h (    16) load
I (180) boot: Loaded app from partition at offset 0x10000
I (180) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (195) cpu_start: Starting app cpu, entry point is 0x40080ff8
0x40080ff8: call_start_cpu1 at /home/quan/Works/ESP-IDF/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (210) cpu_start: Pro cpu start user code
I (210) cpu_start: cpu freq: 160000000
I (210) cpu_start: Application information:
I (214) cpu_start: Project name:     esp32-hello-rust
I (220) cpu_start: App version:      6b4de41
I (225) cpu_start: Compile time:     Oct 30 2021 13:01:12
I (231) cpu_start: ELF file SHA256:  414502fc74f8b587...
I (237) cpu_start: ESP-IDF:          v4.3.1
I (242) heap_init: Initializing. RAM available for dynamic allocation:
I (249) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (255) heap_init: At 3FFB31D8 len 0002CE28 (179 KiB): DRAM
I (261) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (268) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (274) heap_init: At 4008AC54 len 000153AC (84 KiB): IRAM
I (281) spi_flash: detected chip: generic
I (285) spi_flash: flash io: dio
W (289) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (303) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is esp32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 2MB external flash
Free heap: 292396
C validated param #0 = 2
C validated param #1 = 3
Rust calculated 2 + 3 = 5

C validated param #0 = 9
C validated param #1 = 10
Rust calculated (using asm!) 9 + 10 = 19

Restarting in 10 seconds...
```

hongquan, 2021-10-30:
Solved it by removing `CONFIG_ESP_MAIN_TASK_STACK_SIZE` (use default value) and build the fw in _release_ mode.

closed 2021-10-31 by hongquan