esp-rs/esp-idf-template #10: Did not compile for me

jcdc404, 2022-01-30:
running Debian 10
rustup 1.24.3
rustc 1.58.0-dev
Linux ------ 5.10.0-0.bpo.9-amd64 #1 SMP Debian 5.10.70-1~bpo10+1 (2021-10-10) x86_64 GNU/Linux

```
cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cargo
cd esp32rs
cargo build --features native
```
```
error: failed to run custom build command for `esp-idf-sys v0.30.4`
Caused by:
  process didn't exit successfully: `/home/newuser/MHA/ESP32/esp32rs/target/debug/build/esp-idf-sys-aa24d9eff47a728b/build-script-build` (exit status: 1)
```
```
--- stderr
  Cloning into '/home/newuser/MHA/ESP32/esp32rs/.embuild/espressif/esp-idf-v4.3.2'...
  Note: checking out '8bf14a9238329954c7c5062eeeda569529aedf75'.

  You are in 'detached HEAD' state. You can look around, make experimental
  changes and commit them, and you can discard any commits you make in this
  state without impacting any branches by performing another checkout.

  If you want to create a new branch to retain commits you create, you may
  do so (now or later) by using -b with the checkout command again. Example:

    git checkout -b <new-branch-name>
```
virtualenv.py: error: no such option: --seeder
  Traceback (most recent call last):
    File "/home/newuser/MHA/ESP32/esp32rs/.embuild/espressif/esp-idf-v4.3.2/tools/idf_tools.py", line 1818, in <module>
      main(sys.argv[1:])
    File "/home/newuser/MHA/ESP32/esp32rs/.embuild/espressif/esp-idf-v4.3.2/tools/idf_tools.py", line 1814, in main
      action_func(args)
    File "/home/newuser/MHA/ESP32/esp32rs/.embuild/espressif/esp-idf-v4.3.2/tools/idf_tools.py", line 1519, in action_install_python_env
      stdout=sys.stdout, stderr=sys.stderr)
    File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'virtualenv', '--seeder', 'pip', '/home/newuser/MHA/ESP32/esp32rs/.embuild/espressif/python_env/idf4.3_py3.7_env']' returned non-zero exit status 2.
  Error: Could not install esp-idf

  Caused by:
      command '"python3" "/home/newuser/MHA/ESP32/esp32rs/.embuild/espressif/esp-idf-v4.3.2/tools/idf_tools.py" "--idf-path" "/home/newuser/MHA/ESP32/esp32rs/.embuild/espressif/esp-idf-v4.3.2" "--quiet" "--non-interactive" "install-python-env"' exited with non-zero status code 1
```

MabezDev, 2022-01-30:
Seems like you might need to update your python installation?

`python3 -m virtualenv --help` should show `--seeder` as an option.

jcdc404, 2022-01-31:
Apparently it was the version of virtualenv. I had to: `sudo pip3 install virtualenv --upgrade`
Updating and upgrading Linux, will not upgrade this package. The above command was required.

closed 2022-01-31 by jcdc404