esp-rs/esp-idf-template #11: Testing

lkoba, 2022-02-03:
```cargo test``` fails both on cmake and cargo projects:

```
Compiling rust-probando-tests v0.1.0 (/workdir/rust-probando-tests)
error: linking with `ldproxy` failed: exit status: 1
  |
  = note: "ldproxy" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/rust_probando_tests-b7454db5234f6fbd.2kxikby9yetkb74i.rcgu.o" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/rust_probando_tests-b7454db5234f6fbd.3avceqw8aeykl72w.rcgu.o" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/rust_probando_tests-b7454db5234f6fbd.4scxl40zqo84lyn1.rcgu.o" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/rust_probando_tests-b7454db5234f6fbd.5gmgjumh5nex0nfb.rcgu.o" "-Wl,--as-needed" "-L" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps" "-L" "/workdir/rust-probando-tests/target/debug/deps" "-L" "/opt/rustup/toolchains/esp/lib/rustlib/xtensa-esp32-espidf/lib" "-Wl,-Bstatic" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libtest-1240a78d4c2af68d.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libgetopts-e933c34a0d206322.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libunicode_width-a41b277777faa969.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/librustc_std_workspace_std-ed5dfb3c9b12ee67.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libesp_idf_sys-d6b64b34fcb65cdf.rlib" "-Wl,--start-group" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libstd-43840f1f8cbba82e.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libpanic_abort-18987952604e34e8.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libstd_detect-f2dc5c1bd62eb738.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/librustc_demangle-0b6f33d837cead7a.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libhashbrown-23e693654a4f6df1.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/librustc_std_workspace_alloc-c0d9aaa11f51d8d8.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libunwind-4b9acc803481db28.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libcfg_if-5e8ae41a05611c02.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/liblibc-856729009bd346d0.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/liballoc-b0b73273780e9c1e.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/librustc_std_workspace_core-7dd968a8c02328cc.rlib" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libcore-2fc8e6c2716970dd.rlib" "-Wl,--end-group" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/libcompiler_builtins-807d6fa2e73b1452.rlib" "-Wl,-Bdynamic" "-Wl,--eh-frame-hdr" "-Wl,-znoexecstack" "-L" "/opt/rustup/toolchains/esp/lib/rustlib/xtensa-esp32-espidf/lib" "-o" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/rust_probando_tests-b7454db5234f6fbd" "-Wl,--gc-sections" "-no-pie" "-nodefaultlibs" "--ldproxy-linker" "/workdir/rust-probando-tests/.embuild/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc" "--ldproxy-dedup-libs" "-L/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf" "-L/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug" "-L/workdir/rust-probando-tests/.embuild/platformio/packages/framework-espidf/components/esp_wifi/lib/esp32" "-L/workdir/rust-probando-tests/.embuild/platformio/packages/framework-espidf/components/esp_rom/esp32/ld" "-L/workdir/rust-probando-tests/.embuild/platformio/packages/framework-espidf/components/xtensa/esp32" "-L/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/esp32" "-L/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/esp32/ld" "-L/workdir/rust-probando-tests/.embuild/platformio/packages/framework-espidf/components/esp32/ld" "-Wl,--start-group" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/esp_pm/libesp_pm.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/asio/libasio.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/cbor/libcbor.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/unity/libunity.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/cmock/libcmock.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/coap/libcoap.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/console/libconsole.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/esp_adc_cal/libesp_adc_cal.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/esp_hid/libesp_hid.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/protobuf-c/libprotobuf-c.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/protocomm/libprotocomm.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/mdns/libmdns.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/esp_local_ctrl/libesp_local_ctrl.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/esp_websocket_client/libesp_websocket_client.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/expat/libexpat.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/wear_levelling/libwear_levelling.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/fatfs/libfatfs.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/freemodbus/libfreemodbus.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/jsmn/libjsmn.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/json/libjson.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/libsodium/liblibsodium.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/mqtt/libmqtt.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/openssl/libopenssl.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/spiffs/libspiffs.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/wifi_provisioning/libwifi_provisioning.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8cb26051dc7d79af/out/esp-idf/.pio/build/debug/esp-idf/app_trace/libapp_trace.a" "/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/bu [... cut ...]
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `rust-probando-tests` due to previous error
```

Is this a local issue on my installation? Which is the recommended way to test an app developed with this template?

Right now I'm moving all the code that doesn't interact with ESP-IDF functions to a different crate and run tests there.

MabezDev, 2022-02-03:
That _shouldn't_ happen, as we have a patch for pthread rwlocks for esp-idf 4.3. If you enable the `native` feature of esp-idf-sys, and use esp-idf 4.4 by putting the following in `.cargo/config.toml`:

```toml
ESP_IDF_VERSION = { value = "branch:release/v4.4" }
```

That should fix this, but I'm not sure how it happens in the first place. Any ideas @ivmarkov?

ivmarkov, 2022-02-03:
@lkoba Is `esp_idf_sys::link_patches()` being called from your tests?

lkoba, 2022-02-03:
No. After adding it there was some progress, I'm getting a different error now.

main.rs
```
use esp_idf_sys as _;

fn main() {
    esp_idf_sys::link_patches();
    println!("Hello, world!");
}

#[cfg(test)]
mod tests {
    #[test]
    fn one() {
        esp_idf_sys::link_patches();
        assert_eq!(2 + 2, 4);
    }
}
```

Running ```cargo test``` now outputs:
```
   Compiling rust-probando-tests v0.1.0 (/workdir/rust-probando-tests)
    Finished test [optimized + debuginfo] target(s) in 1.81s
     Running unittests (target/xtensa-esp32-espidf/debug/deps/rust_probando_tests-b7454db5234f6fbd)
/workdir/rust-probando-tests/target/xtensa-esp32-espidf/debug/deps/rust_probando_tests-b7454db5234f6fbd: 1: Syntax error: "(" unexpected
error: test failed, to rerun pass '--bin rust-probando-tests'
```

ivmarkov, 2022-02-03:
I'm taking a step back and trying to understand what you are trying to achieve here with the tests. The key here is that you are *cross-compiling*. In other words, the binary (your `main` function) is not complied and cannot run on the host platform:
1. If your tests do contain ESP-specific code (references to `esp-idf-sys`, `esp-idf-svc`, etcetera) then by definition you cannot "run" these tests on the host platform either, as the host platform is *not* an ESP. So you need a way to "plug" an emulator into the cargo `test` capabilities and somehow instruct cargo to (a) compile these tests for ESP and (b) run them _inside_ the emulator. I don't think any of these capabilities are available today with stock `cargo test`.
2. If your tests *do not* contain ESP-specific code - i.e. they are platform-agnostic - you have to push them - together with the platform-agnostic code you are testing into a separate crate that you can compile for the host platform in the first place.

So to summarize, I don't think you can do (1) with `cargo test`. You can do (2) with `cargo test`, but only for your platform-independent code, and you need to isolate it in another crate. This link might shed some light here - and - it also seems to contain info how you can do on-target testing using probe-run and stuff: https://ferrous-systems.com/blog/test-embedded-app/

lkoba, 2022-02-03:
The link you sent is exactly was I was looking for. Thanks!

closed 2022-02-03 by lkoba