esp-rs/esp-idf-template #12: cargo generate does not include features "native"

georgik, 2022-02-14:
Cargo generate does not generate the template which has native feature enabled.
The build is then using pio by default.

ivmarkov, 2022-02-14:
I'm not sure what is the ask. Are you asking for the ["native" feature](https://github.com/esp-rs/esp-idf-template/blob/f080301e91f5b61dcaecb8bed89b22780105e011/cargo/Cargo.toml#L16) to be included in the default features?

We can do that, but only for the case when the user has selected a branch that is supported with native build only.

ivmarkov, 2022-02-21:
If my interpretation of what you request is correct, then this is now fixed with https://github.com/esp-rs/esp-idf-template/commit/94a7d00f82a3c694d2bd1b5b29ceba2fbde4d7a1 .

closed 2022-02-21 by ivmarkov