esp-rs/esp-idf-template #14: Empty no_std project fails to compile

jcard0na, 2022-02-21:
I just tried to build a `no_std` project and it failed to compile.  `std` project compiles and runs successfully.

Project created as:
```
[javier@t420 dev]$ cargo generate --git https://github.com/esp-rs/esp-idf-template cargo
⚠️   Unable to load config file: /home/javier/.cargo/cargo-generate.toml
🤷   Project Name : meh
🔧   Generating template ...
✔ 🤷   Rust toolchain (beware: nightly works only for esp32c3!) · esp
✔ 🤷   STD support · false
✔ 🤷   MCU · esp32
? 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, v5.0 = development; NOTE: applicable only with `cargo build --features✔ 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, v5.0 = development; NOTE: applicable only with `cargo build --features native`) · v4.4
[1/9]   Done: .cargo/config.toml
[2/9]   Done: .cargo
[3/9]   Done: .gitignore
[4/9]   Done: Cargo.toml
[5/9]   Done: build.rs
[6/9]   Done: rust-toolchain.toml
[7/9]   Done: sdkconfig.defaults
[8/9]   Done: src/main.rs
[9/9]   Done: src
🔧   Moving generated files into: `/opt/javier/dev/meh`...
💡   Initializing a fresh Git repository
✨   Done! New project created /opt/javier/dev/meh
``` 
Then built as...
```
[javier@t420 meh]$ cargo build
    Updating crates.io index
   Compiling compiler_builtins v0.1.53
   Compiling proc-macro2 v1.0.36
(...)
error[E0433]: failed to resolve: use of undeclared crate or module `std`
 --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/eventloop.rs:7:5
  |
7 | use std::ptr::NonNull;
  |     ^^^ use of undeclared crate or module `std`

error[E0433]: failed to resolve: use of undeclared crate or module `std`
 --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:5:5
  |
5 | use std::mem::ManuallyDrop;
  |     ^^^ use of undeclared crate or module `std`

error[E0433]: failed to resolve: use of undeclared type `NonNull`
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/eventloop.rs:168:13
    |
168 |             NonNull::dangling().as_ptr() as *const P
    |             ^^^^^^^ not found in this scope
    |
help: consider importing one of these items
    |
1   | use core::ptr::NonNull;
    |
1   | use crate::eventloop::ptr::NonNull;
    |

error[E0412]: cannot find type `Box` in this scope
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:166:28
    |
166 | struct UnsafeCallback(*mut Box<dyn FnMut(esp_mqtt_event_handle_t)>);
    |                            ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

error[E0412]: cannot find type `Box` in this scope
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:169:25
    |
169 |     fn from(boxed: &mut Box<Box<dyn FnMut(esp_mqtt_event_handle_t)>>) -> Self {
    |                         ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

error[E0412]: cannot find type `Box` in this scope
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:169:29
    |
169 |     fn from(boxed: &mut Box<Box<dyn FnMut(esp_mqtt_event_handle_t)>>) -> Self {
    |                             ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

error[E0412]: cannot find type `Box` in this scope
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:190:5
    |
190 |     Box<dyn FnMut(esp_mqtt_event_handle_t)>,
    |     ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

error[E0433]: failed to resolve: use of undeclared type `Box`
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:213:13
    |
213 |             Box::new(move |event_handle| EspMqttConnection::post(&client_connection, event_handle)),
    |             ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

error[E0433]: failed to resolve: use of undeclared type `Box`
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:231:13
    |
231 |             Box::new(move |event_handle| {
    |             ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

error[E0433]: failed to resolve: use of undeclared type `ManuallyDrop`
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:235:37
    |
235 |                     let mut event = ManuallyDrop::new(Some(EspMqttMessage::new_event(event, None)));
    |                                     ^^^^^^^^^^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use core::mem::ManuallyDrop;
    |

error[E0433]: failed to resolve: use of undeclared type `ManuallyDrop`
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:240:25
    |
240 |                         ManuallyDrop::drop(&mut event);
    |                         ^^^^^^^^^^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use core::mem::ManuallyDrop;
    |

error[E0412]: cannot find type `Box` in this scope
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:252:23
    |
252 |         raw_callback: Box<dyn FnMut(esp_mqtt_event_handle_t)>,
    |                       ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

error[E0433]: failed to resolve: use of undeclared type `Box`
   --> /home/javier/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.37.0/src/mqtt/client.rs:257:38
    |
257 |         let mut boxed_raw_callback = Box::new(raw_callback);
    |                                      ^^^ not found in this scope
    |
help: consider importing this struct
    |
1   | use alloc::boxed::Box;
    |

Some errors have detailed explanations: E0412, E0433.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `esp-idf-svc` due to 13 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed
```
Same toolchain, `std` project builds fine...
```
[javier@t420 rust-esp32-std-demo]$ cargo build
   Compiling rust-esp32-std-demo v0.22.8 (/opt/javier/dev/rust-esp32-std-demo)
    Finished dev [optimized + debuginfo] target(s) in 13.40s
```

ivmarkov, 2022-02-23:
Should be fixed now.

closed 2022-02-23 by ivmarkov

ivmarkov, 2022-02-23:
By the way - and I'm not sure what you are trying to achieve - but - using `no_std` on top of ESP-IDF is pretty much pointless.
It only exists "because we can", and "just in case".

Rust support for ESP-IDF *is* a STD compatible environment anyway, so you are not winning anything by living in `no_std` land, and losing a lot. As in how do you plan to e.g. use networking or spin threads?

With that said, `no_std` has value for library crates that depend only on `embedded-hal` and `embedded-svc`, as this makes these portable across various MCUs. But you should not be using the `esp-idf-template` for library MCU-agnostic crates anyway.

jcard0na, 2022-02-23:
Thanks a lot for the clarification @ivmarkov.  
Mi project does not use networking, nor spin threads.  I only use an epd display on waveshare's esp32 driver board. I was looking to use `no_std` because I'm migrating a `no_std` project from a different target and did not want to inadvertently introduce `std` dependencies that would prevent me to migrate back.  For my use case, the ideal environment would be to depend exclusively on `embedded-hal` and `embedded-svc` and not use ESP-IDF at all, but I do not think that is possible with this target, is that right?
Cheers,

ivmarkov, 2022-02-23:
With or without this target, there is no chance to _not_ depend on something which is MCU specific. I mean, e.g. instantiating the SPI and pins is always MCU specific and not covered by e.g. `embedded-hal` which is just a bunch of traits. Regardless of the MCU.

If you want to split your code to MCU-independent part and MCU-dependent part, you should use two crates: a library crate which only depends on `embedded-hal`, and then a binary crate that depends on the MCU specific hal stuff and also on the library crate.

As for `embedded-svc` - given that you just want to draw on an epd display, you don't need that at all.
And, it currently has an implementation for the ESP32 anyway (in `esp-idf-svc`). With a PC (and RPI) implementation likely to follow sometime this year.