esp-rs/esp-idf-template #15: flash size issue causing boot loop on Waveshare ESP-C3-32S-Kit

sgrimee, 2022-03-10:
Trying to build the unmodified template of the waveshare ESP-C3-32S-Kit board, I get the following error at boot.

```
cargo build --features native
```

```
is  v0.1.0 via  v1.61.0-nightly 
embedded/idf-test/esp-idf-template-test ➜  espflash --monitor target/riscv32imc-esp-espidf/debug/esp-idf-template-test 
Serial port: /dev/tty.usbserial-210
Connecting...

[00:00:01] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:11] ########################################      96/96      segment 0x10000
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

�
 �ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (30) boot: compile time 12:10:40
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (44) boot.esp32c3: SPI Speed      : 80MHz
I (45) boot.esp32c3: SPI Mode       : DIO
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08348h ( 33608) map
I (118) esp_image: segment 1: paddr=00018370 vaddr=3fc89a00 size=014fch (  5372) load
I (123) esp_image: segment 2: paddr=00019874 vaddr=40380000 size=067a4h ( 26532) load
I (135) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=17318h ( 95000) map
I (153) esp_image: segment 4: paddr=00037340 vaddr=403867a4 size=03164h ( 12644) load
I (155) esp_image: segment 5: paddr=0003a4ac vaddr=50000010 size=00010h (    16) load
I (161) boot: Loaded app from partition at offset 0x10000
I (165) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (191) cpu_start: Pro cpu start user code
I (191) cpu_start: cpu freq: 160000000
I (191) cpu_start: Application information:
I (194) cpu_start: Project name:     libespidf
I (199) cpu_start: App version:      1
I (203) cpu_start: Compile time:     Mar 10 2022 11:24:18
I (209) cpu_start: ELF file SHA256:  0000000000000000...
I (215) cpu_start: ESP-IDF:          121ddb8
I (220) heap_init: Initializing. RAM available for dynamic allocation:
I (227) heap_init: At 3FC8BDE0 len 00034220 (208 KiB): DRAM
I (234) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (240) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (247) spi_flash: detected chip: generic
I (252) spi_flash: flash io: dio
E (255) spi_flash: Detected size(2048k) smaller than the size in the binary image header(4096k). Probe failed.

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)
Core  0 register dump:
MEPC    : 0x403807ce  RA      : 0x40383596  SP      : 0x3fcde180  GP      : 0x3fc8a200  
TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fcde2df  S1      : 0x00000088  A0      : 0x3fcde1d8  A1      : 0x3fc8a001  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x3fc8b7be  A5      : 0x3fc8b7c6  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcde1d8  S3      : 0x00000001  
S4      : 0x3fcde1d8  S5      : 0x00020020  S6      : 0x00000000  S7      : 0x3fcd7000  
S8      : 0x3c020020  S9      : 0x00010020  S10     : 0x00000006  S11     : 0x3fcde3cc  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fcde180: 0x00000000 0x00000079 0x3c021318 0x4038859a 0x400481b0 0x00000079 0x0000000a 0x3fc8a000
3fcde1a0: 0x00383932 0x00000060 0x00000000 0x00000000 0x00000000 0x3fc8a8c8 0x3c021318 0x3fc8a8ac
3fcde1c0: 0x3c0214e3 0x3fc8a8d8 0x3fcde1a0 0x3fc8a8dc 0x3c021538 0x3fc8a000 0x65737361 0x66207472
3fcde1e0: 0x656c6961 0x64203a64 0x6f635f6f 0x695f6572 0x2074696e 0x72617473 0x2e707574 0x39323a63
3fcde200: 0x66282038 0x6873616c 0x7465725f 0x203d3d20 0x5f505345 0x00294b4f 0x00000104 0x40048722
3fcde220: 0x00000006 0x00020020 0x00000003 0x00000000 0x3fc89db8 0x00000000 0x3fc89de4 0x3fcde254
3fcde240: 0x00000006 0x00010020 0x005e6015 0x4200be50 0x00000000 0x000000ff 0x3c026304 0x00000800
3fcde260: 0x00001000 0x000003e8 0xa0000000 0x0000000a 0x00000002 0x00000000 0x00000000 0x00000001
3fcde280: 0x00000000 0x00000001 0x00000000 0x00000005 0x00000000 0x00000000 0x00000000 0x00000000
3fcde2a0: 0x00000000 0x3fc8bc6c 0x42000020 0x403803ee 0x00000002 0x3c027d68 0x3c027d68 0x42005fce
3fcde2c0: 0x00004d64 0x3ff1c14c 0x3fc8bcc8 0x09896800 0x00000001 0x000001e0 0x00000101 0x30303030
3fcde2e0: 0x30303030 0x30303030 0x30303030 0x40380300 0x0000003e 0x00000000 0x3fc89a50 0x40380616
3fcde300: 0xffffffff 0x3ff1c14c 0x00000000 0x00000007 0x00ffffff 0x3fcd0101 0x00000000 0x7fffffff
3fcde320: 0xffff0000 0x00027317 0x00027337 0x403d0f1a 0x00000000 0x00000000 0x00000000 0x00000000
3fcde340: 0x00000000 0x00000000 0x00000000 0x3fcdf000 0x00004d64 0x3ff1c14c 0x0000ffff 0xffffffff
3fcde360: 0xffffffff 0x3fcde4b0 0xffffffff 0x403d15aa 0x00000000 0x0000000a 0x00010000 0x003f0000
3fcde380: 0x00010000 0x100206e9 0x403803ee 0x000000ee 0x00000005 0x00000000 0x01000000 0x3c020020
3fcde3a0: 0x00008348 0x3fc89a00 0x000014fc 0x40380000 0x000067a4 0x42000020 0x00017318 0x403867a4
3fcde3c0: 0x00003164 0x50000010 0x00000010 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde3e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00010020
3fcde420: 0x00018370 0x00019874 0x00020020 0x00037340 0x0003a4ac 0x00000000 0x00000000 0x00000000
3fcde440: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x0002a4e0
3fcde460: 0x467530c2 0xfa495b5b 0xa106a91a 0x05f4530c 0x89141e6b 0x9d77b5ec 0x209335f9 0x6c0e9924
3fcde480: 0x00004d64 0x3ff1c14c 0x0000ffff 0x3fce0000 0x00000000 0x000000e4 0x00004d70 0x403ce07c
3fcde4a0: 0x8a4d3993 0xd8308225 0x00000000 0xffffffff 0x00000000 0x00000000 0x00010000 0x003f0000
3fcde4c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde540: 0x00000000 0x00000000 0x00000000 0x00000000 0x600c5200 0x600c5200 0x8082dbb8 0x4004a296
3fcde560: 0xb294a2fe 0x24098570 0x3fcde668 0x00000000 0xffffff63 0x00000000 0x00010000 0x2f0203e9

ELF file SHA256: 0000000000000000

Rebooting...
```

Compilation settings:
```
[build]
target = "riscv32imc-esp-espidf"

[env]
ESP_IDF_VERSION = { value = "branch:release/v4.4" }
...
[features]
default = ["native"]
native = ["esp-idf-sys/native"]
```

ivmarkov, 2022-03-10:
You have one of these rare(er) chips with 2MB flash instead of the usual 4MB.

@jessebraham - I do not remember - what was our support story for these in espflash?

MabezDev, 2022-03-10:
I believe you can pass `--partition-table PATH`, and use the csv format described in [the esp-idf docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html) to reduce your factory partition size to fit in your flash.

ivmarkov, 2022-03-10:
But you also need a 2mb bootloader?

MabezDev, 2022-03-10:
Ah sorry I've misunderstood slightly. I though it was possible, but it seems the image we create is hardcoded to 4mb (among other things). I've opened https://github.com/esp-rs/espflash/issues/140 to solve this.

MabezDev, 2022-03-11:
There is now a [PR](https://github.com/esp-rs/espflash/pull/141) for espflash which should allow the override of header parameters.

Note there is a second way of doing this, you can set the bootloader flash size in sdkconfig, once set you can use the esp-idf bootloader built from your project instead of the builtin bootloader. Passing something like `--bootloader target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-3b7438c03099654c/out/build/bootloader/bootloader.bin` should work.