esp-rs/esp-idf-template #16: Fails to build due to esp-idf-sys

recmo, 2022-03-12:
Empty project using cargo-generate.

```
cargo build
```

```
   Compiling esp-idf-sys v0.30.6
error: failed to run custom build command for `esp-idf-sys v0.30.6`

Caused by:
  process didn't exit successfully: `/Users/remco/WV/esp32c3/esp32c3-test/target/debug/build/esp-idf-sys-e5d7a772d4b05ce4/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/python_env/idf5.0_py3.9_env
  PATH=/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/ninja/1.10.2/:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/python_env/idf5.0_py3.9_env/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/esp-idf-master/tools:$PATH
  Skipping riscv32-esp-elf@esp-2021r2-patch3-8.4.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/python_env/idf5.0_py3.9_env
  PATH=/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/ninja/1.10.2/:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/python_env/idf5.0_py3.9_env/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/esp-idf-master/tools:$PATH
  Skipping cmake@3.20.3 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/python_env/idf5.0_py3.9_env
  PATH=/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/python_env/idf5.0_py3.9_env/bin:/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/esp-idf-master/tools:$PATH
  cargo:rerun-if-changed=/Users/remco/WV/esp32c3/esp32c3-test/sdkconfig.defaults
  CMAKE_PREFIX_PATH_riscv32imc-esp-espidf = None
  CMAKE_PREFIX_PATH_riscv32imc_esp_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_riscv32imc-esp-espidf = None
  CMAKE_riscv32imc_esp_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/Users/remco/WV/esp32c3/esp32c3-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-2f34b9498a53221f/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/esp-idf-master/tools/cmake/toolchain-esp32c3.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/Users/remco/WV/esp32c3/esp32c3-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-2f34b9498a53221f/out" "-DCMAKE_C_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_CXX_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany"
  -- Checking Python dependencies...
  Python requirements are satisfied.
  Skipping the download of /Users/remco/.espressif/espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
  Constraint file: /Users/remco/.espressif/espidf.constraints.v5.0.txt
  Requirement files:
   - /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/esp-idf-master/tools/requirements/requirements.core.txt
  Python being checked: /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/python_env/idf5.0_py3.9_env/bin/python
  -- Project sdkconfig file /Users/remco/WV/esp32c3/esp32c3-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-2f34b9498a53221f/out/sdkconfig
  Loading defaults file /Users/remco/WV/esp32c3/esp32c3-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-2f34b9498a53221f/out/gen-sdkconfig.defaults...
  Loading defaults file /Users/remco/WV/esp32c3/esp32c3-test/sdkconfig.defaults...
  -- Configuring incomplete, errors occurred!
  See also "/Users/remco/WV/esp32c3/esp32c3-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-2f34b9498a53221f/out/build/CMakeFiles/CMakeOutput.log".
  See also "/Users/remco/WV/esp32c3/esp32c3-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-2f34b9498a53221f/out/build/CMakeFiles/CMakeError.log".

  --- stderr
  CMake Error at /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/esp-idf-master/CMakeLists.txt:226 (message):
    Failed to determine sizeof(time_t)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /Users/remco/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

recmo, 2022-03-12:
`CMakeError.log`:

```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc 
Build flags: -march=rv32imc;;-ffunction-sections;-fdata-sections;-mabi=ilp32;-mcmodel=medany
Id flags:  

The output was:
1
/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../libexec/gcc/riscv32-esp-elf/8.4.0/liblto_plugin.so: error loading plugin: dlopen(/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../libexec/gcc/riscv32-esp-elf/8.4.0/liblto_plugin.so, 0x0002): Library not loaded: /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib
  Referenced from: /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/libexec/gcc/riscv32-esp-elf/8.4.0/liblto_plugin.0.so
  Reason: tried: '/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib' (no such file), '/Users/remco/WV/esp32c3/esp32c3-test/target/debug/deps/libstdc++.6.dylib' (no such file), '/Users/remco/WV/esp32c3/esp32c3-test/target/debug/libstdc++.6.dylib' (no such file), '/Users/remco/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/libstdc++.6.dylib' (no such file), '/Users/remco/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/libstdc++.6.dylib' (no such file), '/Users/remco/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/libstdc++.6.dylib' (no such file), '/usr/local/opt/openssl/lib/libstdc++.6.dylib' (no such file), '/libstdc++.6.dylib' (no such file), '/Users/remco/WV/esp32c3/esp32c3-test/target/debug/build/ring-9ab5bf82c8a3f3c5/out/libstdc++.6.dylib' (no such file)
collect2: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ 
Build flags: -march=rv32imc;;-ffunction-sections;-fdata-sections;-mabi=ilp32;-mcmodel=medany
Id flags:  

The output was:
1
/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../libexec/gcc/riscv32-esp-elf/8.4.0/liblto_plugin.so: error loading plugin: dlopen(/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../libexec/gcc/riscv32-esp-elf/8.4.0/liblto_plugin.so, 0x0002): Library not loaded: /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib
  Referenced from: /Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/libexec/gcc/riscv32-esp-elf/8.4.0/liblto_plugin.0.so
  Reason: tried: '/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib' (no such file), '/Users/remco/WV/esp32c3/esp32c3-test/target/debug/deps/libstdc++.6.dylib' (no such file), '/Users/remco/WV/esp32c3/esp32c3-test/target/debug/libstdc++.6.dylib' (no such file), '/Users/remco/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/libstdc++.6.dylib' (no such file), '/Users/remco/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/libstdc++.6.dylib' (no such file), '/Users/remco/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/libstdc++.6.dylib' (no such file), '/usr/local/opt/openssl/lib/libstdc++.6.dylib' (no such file), '/libstdc++.6.dylib' (no such file), '/Users/remco/WV/esp32c3/esp32c3-test/target/debug/build/ring-9ab5bf82c8a3f3c5/out/libstdc++.6.dylib' (no such file)
collect2: error: ld returned 1 exit status

Determining if the include file sys/types.h exists failed with the following output:
Change Dir: /Users/remco/WV/esp32c3/esp32c3-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-2f34b9498a53221f/out/build/CMakeFiles/CMakeTmp

Run Build Command(s):/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/ninja/1.10.2/ninja cmTC_e8fd7 && [1/2] Building C object CMakeFiles/cmTC_e8fd7.dir/CheckIncludeFile.c.obj
FAILED: CMakeFiles/cmTC_e8fd7.dir/CheckIncludeFile.c.obj 
/Users/remco/WV/esp32c3/esp32c3-test/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc   -march=rv32imc  -ffunction-sections -fdata-sections -mabi=ilp32 -mcmodel=medany -o CMakeFiles/cmTC_e8fd7.dir/CheckIncludeFile.c.obj -c CheckIncludeFile.c
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/types.h:75,
                 from CheckIncludeFile.c:1:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:837:2: error: #error Unsupported architecture
 #error Unsupported architecture
  ^~~~~
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/types.h:78,
[... cut ...]
[... 337 lines cut ...]
static char info_size[] =  {'I', 'N', 'F', 'O', ':', 's','i','z','e','[',
  ('0' + ((SIZE / 10000)%10)),
  ('0' + ((SIZE / 1000)%10)),
  ('0' + ((SIZE / 100)%10)),
  ('0' + ((SIZE / 10)%10)),
  ('0' +  (SIZE    % 10)),
  ']',
#ifdef KEY
  ' ','k','e','y','[', KEY, ']',
#endif
  '\0'};

#ifdef __CLASSIC_C__
int main(argc, argv) int argc; char *argv[];
#else
int main(int argc, char *argv[])
#endif
{
  int require = 0;
  require += info_size[argc];
  (void)argv;
  return require;
}

```

MabezDev, 2022-03-15:
v5.0 of esp-idf is not officially released (master is currently 5.0-dev), hence issues like this can be expected. Easier way to sort this would be to use the latest stable v4.4 release.

closed 2022-03-15 by MabezDev