esp-rs/esp-idf-template #17: build ok but run failure genarated code

IoTServ, 2022-03-13:
config:
``` toml
ESP_IDF_VERSION = { value = "master" }
ESP_IDF_TOOLS_INSTALL_DIR="global"
```
code:
``` rust
use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported

fn main() {
    // Temporary. Will disappear once ESP-IDF 4.4 is released, but for now it is necessary to call this function once,
    // or else some patches to the runtime implemented by esp-idf-sys might not link properly.
    esp_idf_sys::link_patches();
    println!("Hello, world!");
}
```
Chicp: esp32c3
Rust toolchain: esp or nightly
build and minitor:
```
espflash  --monitor /dev/tty.wchusbserial542A0047881 target/riscv32imc-esp-espidf/debug/rust-native
Serial port: /dev/tty.wchusbserial542A0047881
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       60:55:f9:72:dd:f4
[00:00:01] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:10] ########################################      94/94      segment 0x10000

Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

�
 �ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
```

ivmarkov, 2022-03-13:
Maybe you could first try one of the more stable releases rather than master?

MabezDev, 2022-03-13:
> ets_loader.c 78

This is stuck in the ROM bootloader, chances are you have some weird flash that you have to tell the bootloader about, or the second stage bootloader on the master branch in borked.

IoTServ, 2022-03-14:
> > ets_loader.c 78
> 
> This is stuck in the ROM bootloader, chances are you have some weird flash that you have to tell the bootloader about, or the second stage bootloader on the master branch in borked.

ok,thanks!
I will try v4.4 IDF

IoTServ, 2022-03-14:
> Maybe you could first try one of the more stable releases rather than master?

> Maybe you could first try one of the more stable releases rather than master?

ok,thanks!
I will try v4.4 or other version IDF