esp-rs/esp-idf-template #18: Failed at esp-idf-sys: `error: unknown target triple 'xtensa', please use -triple or -arch`

xloc, 2022-04-05:
I failed to build the empty project. Can I ask for some help?

The build was for `esp32` and `with std`. I followed the toolchain setup tutorial on [esp-rs](https://esp-rs.github.io/book/dependencies/installing-rust.html), and wanted to use this to test if my setup is working.

`esp-idf-sys` complains about
```
process didn't exit successfully: `/Users/olir/workspace/rust-m5stick/test2/target/debug/build/esp-idf-sys-ad48d41d0437a733/build-script-build` (exit status: 101)
```

And it seems to do with `bindgen.rs`
```
  Output: "/Users/olir/workspace/rust-m5stick/test2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8a41fc2756c3eccc/out/bindings.rs"
  Bindgen builder flags: ["/Users/olir/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.0/src/include/esp-idf/bindings.h", "--rust-target", "1.47", "--blocklist-function", "strtold", "--blocklist-function", "_strtold_r", "--blocklist-function", "v.*printf", "--blocklist-function", "v.*scanf", "--blocklist-function", "_v.*printf_r", "--blocklist-function", "_v.*scanf_r", "--blocklist-function", "esp_log_writev", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-DHAVE_CONFIG_H", "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "-DUNITY_INCLUDE_CONFIG_H", "-DWITH_POSIX", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_ringbuf/include", "-I/Users/olir/workspace/rust-m5stick/test2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8a41fc2756c3eccc/out/build/config", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/newlib/platform_include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/include/esp_additions/freertos", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/port/xtensa/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/include/esp_additions", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/include/soc", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/include/soc/esp32", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/port/esp32/.", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/port/esp32/private_include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/heap/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/log/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/lwip/include/apps", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/lwip/include/apps/sntp", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/lwip/lwip/src/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/lwip/port/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/lwip/port/esp32/include/arch", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/soc/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/soc/esp32/.", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/soc/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/hal/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/hal/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/hal/platform_port/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_rom/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_rom/include/esp32", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_rom/esp32", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_common/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_system/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_system/port/soc", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_system/port/public_compat", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/xtensa/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/xtensa/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/driver/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/driver/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_pm/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/efuse/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/efuse/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/vfs/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_wifi/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_event/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_netif/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_eth/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/tcpip_adapter/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_phy/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_phy/esp32/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_ipc/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/app_trace/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_timer/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/mbedtls/port/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/mbedtls/mbedtls/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/mbedtls/esp_crt_bundle/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/app_update/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/spi_flash/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/bootloader_support/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/nvs_flash/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/pthread/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_gdbstub/include", "-I/Users/olir/worksp [... cut ...] IDF_COMP_FATFS_ENABLED", "-DESP_IDF_COMP_MDNS_ENABLED", "-DESP_IDF_COMP_MQTT_ENABLED", "-DESP_IDF_COMP_NVS_FLASH_ENABLED", "-DESP_IDF_COMP_SPI_FLASH_ENABLED", "-DESP_IDF_COMP_PTHREAD_ENABLED", "-DESP_IDF_COMP_SOC_ENABLED", "-DESP_IDF_COMP_SPIFFS_ENABLED", "-DESP_IDF_COMP_VFS_ENABLED", "-DESP_IDF_COMP_ESP_WIFI_PROVISIONING_ENABLED", "-target", "xtensa"]
  error: unknown target triple 'xtensa', please use -triple or -arch
```

The full log is shown below. Do you have any idea of what is happening?

xloc, 2022-04-05:
```

    Updating crates.io index
   Compiling compiler_builtins v0.1.67
   Compiling core v0.0.0 (/Users/olir/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.36
   Compiling memchr v2.4.1
   Compiling unicode-xid v0.2.2
   Compiling libc v0.2.121
   Compiling cfg-if v1.0.0
   Compiling syn v1.0.90
   Compiling log v0.4.16
   Compiling regex-syntax v0.6.25
   Compiling serde_derive v1.0.136
   Compiling glob v0.3.0
   Compiling lazy_static v1.4.0
   Compiling crossbeam-utils v0.8.8
   Compiling rustversion v1.0.6
   Compiling serde v1.0.136
   Compiling unicode-width v0.1.9
   Compiling same-file v1.0.6
   Compiling fnv v1.0.7
   Compiling bitflags v1.3.2
   Compiling unicode-segmentation v1.9.0
   Compiling libc v0.2.120
   Compiling minimal-lexical v0.2.1
   Compiling once_cell v1.10.0
   Compiling bindgen v0.59.2
   Compiling either v1.6.1
   Compiling vec_map v0.8.2
   Compiling ansi_term v0.12.1
   Compiling anyhow v1.0.56
   Compiling serde_json v1.0.79
   Compiling strsim v0.8.0
   Compiling humantime v2.1.0
   Compiling termcolor v1.1.3
   Compiling cc v1.0.69
   Compiling remove_dir_all v0.5.3
   Compiling zero v0.1.2
   Compiling itoa v1.0.1
   Compiling rustc-hash v1.1.0
   Compiling peeking_take_while v0.1.2
   Compiling cc v1.0.73
   Compiling fastrand v1.7.0
   Compiling shlex v1.1.0
   Compiling ryu v1.0.9
   Compiling lazycell v1.3.0
   Compiling std v0.0.0 (/Users/olir/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling paste v1.0.7
   Compiling libloading v0.7.3
   Compiling clang-sys v1.3.1
   Compiling textwrap v0.11.0
   Compiling walkdir v2.3.2
   Compiling thread_local v1.1.4
   Compiling heck v0.3.3
   Compiling xmas-elf v0.8.0
   Compiling cmake v0.1.48
   Compiling unwind v0.0.0 (/Users/olir/.rustup/toolchains/esp/lib/rustlib/src/rust/library/unwind)
   Compiling quote v1.0.17
   Compiling aho-corasick v0.7.18
   Compiling bstr v0.2.17
   Compiling nom v7.1.1
   Compiling atty v0.2.14
   Compiling which v4.2.5
   Compiling dirs-sys v0.3.7
   Compiling tempfile v3.3.0
   Compiling remove_dir_all v0.7.0
   Compiling clap v2.34.0
   Compiling dirs v4.0.0
   Compiling regex v1.5.5
[... 1527 lines cut ...]
  From https://github.com/espressif/tinyusb
   * branch            c4badd394eda18199c0196ed0be1e2d635f0a5f6 -> FETCH_HEAD
  From https://github.com/ThrowTheSwitch/Unity
   * branch            7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 -> FETCH_HEAD
  From https://github.com/leethomason/tinyxml2
   * branch            7e8e249990ec491ec15990cf95b6d871a66cf64a -> FETCH_HEAD
  From https://github.com/espressif/esp-cryptoauthlib
   * branch            36d0642e66ff5b1c7a291873f24c498ca6ffedef -> FETCH_HEAD
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  Output: "/Users/olir/workspace/rust-m5stick/test2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8a41fc2756c3eccc/out/bindings.rs"
  Bindgen builder flags: ["/Users/olir/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.0/src/include/esp-idf/bindings.h", "--rust-target", "1.47", "--blocklist-function", "strtold", "--blocklist-function", "_strtold_r", "--blocklist-function", "v.*printf", "--blocklist-function", "v.*scanf", "--blocklist-function", "_v.*printf_r", "--blocklist-function", "_v.*scanf_r", "--blocklist-function", "esp_log_writev", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-DHAVE_CONFIG_H", "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "-DUNITY_INCLUDE_CONFIG_H", "-DWITH_POSIX", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_ringbuf/include", "-I/Users/olir/workspace/rust-m5stick/test2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8a41fc2756c3eccc/out/build/config", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/newlib/platform_include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/include/esp_additions/freertos", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/port/xtensa/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/freertos/include/esp_additions", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/include/soc", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/include/soc/esp32", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/port/esp32/.", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/esp_hw_support/port/esp32/private_include", "-I/Users/olir/workspace/rust-m5stick/test2/.embuild/espressif/esp-idf-release/v4.4/components/heap/include", "-I/User [... cut ...]
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /Users/olir/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.59.2/src/ir/context.rs:538:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

MabezDev, 2022-04-05:
Please follow the instructions to install the toolchain: https://esp-rs.github.io/book/dependencies/installing-rust.html#using-a-pre-built-release.

FYI you're missing the Xtensa enabled clang, you can download it separately if you don't want to use the script.

closed 2022-04-05 by MabezDev

SergioGasquez, 2023-04-12:
> however the toolchain link was lost, should i use the espup? https://esp-rs.github.io/book/installation/installation.html#espup

Yes, please, follow the instructions in that section!

kyrosle, 2023-04-13:
hello, i want know that, the export-esp.ps1 would work after i execute it once? in windows, and then i don't have to execute it every time i use the terminal?

SergioGasquez, 2023-04-13:
> hello, i want know that, the export-esp.ps1 would work after i execute it once? in windows, and then i don't have to execute it every time i use the terminal?

You need to source it in every terminal or add the contents to your environment variables