esp-rs/esp-idf-template #19: Build failed

turgu1, 2022-04-08:
Hello,
I've just tried to create a test project using the template:

OS: Ubuntu 20.04
ESP-IDF: V4.4
Rust ESP version: 1.60

After generating the project, I used `cargo build` and it failed to build the project. As a newbie with Rust, I've no clue on how to resolve the issue. Here is the output from the terminal:

```
turgu1@phobos:~/Dev/rust-trials$ cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cargo
⚠️   Unable to load config file: /home/turgu1/.cargo/cargo-generate.toml
🤷   Project Name : test
🔧   Generating template ...
? 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, v5.0 = development; NOTE: applicable only with `cargo build --features native`✔ 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, v5.0 = development; NOTE: applicable only with `cargo build --features native`) · v4.4
✔ 🤷   Rust toolchain (beware: nightly works only for esp32c3!) · esp
✔ 🤷   MCU · esp32
✔ 🤷   STD support · true
[1/9]   Done: .cargo/config.toml
[2/9]   Done: .cargo
[3/9]   Done: .gitignore
[4/9]   Done: Cargo.toml
[5/9]   Done: build.rs
[6/9]   Done: rust-toolchain.toml
[7/9]   Done: sdkconfig.defaults
[8/9]   Done: src/main.rs
[9/9]   Done: src
🔧   Moving generated files into: `/home/turgu1/Dev/rust-trials/test`...
💡   Initializing a fresh Git repository
✨   Done! New project created /home/turgu1/Dev/rust-trials/test
turgu1@phobos:~/Dev/rust-trials$ cd test
turgu1@phobos:~/Dev/rust-trials/test$ ls
build.rs  Cargo.toml  rust-toolchain.toml  sdkconfig.defaults  src
turgu1@phobos:~/Dev/rust-trials/test$ echo $PATH
/home/turgu1/esp/esp-idf/components/esptool_py/esptool:/home/turgu1/esp/esp-idf/components/espcoredump:/home/turgu1/esp/esp-idf/components/partition_table:/home/turgu1/esp/esp-idf/components/app_update:/home/turgu1/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/turgu1/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/turgu1/.espressif/tools/ninja/1.10.2/:/home/turgu1/.espressif/python_env/idf4.4_py3.8_env/bin:/home/turgu1/esp/esp-idf/tools:/home/turgu1/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/:/home/turgu1/.cargo/bin:/home/turgu1/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/turgu1/.platformio/penv/bin
turgu1@phobos:~/Dev/rust-trials/test$ cargo build
    Updating crates.io index
   Compiling compiler_builtins v0.1.70
   Compiling core v0.0.0 (/home/turgu1/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.37
   Compiling libc v0.2.122
   Compiling memchr v2.4.1
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.91
   Compiling cfg-if v1.0.0
   Compiling log v0.4.16
   Compiling regex-syntax v0.6.25
   Compiling glob v0.3.0
   Compiling serde_derive v1.0.136
   Compiling rustversion v1.0.6
   Compiling crossbeam-utils v0.8.8
   Compiling serde v1.0.136
   Compiling lazy_static v1.4.0
   Compiling unicode-width v0.1.9
   Compiling same-file v1.0.6
   Compiling bitflags v1.3.2
   Compiling once_cell v1.10.0
   Compiling minimal-lexical v0.2.1
   Compiling libc v0.2.120
   Compiling unicode-segmentation v1.9.0
   Compiling fnv v1.0.7
   Compiling anyhow v1.0.56
   Compiling vec_map v0.8.2
   Compiling either v1.6.1
   Compiling strsim v0.8.0
   Compiling termcolor v1.1.3
   Compiling serde_json v1.0.79
   Compiling ansi_term v0.12.1
   Compiling humantime v2.1.0
   Compiling bindgen v0.59.2
[... 213 lines cut ...]
  From https://github.com/tatsuhiro-t/neverbleed
   * branch            b967ca054f48a36f82d8fcdd32e54ec5144f2751 -> FETCH_HEAD
  From https://github.com/espressif/esp-thread-lib
   * branch            9a8d34d8f698cad2c9468468b473e26a3dda51b9 -> FETCH_HEAD
  From https://github.com/espressif/openthread
   * branch            c36c0e77a2465355bcf13bd7dc718d8c9aa6ff64 -> FETCH_HEAD
  From https://github.com/protobuf-c/protobuf-c
   * branch            dac1a65feac4ad72f612aab99f487056fbcf5c1a -> FETCH_HEAD
  From https://github.com/pellepl/spiffs
   * branch            f5e26c4e933189593a71c6b82cda381a7b21e41c -> FETCH_HEAD
  From https://github.com/espressif/tinyusb
   * branch            c4badd394eda18199c0196ed0be1e2d635f0a5f6 -> FETCH_HEAD
  From https://github.com/ThrowTheSwitch/Unity
   * branch            7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 -> FETCH_HEAD
  From https://github.com/leethomason/tinyxml2
   * branch            7e8e249990ec491ec15990cf95b6d871a66cf64a -> FETCH_HEAD
  From https://github.com/espressif/esp-cryptoauthlib
   * branch            36d0642e66ff5b1c7a291873f24c498ca6ffedef -> FETCH_HEAD
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  Error: Could not install esp-idf

  Caused by:
      command '"python3" "/home/turgu1/Dev/rust-trials/test/.embuild/espressif/esp-idf-release/v4.4/tools/idf_tools.py" "--idf-path" "/home/turgu1/Dev/rust-trials/test/.embuild/espressif/esp-idf-release/v4.4" "--quiet" "--non-interactive" "install-python-env"' exited with non-zero status code 1
```

turgu1, 2022-04-09:
I found the problem: I needed to set the default toolchain to esp: `rustup default esp`. Please update the README file to indicate that.

Thanks!

MabezDev, 2022-04-10:
Are you sure that was it? That _shouldn't_ be needed, as we have this [rust-toolchain.toml](https://github.com/esp-rs/esp-idf-template/blob/master/cargo/rust-toolchain.toml) file to use the esp toolchain.

turgu1, 2022-04-10:
Well...
I have tried without selecting the esp toolchain and it worked. Sorry for this. I've tried so many things to get it working that I may have lost the path of actions I have used.

Thanks for your answer!
Guy

closed 2022-04-10 by turgu1