esp-rs/esp-idf-template #20: Error when compiling ESP-IDF:Remote branch v5.0 not found in upstream origin

qwerty-liu, 2022-04-09:
It seems that v5.0 is not a branch of ESP-IDF. Or is it just temporarily unavailable? 

Compiling esp-idf-sys v0.31.0

error: failed to run custom build command for `esp-idf-sys v0.31.0`

Caused by:

  process didn't exit successfully: `/home/leohan/Code/test/target/debug/build/esp-idf-sys-991ede0d5d6a6d4d/build-script-build` (exit status: 1)

  --- stdout

  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR

  cargo:rerun-if-env-changed=ESP_IDF_VERSION

  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY

  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS

  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG

  cargo:rerun-if-env-changed=MCU

  --- stderr

  Cloning into '/home/leohan/Code/test/.embuild/espressif/esp-idf-v5.0'...

  warning: Could not find remote branch v5.0 to clone.

  fatal: Remote branch v5.0 not found in upstream origin
  Compiling esp-idf-sys v0.31.0

error: failed to run custom build command for `esp-idf-sys v0.31.0`

Caused by:

  process didn't exit successfully: `/home/leohan/Code/test/target/debug/build/esp-idf-sys-991ede0d5d6a6d4d/build-script-build` (exit status: 1)

  --- stdout

  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR

  cargo:rerun-if-env-changed=ESP_IDF_VERSION

  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY

  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS

  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG

  cargo:rerun-if-env-changed=MCU

  --- stderr

  Cloning into '/home/leohan/Code/test/.embuild/espressif/esp-idf-v5.0'...

  warning: Could not find remote branch v5.0 to clone.

  fatal: Remote branch v5.0 not found in upstream origin

  Error: Could not install esp-idf

  Caused by:

      command '"git" "clone" "--recursive" "--depth" "1" "--shallow-submodules" "--branch" "v5.0" "https://github.com/espressif/esp-idf.git" "/home/leohan/Code/test/.embuild/espressif/esp-idf-v5.0"' exited with non-zero status code 128

  Error: Could not install esp-idf

  Caused by:

      command '"git" "clone" "--recursive" "--depth" "1" "--shallow-submodules" "--branch" "v5.0" "https://github.com/espressif/esp-idf.git" "/home/leohan/Code/test/.embuild/espressif/esp-idf-v5.0"' exited with non-zero status code 128

igrr, 2022-04-09:
That's correct, ESP-IDF v5.0 is in development and hasn't been released yet. The latest stable release at the moment is v4.4.

qwerty-liu, 2022-04-09:
Sorry for not being clear:
I did not choose v5.0.during cargo generate. Choosing v4.4 gave me the above error.

however, on another machine choosing v4.4 works fine. 

any suggestion?

MabezDev, 2022-04-10:
If you can reproduce the buggy project generation, then please open another issue. If possible, please attach, or link the generated project so we can inspect it.

closed 2022-04-10 by MabezDev