esp-rs/esp-idf-template #25: Unable to build: Could not install esp-idf

ruabmbua, 2022-04-20:
Here is the part that fails (build-script of esp-idf-sys):

```
Running `/home/rruckerbauer/code/esp/rust-esp-demo/target/debug/build/esp-idf-sys-1d697082439a814a/build-script-build`
error: failed to run custom build command for `esp-idf-sys v0.31.0`

Caused by:
  process didn't exit successfully: `/home/rruckerbauer/code/esp/rust-esp-demo/target/debug/build/esp-idf-sys-1d697082439a814a/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  PATH=/home/rruckerbauer/code/esp/rust-esp-demo/.embuild/espressif/esp-idf-release/v4.4/tools:$PATH

  --- stderr
  fatal: Keine Namen gefunden, kann nichts beschreiben.
  fatal: Keine Namen gefunden, kann nichts beschreiben.
  Error: Could not install esp-idf

  Caused by:
      command '"python3" "/home/rruckerbauer/code/esp/rust-esp-demo/.embuild/espressif/esp-idf-release/v4.4/tools/idf_tools.py" "--idf-path" "/home/rruckerbauer/code/esp/rust-esp-demo/.embuild/espressif/esp-idf-release/v4.4" "--quiet" "--non-interactive" "install-python-env"' exited with non-zero status code 1
warning: build failed, waiting for other jobs to finish...
```

N3xed, 2022-04-23:
Could you try running
```bash
python3 "/home/rruckerbauer/code/esp/rust-esp-demo/.embuild/espressif/esp-idf-release/v4.4/tools/idf_tools.py" --idf-path "/home/rruckerbauer/code/esp/rust-esp-demo/.embuild/espressif/esp-idf-release/v4.4" install-python-env
```
and see why that fails?

ruabmbua, 2022-04-23:
Same thing happens. `(fatal: Keine Namen gefunden, kann nichts beschreiben.)`

But I think I figured out the problem, I had a different installation of esp-idf tools in my path already. I removed it temporarily, and after doing a fresh cargo build + cleaning the .embuild folder it works!

N3xed, 2022-04-24:
Yeah, but this error (`fatal: Keine Namen gefunden, kann nichts beschreiben.`) is not actually causing it to fail, so it must be something else. In the command I sent I removed the `--quiet` flag which should print more information.
Nonetheless, it's very weird that `idf_tools.py` fails.

> But I think I figured out the problem, I had a different installation of esp-idf tools in my path already. I removed it temporarily, and after doing a fresh cargo build + cleaning the .embuild folder it works!

👍

Okay, with that I'll close the issue (you can always reopen though!).

closed 2022-04-24 by N3xed