esp-rs/esp-idf-template #27: Boot loop after flash

Trbe, 2022-05-28:
After I did `cargo espflash --monitor /dev/ttyCH343USB0` then I get this
```
Serial port: /dev/ttyCH343USB0
Connecting...

    Finished dev [optimized + debuginfo] target(s) in 0.05s
Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       60:55:f9:77:57:38
[00:00:01] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:23] ########################################     212/212     segment 0x10000

Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

�
 �ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42 [chip726_phyrom_version_num:??:??]
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42 [chip726_phyrom_version_num:??:??]
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42 [chip726_phyrom_version_num:??:??]
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42 [chip726_phyrom_version_num:??:??]
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x3ce00000,len:0x92840
ets_loader.c 78
...
```
loop in rst:0x7 and rst:0x10
While I re-plug the board, it boots normally.
```
$~/.cargo/bin/espmonitor /dev/ttyCH343USB0

ESPMonitor 0.7.0

Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

Opening /dev/ttyCH343USB0 with speed 115200
Resetting device... done
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (30) boot: compile time 12:10:40
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32c3: SPI Speed      : 80MHz
I (45) boot.esp32c3: SPI Mode       : DIO
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=19228h (102952) map
I (129) esp_image: segment 1: paddr=00029250 vaddr=3fc89c00 size=0152ch (  5420) load
I (130) esp_image: segment 2: paddr=0002a784 vaddr=40380000 size=05894h ( 22676) load
I (139) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=47f7ch (294780) map
I (187) esp_image: segment 4: paddr=00077fa4 vaddr=40385894 size=041b4h ( 16820) load
I (191) esp_image: segment 5: paddr=0007c160 vaddr=50000010 size=00010h (    16) load
I (196) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
I (216) cpu_start: Pro cpu up.
I (224) cpu_start: Pro cpu start user code
I (224) cpu_start: cpu freq: 160000000
I (224) cpu_start: Application information:
I (227) cpu_start: Project name:     libespidf
I (232) cpu_start: App version:      1
I (237) cpu_start: Compile time:     May 27 2022 23:26:18
I (243) cpu_start: ELF file SHA256:  0000000000000000...
I (249) cpu_start: ESP-IDF:          692c1a6
I (254) heap_init: Initializing. RAM available for dynamic allocation:
I (261) heap_init: At 3FC8C040 len 00033FC0 (207 KiB): DRAM
I (267) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (274) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (281) spi_flash: detected chip: mxic
I (285) spi_flash: flash io: dio
I (289) sleep: Configure to isolate all GPIO pins in sleep state
I (295) sleep: Enable automatic switching of GPIO sleep configuration
I (303) cpu_start: Starting scheduler.
Hello, ESP32-C3!
```
It seems like the loading address is wrong.

MabezDev, 2022-05-30:
> Saved PC:0x40049a42 [chip726_phyrom_version_num:??:??]

This means it's hanging inside the ROM bootloader, the second stage bootloader (part of what is flashed to the board) isn't even running yet.

The fact when you cycle the power it starts working shows that everything is flashed correctly. Very strange :thinking:. What happens when you try and flash the board with esptool.py?

MabezDev, 2022-07-14:
Hey sorry this dropped off my radar, could you try again with the 1.6 release of espflash? Hopefully that fixes these strange issues!

Trbe, 2022-07-16:
Oops, new errors.

<details>
<summary>build and flash</summary>

```
% cargo espflash --monitor /dev/ttyCH343USB0
Serial port: /dev/ttyCH343USB0
Connecting...

    Finished dev [optimized + debuginfo] target(s) in 0.05s
Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       60:55:f9:77:57:38
[00:00:01] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:22] ########################################     209/209     segment 0x10000

Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

�
 �ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x3ce00000,len:0x93040
load:0x05436348,len:0x1384aaff
Invalid image block, can't boot.
ets_main.c 333
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
0x40047ed2 - chip726_phyrom_version_num
    at ??:??
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x3ce00000,len:0x93040
load:0x05436348,len:0x1384aaff
Invalid image block, can't boot.
ets_main.c 333
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
0x40047ed2 - chip726_phyrom_version_num
    at ??:??
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x3ce00000,len:0x93040
load:0x05436348,len:0x1384aaff
Invalid image block, can't boot.
ets_main.c 333
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x3ce00000,len:0x93040
load:0x05436348,len:0x1384aaff
Invalid image block, can't boot.
ets_main.c 333
...
```
</details>

<details>
<summary>re-plug</summary>

```
% ~/.cargo/bin/espmonitor /dev/ttyCH343USB0
ESPMonitor 0.10.0

Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

Opening /dev/ttyCH343USB0 with speed 115200
Resetting device... done
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
SHA-256 comparison failed:
Calculated: 1d0b6cb886e242c146ec7b08edea125baaad603db54764d975ad0ede9ca690e1
Expected: ce79d81cbc9eb8b3c144ab91af3b2017cd1705f94c238631ad1c82817d418aa9
Attempting to boot anyway...
entry 0x403ce000
I (48) boot: ESP-IDF ee2029c 2nd stage bootloader
I (49) boot: compile time 01:39:15
I (49) boot: chip revision: 3
I (50) boot.esp32c3: SPI Speed      : 80MHz
I (55) boot.esp32c3: SPI Mode       : DIO
I (60) boot.esp32c3: SPI Flash Size : 4MB
I (65) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (107) boot_comm: chip revision: 3, min. application chip revision: 0
I (115) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=17418h ( 95256) map
I (138) esp_image: segment 1: paddr=00027440 vaddr=3fc89c00 size=014ech (  5356) load
I (139) esp_image: segment 2: paddr=00028934 vaddr=40380000 size=076e4h ( 30436) load
I (150) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=45208h (283144) map
I (197) esp_image: segment 4: paddr=00075230 vaddr=403876e4 size=023a4h (  9124) load
I (199) esp_image: segment 5: paddr=000775dc vaddr=50000010 size=00010h (    16) load
I (205) boot: Loaded app from partition at offset 0x10000
I (208) boot: Disabling RNG early entropy source...
I (225) cpu_start: Pro cpu up.
I (234) cpu_start: Pro cpu start user code
I (234) cpu_start: cpu freq: 160000000
I (234) cpu_start: Application information:
I (237) cpu_start: Project name:     libespidf
I (242) cpu_start: App version:      1
I (246) cpu_start: Compile time:     Jul 17 2022 01:39:13
I (252) cpu_start: ELF file SHA256:  0000000000000000...
I (258) cpu_start: ESP-IDF:          ee2029c
I (263) heap_init: Initializing. RAM available for dynamic allocation:
I (270) heap_init: At 3FC8C000 len 00034000 (208 KiB): DRAM
I (277) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (283) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (290) spi_flash: detected chip: mxic
I (294) spi_flash: flash io: dio
I (298) sleep: Configure to isolate all GPIO pins in sleep state
I (305) sleep: Enable automatic switching of GPIO sleep configuration
I (312) cpu_start: Starting scheduler.
Hello, world!
```
</details>

hyx0329, 2022-11-17:
I encounter the issue as well, posting my logs

+ espflash v1.7.0 and v2.0.0-rc.1 behaves (almost) the same by default

It's okay to

<details>
<summary>use esptool.py</summary>

```
> esptool.py -c esp32c3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o hi.bin target/riscv32imc-esp-espidf/debug/esp32c3-rust-hands-on
esptool.py v4.3
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.

> file hi.bin
hi.bin: DOS executable (COM), start instruction 0xe905022f 42043840

> esptool.py -c esp32c3 --port /dev/ttyACM0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x10000 hi.bin
esptool.py v4.3
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: redacted
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00010000 to 0x0007bfff...
Compressed 441728 bytes to 221467...
Wrote 441728 bytes (221467 compressed) at 0x00010000 in 6.4 seconds (effective 556.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

> espmonitor /dev/ttyACM0
ESPMonitor 0.10.0

Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

Opening /dev/ttyACM0 with speed 115200
Resetting device... done
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (30) boot: compile time 12:10:40
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32c3: SPI Speed      : 80MHz
I (45) boot.esp32c3: SPI Mode       : DIO
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[... 250 lines cut ...]
I (195) boot: Loaded app from partition at offset 0x10000
I (195) boot: Disabling RNG early entropy source...
I (207) cpu_start: Pro cpu up.
I (215) cpu_start: Pro cpu start user code
I (216) cpu_start: cpu freq: 160000000
I (216) cpu_start: Application information:
I (218) cpu_start: Project name:     libespidf
I (223) cpu_start: App version:      1
I (228) cpu_start: Compile time:     Nov 17 2022 14:18:57
I (234) cpu_start: ELF file SHA256:  0000000000000000...
I (240) cpu_start: ESP-IDF:          76caa64-dirty
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (253) heap_init: At 3FC8C190 len 00050580 (321 KiB): DRAM
I (259) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (266) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (272) spi_flash: detected chip: mxic
I (276) spi_flash: flash io: dio
I (281) sleep: Configure to isolate all GPIO pins in sleep state
I (287) sleep: Enable automatic switching of GPIO sleep configuration
I (294) cpu_start: Starting scheduler.
hi there!

```

</details>

MabezDev, 2022-12-22:
So I found out that some flash chips soldered to dev boards (I believe some XMC chips) behave very strangely and the ROM bootloader may not know about this, which could be why it fails. The stub has workarounds for these chips so that should be used instead. Soon we will switch to using the stub by default but for now you should pass `--use-stub` to espflash.

closed 2022-12-22 by MabezDev