esp-rs/esp-idf-template #37: Container failing on wokwi-server compile

mortenhauan, 2022-07-08:
In VSCode on starting DevContainer for first time, it stops with the following error:

```
#0 397.9    Compiling espflash v1.5.2-dev (https://github.com/esp-rs/espflash#d0bd11b7)
#0 399.6    Compiling wokwi-server
[2022-07-08T09:37:44.529Z] v0.1.0-dev (/home/esp/.cargo/git/checkouts/wokwi-server-86f94bc568cd28c9/7df8ad1)
#0 408.5 error[E0432]: unresolved import `tokio::task::JoinSet`
#0 408.5   --> src/main.rs:11:5
#0 408.5    |
#0 408.5 11 | use tokio::task::JoinSet;
#0 408.5    |     ^^^^^^^^^^^^^^^^^^^^ no `JoinSet` in `task`
#0 408.5
#0 408.5 error[E0432]: unresolved import `espflash::elf::FirmwareImageBuilder`
#0 408.5   --> src/main.rs:15:5
#0 408.5    |
#0 408.5 15 | use espflash::elf::FirmwareImageBuilder;
#0 408.5    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `FirmwareImageBuilder` in `elf`
#0 408.5
#0 408.5 error[E0061]: this function takes 8 arguments but 5 arguments were supplied
#0 408.5    --> src/main.rs:155:27
#0 408.5     |
#0 408.5 155 |     let image = opts.chip.get_flash_image(&firmware, b, p, None, None)?;
#0 408.5     |                           ^^^^^^^^^^^^^^^ ---------  -  -  ----  ---- supplied 5 arguments
#0 408.5     |                           |
#0 408.5     |                           expected
[2022-07-08T09:37:44.529Z] 8 arguments
#0 408.5     |
#0 408.5 note: associated function defined here
#0 408.5    --> /home/esp/.cargo/git/checkouts/espflash-85399b032e8bdfe6/d0bd11b/espflash/src/chip/mod.rs:197:12
#0 408.5     |
#0 408.5 197 |     pub fn get_flash_image<'a>(
#0 408.5     |            ^^^^^^^^^^^^^^^
#0 408.5
#0 408.6 Some errors have detailed explanations: E0061, E0432.
#0 408.6 For more information about an error, try `rustc --explain E0061`.
#0 408.6 error: could not compile `wokwi-server` due to 3 previous errors
#0 408.6 error: failed to compile `wokwi-server v0.1.0-dev (https://github.com/MabezDev/wokwi-server#7df8ad12)`, intermediate artifacts can be found at `/tmp/cargo-install2T0YkN`
------
[2022-07-08T09:37:44.530Z] error: failed to solve: executor failed running [/bin/sh -c chmod a+x ${INSTALL_RUST_TOOLCHAIN}     && ./${INSTALL_RUST_TOOLCHAIN}     --extra-crates "ldproxy cargo-espflash wokwi-server web-flash"     --clear-cache "YES" --export-file /home/${CONTAINER_USER}/export-esp.sh     --esp-idf-version "${ESP_IDF_VERSION}"     --minified-esp-idf "YES"     --build-target "${ESP_BOARD}"     && rustup component add clippy rustfmt]: exit code: 101
```

I can see that it is related to [this issue](https://github.com/MabezDev/wokwi-server/issues/9) in wokwi-server. I tried adding `ENV RUSTFLAGS="--cfg tokio_unstable"` to the Dockerfile. That gets around the `no JoinSet in task` error, but it still failes on the other errors.

mortenhauan, 2022-07-08:
Removing wokwi-server from `--extra-crates "ldproxy cargo-espflash wokwi-server web-flash" \` in Dockerfile fixes it for now.

MabezDev, 2022-07-08:
You also need the `--locked` flag, see https://github.com/MabezDev/wokwi-server#installation. This was fixed in [this](https://github.com/georgik/rustzx-esp32/blob/d403dec03c49f8b8b7c146275c705b9121b624b3/.gitpod.Dockerfile#L35)  Dockerfile, but didn't get updated here.

mortenhauan, 2022-07-08:
Thank you @MabezDev. My current working Dockerfile is this:
```
ARG VARIANT=bullseye
FROM debian:${VARIANT}
ENV DEBIAN_FRONTEND=noninteractive
ENV LC_ALL=C.UTF-8
ENV LANG=C.UTF-8

# Arguments
ARG CONTAINER_USER=esp
ARG CONTAINER_GROUP=esp
ARG TOOLCHAIN_VERSION=1.61.0.0
ARG ESP_IDF_VERSION=release/v4.4
ARG ESP_BOARD=esp32
ARG INSTALL_RUST_TOOLCHAIN=install-rust-toolchain.sh

# Install dependencies
RUN apt-get update \
    && apt-get install -y git curl gcc clang ninja-build libudev-dev unzip xz-utils\
    python3 python3-pip python3-venv libusb-1.0-0 libssl-dev pkg-config libtinfo5 libpython2.7 \
    && apt-get clean -y && rm -rf /var/lib/apt/lists/* /tmp/library-scripts

# Set users
RUN adduser --disabled-password --gecos "" ${CONTAINER_USER}
USER ${CONTAINER_USER}
WORKDIR /home/${CONTAINER_USER}

# Install Rust toolchain, extra crates and esp-idf
ENV PATH=${PATH}:/home/${CONTAINER_USER}/.cargo/bin:/home/${CONTAINER_USER}/opt/bin

ADD --chown=${CONTAINER_USER}:${CONTAINER_GROUP} \
    https://github.com/esp-rs/rust-build/releases/download/v${TOOLCHAIN_VERSION}/${INSTALL_RUST_TOOLCHAIN} \
    /home/${CONTAINER_USER}/${INSTALL_RUST_TOOLCHAIN}

RUN chmod a+x ${INSTALL_RUST_TOOLCHAIN} \
    && ./${INSTALL_RUST_TOOLCHAIN} \
    --extra-crates "ldproxy cargo-espflash web-flash" \
    --clear-cache "YES" --export-file /home/${CONTAINER_USER}/export-esp.sh \
    --esp-idf-version "${ESP_IDF_VERSION}" \
    --minified-esp-idf "YES" \
    --build-target "${ESP_BOARD}" \
    && rustup component add clippy rustfmt

# Install wokwi-server
RUN RUSTFLAGS="--cfg tokio_unstable" cargo install wokwi-server --git https://github.com/MabezDev/wokwi-server --locked

# Activate ESP environment
RUN echo "source /home/${CONTAINER_USER}/export-esp.sh" >> ~/.bashrc

CMD [ "/bin/bash" ]
```

SergioGasquez, 2022-07-08:
Which platform are you using? I assume you are using aarch64 (m1), is that correct? As you mentioned, at this moment, installing `wokwi-server` via `--extra-crates` argument of the installer fails as it does not set `RUSTFLAGS` and does not use `--locked`. This [is already fixed](https://github.com/esp-rs/rust-build/pull/91), but since the Dockerfile pulls the installer from the releases, those changes are not present, I would recommend using your Dockerfile as a temporary solution, and waiting for `1.62.0.0` release which will probably come up next week and would solve the issue.

MabezDev, 2022-07-08:
Also https://github.com/tokio-rs/tokio/issues/4535 is nearing stabilization, so we won't need the unstable cfg soon.

mortenhauan, 2022-07-12:
I'm using M1 yes, @SergioGasquez. It is no problem using my own Dockerfile as a temporary solution for now.

SergioGasquez, 2022-07-13:
Update: Release 1.62.0.0 is out! Using your custom Dockerfile should no longer be required, I'll try to do some testing later on today to confirm that is working fine

SergioGasquez, 2022-07-22:
Hello @mortenhauan, did you have a chance to confirm that is working for you? I would like to close the issue as it should be already fixed with the `1.62.0.0` release

closed 2022-08-09 by SergioGasquez