esp-rs/esp-idf-template #45: Drop support for `4.3.2`

SergioGasquez, 2022-07-22:
With the new release of `esp-idf v5.0` coming up pretty soon. What would you think about dropping the support for `v4.3.2`? I suppose we should be pushing users to use the latest stable (which should be `v4.4`) or the latest unstable (`v5.0`).

Maybe is still too soon as `v5.0` is still very unstable, but wanted to gather opinions about this topic

MabezDev, 2022-07-22:
I think leaving `v4.3.2` as an option is fine, there are probably some people using it. We should probably define somewhere the level of support. For instance, 4.3.x is no longer getting Rust related backports, we are only backporting to 4.4 now.

SergioGasquez, 2022-07-26:
As we accorded in our 2[6/7/2022 community meeting](https://github.com/esp-rs/rust/discussions/130) we will close this issue until esp-idf `v5.0` gets released and stabilized for Rust, once this happens we will revisit this issue and evaluate it again

closed 2022-07-26 by SergioGasquez