esp-rs/esp-idf-template #47: failed to compile:  thread 'main' panicked at 'libclang error;

mutantbob, 2022-07-26:
I am having difficulty building the project from the template using `cargo +esp build --release` .

I built a docker container using 

```Dockerfile
FROM ubuntu:focal

RUN apt clean

RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y \
    git \
    curl \
    gcc \
    clang \
    ninja-build \
    cmake \
    libudev-dev \
    unzip \
    xz-utils\
    python3 \
    python3-pip \
    python3-venv \
    libusb-1.0-0 \
    libssl-dev \
    pkg-config \
    libtinfo5 \
    libpython2.7

RUN git clone https://github.com/esp-rs/rust-build.git
RUN cd rust-build && \
    curl -LO https://github.com/esp-rs/rust-build/releases/download/v1.62.1.0/install-rust-toolchain.sh && \
    chmod a+x install-rust-toolchain.sh && \
    ./install-rust-toolchain.sh

#RUN rustup update

ENV PATH=/root/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
RUN echo $PATH

RUN cargo install cargo-generate
```

Inside that container I ran 
```
# cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template 
✔  Which template should be expanded? · cargo
 Project Name : blink2
 Destination: /esp32-experiments/blink2 ...
 Generating template ...
✔  MCU · esp32
✔  Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not available for esp-idf v4.3.2) · false
✔  STD support · true
✔  ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, mainline = UNSTABLE) · v4.4
[ 1/10]   Done: .cargo/config.toml
[ 2/10]   Done: .cargo
[ 3/10]   Done: .gitignore
[ 4/10]   Done: .vscode
[ 5/10]   Done: Cargo.toml
[ 6/10]   Done: build.rs
[ 7/10]   Done: rust-toolchain.toml
[ 8/10]   Done: sdkconfig.defaults
[ 9/10]   Done: src/main.rs
[10/10]   Done: src
 Moving generated files into: `/esp32-experiments/blink2`...
 Done! New project created /esp32-experiments/blink2
# cargo +esp build --release
    Updating crates.io index
   Compiling compiler_builtins v0.1.71
   Compiling core v0.0.0 (/root/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.42

...

  From https://github.com/leethomason/tinyxml2
   * branch            7e8e249990ec491ec15990cf95b6d871a66cf64a -> FETCH_HEAD
  From https://github.com/espressif/esp-cryptoauthlib
   * branch            36d0642e66ff5b1c7a291873f24c498ca6ffedef -> FETCH_HEAD
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.1 at '/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4'
  fatal: No names found, cannot describe anything.
  Output: "/esp32-experiments/blink2/target/xtensa-esp32-espidf/release/build/esp-idf-sys-d77b6f1aaf74c425/out/bindings.rs"
  Bindgen builder flags: ["/root/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.6/src/include/esp-idf/bindings.h", "--rust-target", "1.47", "--blocklist-function", "strtold", "--blocklist-function", "_strtold_r", "--blocklist-function", "v.*printf", "--blocklist-function", "v.*scanf", "--blocklist-function", "_v.*printf_r", "--blocklist-function", "_v.*scanf_r", "--blocklist-function", "esp_log_writev", "--no-layout-tests", "--with-derive-default", "--ctypes-prefix", "c_types", "--generate", "functions,types,vars,methods,constructors,destructors", "--use-core", "--no-rustfmt-bindings", "--", "-D__bindgen", "-DHAVE_CONFIG_H", "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"", "-DUNITY_INCLUDE_CONFIG_H", "-DWITH_POSIX", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_ringbuf/include", "-I/esp32-experiments/blink2/target/xtensa-esp32-espidf/release/build/esp-idf-sys-d77b6f1aaf74c425/out/build/config", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/newlib/platform_include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/freertos/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/freertos/include/esp_additions/freertos", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/freertos/port/xtensa/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/freertos/include/esp_additions", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/include/soc", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/include/soc/esp32", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/port/esp32/.", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/port/esp32/private_include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/heap/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/log/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/lwip/include/apps", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/lwip/include/apps/sntp", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/lwip/lwip/src/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/lwip/port/esp32/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/lwip/port/esp32/include/arch", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/soc/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/soc/esp32/.", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/soc/esp32/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/hal/esp32/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/hal/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/hal/platform_port/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/include/esp32", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_common/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/port/soc", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/port/public_compat", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp32/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/xtensa/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/xtensa/esp32/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/driver/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/driver/esp32/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_pm/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/efuse/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/efuse/esp32/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/vfs/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_wifi/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_event/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_netif/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_eth/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/tcpip_adapter/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_phy/include", "-I/esp32-experiments/blink2/.embuild/espressif/esp-idf/release-v4.4/components/esp_phy/esp32/include", "-I/esp32-experiments/blink2/.embuild/e [... cut ...]
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.59.2/src/ir/context.rs:538:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Do you have any suggestions for how to build the template application?

mutantbob, 2022-07-27:
I banged on it with a rock until I figured out what the problem was.

I had to switch to
`./install-rust-toolchain.sh --export-file export-esp-rust.sh `
And then I could 
`. /rust-build/export-esp-rust.sh ; cargo +esp build`

It seems the --export-file was not listed in the Installation commands primary example from https://github.com/esp-rs/rust-build .

closed 2022-07-27 by mutantbob

ivmarkov, 2022-07-27:
Jfyi the original error is due to the build not finding the xtensa clang compiler on the path. It is installed together with the esp rust compiler.