esp-rs/esp-idf-template #49: ESP-IDF Environment Auto-Loading

vpochapuis, 2022-07-29:
Sorry to bother, I am new with those tools.
I previously used ESP-IDF with C. I know we needed to always do a `get_idf` before compiling, but for esp-rs, I thought that given this templates will generate at build a full ESP-IDF environment in the `.embuild` it was designed to automatically load the environment when doing `cargo build`? (Its a supposition it might not be the case, please let me know).

So my question is: ***Is my setup incorrect, or should we always manually source the `export.sh` file?*** (Manually doing it would be quite troublesome as when using Rust Analyzer with Flycheck, as they are not in the same env, they would break in VsCode for example and I didn't find yet a workaround).

For further information on my issue when launching `cargo check` or `cargo build`, here is the output (I think its the usual output for a missing ESP-IDF Python Env, but there might be other information useful if it was supposed to be loaded automatically):

```
cargo check
   Compiling esp-idf-sys v0.31.6
error: failed to run custom build command for `esp-idf-sys v0.31.6`

Caused by:
  process didn't exit successfully: `/[...]/rust-tests/nrg-hub/target/debug/build/esp-idf-sys-01984a316360a59f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/[...]/rust-tests/nrg-hub/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/cmake/3.23.1/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/ninja/1.10.2/:/[...]/rust-tests/nrg-hub/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  Current system platform: linux-amd64
  Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
  IDF_PYTHON_ENV_PATH=/[...]/rust-tests/nrg-hub/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/cmake/3.23.1/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/ninja/1.10.2/:/[...]/rust-tests/nrg-hub/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  cargo:rerun-if-changed=/[...]/rust-tests/nrg-hub/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/[...]/rust-tests/nrg-hub/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/[...]/rust-tests/nrg-hub/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/[...]/rust-tests/nrg-hub/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  The following Python requirements are not satisfied:
  click>=7.0
  pyserial>=3.3
  future>=0.15.2
  cryptography>=2.1.4
  pyparsing>=2.0.3,<2.4.0
  pyelftools>=0.22
  idf-component-manager~=1.0
  gdbgui==0.13.2.0
  pygdbmi<=0.9.0.2
  python-socketio<5
  jinja2<3.1  # See https://github.com/espressif/esp-idf/issues/8760
  itsdangerous<2.1
  kconfiglib==13.7.1
  reedsolo>=1.5.3,<=1.5.4
  bitstring>=3.1.6
  ecdsa>=0.16.0
  construct==2.10.54
  Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
  Diagnostic information:
      IDF_PYTHON_ENV_PATH: (not set)
      Python interpreter used: /[...]/rust-tests/nrg-hub/.embuild/espressif/python_env/idf4.4_py3.10_env/bin/python
      Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
      PATH: /[...]/rust-tests/nrg-hub/.embuild/espressif/tools/cmake/3.23.1/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/ninja/1.10.2/:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/esp-idf/release-v4.4/tools:/[...]/rust-tests/nrg-hub/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/[...]/rust-tests/nrg-hub/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/home/vchapuis/.espressif/tools/xtensa-esp32-elf-gcc/8_4_0-esp-2021r2-patch3-x86_64-unknown-linux-gnu/bin/:/home/vchapuis/.espressif/tools/xtensa-esp32s2-elf-gcc/8_4_0-esp-2021r2-patch3-x86_64-unknown-linux-gnu/bin/:/home/vchapuis/.espressif/tools/xtensa-esp32s3-elf-gcc/8_4_0-esp-2021r2-patch3-x86_64-unknown-linux-gnu/bin/:/home/vchapuis/.espressif/tools/:/home/vchapuis/.emacs.d/bin:/home/vchapuis/.local/bin:/home/vchapuis/.cargo/bin:/home/vchapuis/.cargo/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
  -- Configuring incomplete, errors occurred!
  See also "/[...]/rust-tests/nrg-hub/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.1 at '/[...]/rust-tests/nrg-hub/.embuild/espressif/esp-idf/release-v4.4'
  fatal: No names found, cannot describe anything.
  CMake Error at /[...]/rust-tests/nrg-hub/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/build.cmake:287 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /[...]/rust-tests/nrg-hub/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/build.cmake:416 (__build_check_python)
    CMakeLists.txt:6 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /home/vchapuis/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Thanks for your help.

ivmarkov, 2022-07-29:
It is suppised to download and fetch everything
 Are you running in a python virtual environment? If yes, exit that.

vpochapuis, 2022-07-29:
> It is supposed to download and fetch everything Are you running in a python virtual environment? If yes, exit that.

I tried again from scratch and it worked when building manually.
What I think happened, is like you said, I must have `cargo build` for the first time after doing an `get_idf` command and must have messed with the configs. 
I think I was mixed up with the *cargo first* and *idf first* approaches. 

So it works well when I do the command myself.
However when opening the project in VsCode, the Rust-Analyzer build fails with:

```
[ERROR rust_analyzer::lsp_utils] rust-analyzer failed to run build scripts

error: failed to run custom build command for `esp-idf-sys v0.31.6`

Caused by:
  process didn't exit successfully: `/tmp/test-project/target/debug/build/esp-idf-sys-01984a316360a59f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=MCU
  IDF_PYTHON_ENV_PATH=/tmp/test-project/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/tmp/test-project/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/tmp/test-project/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/tmp/test-project/.embuild/espressif/tools/cmake/3.23.1/bin:/tmp/test-project/.embuild/espressif/tools/ninja/1.10.2/:/tmp/test-project/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  Current system platform: linux-amd64
  Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
  IDF_PYTHON_ENV_PATH=/tmp/test-project/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/tmp/test-project/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/tmp/test-project/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/tmp/test-project/.embuild/espressif/tools/cmake/3.23.1/bin:/tmp/test-project/.embuild/espressif/tools/ninja/1.10.2/:/tmp/test-project/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  cargo:rerun-if-changed=/tmp/test-project/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/tmp/test-project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DCMAKE_INSTALL_PREFIX=/tmp/test-project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/requirements.txt are satisfied.
  -- Project sdkconfig file /tmp/test-project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out/sdkconfig
  Loading defaults file /tmp/test-project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out/gen-sdkconfig.defaults...
  Loading defaults file /tmp/test-project/sdkconfig.defaults...
  -- App "libespidf" version: 1
  -- Adding linker script /tmp/test-project/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fdd86283b5a21988/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
  -- Adding linker script /tmp/test-project/.embuild/espressif/esp-idf/release-v4.4/components/soc/esp32/ld/esp32.peripherals.ld
  -- Configuring done
  -- Generating done
[... cut ...]
               at /home/vchapuis/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.29.3/src/bindgen.rs:162:20
    10: embuild::bindgen::run
               at /home/vchapuis/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.29.3/src/bindgen.rs:149:5
    11: build_script_build::main
               at ./build/build.rs:91:25
    12: core::ops::function::FnOnce::call_once
               at /home/rust/rust/library/core/src/ops/function.rs:248:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

```

Any idea why would this happen?

Thank you.

ivmarkov, 2022-07-29:
Yes. When you install the rust compiler using `rust-build` it tells you at the end of the installation to put something in your `~/.bashrc`. Please do that. That something is the Xtensa Clang compiler. Without it, you cannot really generate the Rust bindings for ESP-IDF.

ivmarkov, 2022-07-29:
If it is not clear what is happening: your shell does have Xtensa clang in your $path, while VScode does not have access to it, because you did not put it in `.bashrc` and did not restart VScode.

closed 2022-07-29 by ivmarkov

vpochapuis, 2022-07-29:
Thanks for everything @ivmarkov !

I restarted everything and checked my `.bashrc` the export PATH was there. VsCode was still unhappy so I added the CLANG Path to the `.vscode/settings.json`  `"rust-analyzer.server.extraEnv"` and it just works now! 
Thank you for showing me it was a CLANG path issue, it was very helpful!

I am grateful to have all this help.
Wish you all the best.