esp-rs/esp-idf-template #51: Compiling hello world - esp-idf-sys - stderr Build configuration: BuildConfig

carlin54, 2022-08-10:
Hey, 

I am quite excited to get this to work. I have had quite a bit of trouble.

What I am trying to do is to get something compiled and onto my ESP32. 

Might be more related to esp-idf-sys. 

Here is one of the things that I have tried to do. 

➜  Git cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cargo
🤷   Project Name : project
🔧   Destination: /home/richard/Git/project ...
🔧   Generating template ...
? 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not av✔ 🤷   Configure project to use Dev Containers (VS Code, GitHub Codespaces and Gitpod)? (beware: Dev Containers not available for esp-idf v4.3.2) · true
✔ 🤷   MCU · esp32
✔ 🤷   ESP-IDF native build version (v4.3.2 = previous stable, v4.4 = stable, mainline = UNSTABLE) · v4.4
✔ 🤷   STD support · true
[ 1/24]   Done: .cargo/config.toml
[ 2/24]   Done: .cargo
[ 3/24]   Done: .devcontainer/Dockerfile
[ 4/24]   Done: .devcontainer/devcontainer.json
[ 5/24]   Done: .devcontainer
[ 6/24]   Done: .dockerignore
[ 7/24]   Done: .gitignore
[ 8/24]   Done: .gitpod.Dockerfile
[ 9/24]   Done: .gitpod.yml
[10/24]   Done: .vscode/launch.json
[11/24]   Done: .vscode/tasks.json
[12/24]   Done: .vscode
[13/24]   Done: Cargo.toml
[14/24]   Done: build.rs
[15/24]   Done: docs/README.md
[16/24]   Done: docs
[17/24]   Done: rust-toolchain.toml
[18/24]   Done: scripts/build.sh
[19/24]   Done: scripts/flash.sh
[20/24]   Done: scripts/run-wokwi.sh
[21/24]   Done: scripts
[22/24]   Done: sdkconfig.defaults
[23/24]   Done: src/main.rs
[24/24]   Done: src
🔧   Moving generated files into: `/home/richard/Git/project`...
✨   Done! New project created /home/richard/Git/project
➜  Git cd /home/richard/Git/project

➜  esp-idf git:(release/v4.4) ✗ git checkout release/v4.4 
M	components/bt/host/nimble/nimble
M	components/esp_phy/lib
M	components/esp_wifi/lib
M	components/esptool_py/esptool
M	components/ieee802154/lib
M	components/lwip/lwip
M	components/mbedtls/mbedtls
M	components/mqtt/esp-mqtt
M	components/openthread/lib
M	components/openthread/openthread
M	components/tinyusb/tinyusb
M	examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib
Already on 'release/v4.4'
Your branch is up to date with 'origin/release/v4.4'.
➜  esp-idf git:(release/v4.4) ✗ pwd
/home/richard/esp/esp-idf
➜  esp-idf git:(release/v4.4) ✗ git submodule update --init --recursive 
Submodule path 'components/bt/host/nimble/nimble': checked out 'dac3a2a0f54589b10a4cff2a99526804747089df'
Submodule path 'components/esp_phy/lib': checked out 'ff0d771b8e33e320e11634567ee53b9cd78e6be1'
Submodule path 'components/esp_wifi/lib': checked out '85d6197b8f4271f51a409c0cd7e293ae2694145c'
Submodule path 'components/ieee802154/lib': checked out 'f7b5e8059a3bb6f321e79ac3bf2aa4d2a9b93326'
Submodule path 'components/lwip/lwip': checked out '76303df2386902e0d7873be4217f1d9d1b50f982'
Submodule path 'components/mbedtls/mbedtls': checked out '89da85968417ce4a163b7275cb22f159995c271d'
[... 110 lines cut ...]
     1:     0x5592d71b059e - core::fmt::write::hd3936cb33cf524d4
     2:     0x5592d716ddd1 - std::io::Write::write_fmt::hdd80ac53afad4740
     3:     0x5592d7173375 - std::panicking::default_hook::{{closure}}::ha1fd9d555c583f5d
     4:     0x5592d7172ffa - std::panicking::default_hook::hc37a1a17cf5ce086
     5:     0x5592d717387c - std::panicking::rust_panic_with_hook::h818296db12e1c172
     6:     0x5592d7192bb7 - std::panicking::begin_panic_handler::{{closure}}::hcb2048160d2bd008
     7:     0x5592d7191f64 - std::sys_common::backtrace::__rust_end_short_backtrace::h8106bfaf0dca55fa
     8:     0x5592d7173542 - rust_begin_unwind
     9:     0x5592d698fbe3 - core::panicking::panic_fmt::ha8187868327e1820
    10:     0x5592d6c1e9fd - cmake::fail::hf8ea13342af4ecc2
    11:     0x5592d6c1e43e - cmake::run::hdc82ba4860964bbd
    12:     0x5592d6c1b01c - cmake::Config::build::h309f8b6a3bf11c84
    13:     0x5592d69ec84d - build_script_build::native::cargo_driver::build::h0928d80ad948051d
    14:     0x5592d69e0dfc - build_script_build::native::build::h9ed16d7a30efc811
    15:     0x5592d69b7492 - build_script_build::main::h662f52b46fdc2986
    16:     0x5592d6a04d06 - core::ops::function::FnOnce::call_once::h8ab18b3a7e586893
    17:     0x5592d69b1e59 - std::sys_common::backtrace::__rust_begin_short_backtrace::h51be1fe5382073af
    18:     0x5592d69e326c - std::rt::lang_start::{{closure}}::hb40aa8284739eabc
    19:     0x5592d716d751 - std::rt::lang_start_internal::h4f59c50d5a45d217
    20:     0x5592d69e3251 - std::rt::lang_start::h35ae57f21736ef18
    21:     0x5592d69ba423 - main
    22:     0x7f4495708083 - __libc_start_main
                                 at /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:308:16
    23:     0x5592d698fe8e - _start
    24:                0x0 - <unknown>

ivmarkov, 2022-08-10:
I think you have an activated esp idf environment, or at least an active python venv when trying to build the project, which is not supported (yet). Please exit all of that and try the build again.

migepatschen, 2022-12-04:
Hi all, just something similar.

I just set up the `esp-rs` toolchain using `espup` on a Win10 machine not previously used for ESP32 development.
The no-std "hello world" works, but the generated code from the `esp-idf-template` does not compile because:
`IDF_PYTHON_ENV_PATH: (not set)`

I did run the `export-esp.ps1` before compiling.

I did a setup yesterday on a Win10 previously used for ESP32 development with the Arduino IDE as well as PlattformIO and everything worked out of the box with the `esp-rs` toolchain on that PC.

SergioGasquez, 2023-06-23:
Since there has been no activity in the issue, and it works for other users and in CI, I'll close the issue. Feel free to reopen is the issue is still not resolved.

closed 2023-06-23 by SergioGasquez