esp-rs/esp-idf-template #54: error: components/driver/include/driver/i2s_types.h: No such file or directory

tradeJmark, 2022-09-04:
This is the result I get when I try to do a `cargo build`, and I haven't been able to find anyone else who had the same issue. As far as I know, I've exactly followed the setup instructions. More output for context:
```
$ cargo build
   Compiling esp-idf-sys v0.31.8
error: failed to run custom build command for `esp-idf-sys v0.31.8`

Caused by:
  process didn't exit successfully: `/home/tim/tmp/rst/hello-world/hello-world/target/debug/build/esp-idf-sys-052a90a0c1dd4292/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Branch(
                  "release/v4.4",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/home/tim/.espressif/frameworks/esp-idf-v4.4",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v4.4.0 environment at '/home/tim/.espressif/frameworks/esp-idf-v4.4'
  error: components/driver/include/driver/i2s_types.h: No such file or directory
  error: components/driver/include/driver/i2s_types.h: No such file or directory
  Error: command '"git" "--git-dir" "/home/tim/.espressif/frameworks/esp-idf-v4.4/.git" "--work-tree" "/home/tim/.espressif/frameworks/esp-idf-v4.4" "apply" "/home/tim/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.8/patches/i2s_types_master.diff"' exited with non-zero status code 1
```

N3xed, 2022-09-07:
Yup, this is definitely a bug in esp-idf-sys. I'll fix it shortly.

N3xed, 2022-09-19:
This was indirectly fixed by esp-rs/esp-idf-sys@55e11cdfef298ec95f59fe596e7267f7c8e1bbac (so works on the latest release) and is now directly fixed by https://github.com/esp-rs/esp-idf-sys/commit/527f7fc42e6a6ca870494de3bf3de6551520c833.

closed 2022-09-19 by N3xed