esp-rs/esp-idf-template #57: building template fails.

zee-mzha, 2022-09-18:
Currently building the template fails for me with the error log below. For the sake of making sure I didn't do something truly horrible I tried building the other non-std template (https://github.com/esp-rs/esp-template) and that one builds just fine with no problems. Any idea what's going on here?
```
[zee@zeeLT engr290-server]$ cargo build
   Compiling esp-idf-sys v0.31.9
error: failed to run custom build command for `esp-idf-sys v0.31.9`

Caused by:
  process didn't exit successfully: `/home/zee/projects/engr290/engr290-server/target/debug/build/esp-idf-sys-9674ad052d4ec0a9/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  IDF_PYTHON_ENV_PATH=/home/zee/projects/engr290/engr290-server/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/home/zee/projects/engr290/engr290-server/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch4-8.4.0/xtensa-esp32-elf/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/tools/cmake/3.23.1/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  Current system platform: linux-amd64
  Skipping xtensa-esp32-elf@esp-2021r2-patch4-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=/home/zee/projects/engr290/engr290-server/.embuild/espressif/python_env/idf4.4_py3.10_env
  PATH=/home/zee/projects/engr290/engr290-server/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch4-8.4.0/xtensa-esp32-elf/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/tools/cmake/3.23.1/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/python_env/idf4.4_py3.10_env/bin:/home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  cargo:rerun-if-changed=/home/zee/projects/engr290/engr290-server/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/zee/projects/engr290/engr290-server/.embuild/espressif/python_env/idf4.4_py3.10_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/requirements.txt are satisfied.
  -- Project sdkconfig file /home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out/sdkconfig
  Loading defaults file /home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out/gen-sdkconfig.defaults...
  Loading defaults file /home/zee/projects/engr290/engr290-server/sdkconfig.defaults...
  -- App "libespidf" version: 1
  -- Adding linker script /home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/soc/esp32/ld/esp32.peripherals.ld
  -- Configuring done
  -- Generating done
  -- Build files have been written to: /home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out/build
  running: "cmake" "--build" "." "--config" "MinSizeRel" "--parallel" "8"
  [1/7] Generating ld/sections.ld
  [2/7] Performing build step for 'bootloader'
  [0/1] Re-running CMake...
  -- Building ESP-IDF components for target esp32
  -- Project sdkconfig file /home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out/sdkconfig
  Loading defaults file /home/zee/projects/engr290/engr290-server/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-734234081bd281a0/out/gen-sdkconfig.defaults...
  Loading defaults file /home/zee/projects/engr290/engr290-server/sdkconfig.defaults...
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/soc/esp32/ld/esp32.peripherals.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
  -- Adding linker script /home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
[... 32 lines cut ...]
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.2 at '/home/zee/projects/engr290/engr290-server/.embuild/espressif/esp-idf/release-v4.4'
  fatal: No names found, cannot describe anything.
  Built components: esp_ringbuf, efuse, esp_ipc, driver, esp_pm, mbedtls, bootloader, esptool_py, partition_table, app_update, bootloader_support, spi_flash, nvs_flash, pthread, esp_gdbstub, espcoredump, esp_phy, esp_system, esp_rom, hal, vfs, esp_eth, tcpip_adapter, esp_netif, esp_event, wpa_supplicant, esp_wifi, ieee [... cut ...]
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /home/zee/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.60.1/src/ir/context.rs:529:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

zee-mzha, 2022-09-18:
aaaaaand nevermind, apparently I forgot to follow some of the instructions in the guide (adding target and adding env vars to end of bashrc). Figures that I solve it after hours soon as I post something. Sorry for the useless issue!

closed 2022-09-18 by zee-mzha

kevin406972552, 2023-03-12:
How exactly did you do that? Thank you