esp-rs/esp-idf-template #59: Build error "fatal error: bits/os_defines.h: No such file or directory"

Brad-Hesson, 2022-10-07:
Trying to build my project after a re-install of windows but I am getting this error.  I then tried to build this template to test if it was a problem specifit to my project, but I got the same error.  I have included more of the build output here:

```
process didn't exit successfully: `C:\Users\slice\Desktop\code\esp-idf-bare\target\release\build\esp-idf-sys-56fcf0fbd311e6fe\build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  IDF_PYTHON_ENV_PATH=C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\python_env\idf4.4_py3.10_env
  PATH=C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\tools\cmake\3.23.1\bin;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\tools\ninja\1.10.2\;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\python_env\idf4.4_py3.10_env\Scripts;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  Current system platform: win64
  Skipping riscv32-esp-elf@esp-2021r2-patch5-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\python_env\idf4.4_py3.10_env
  PATH=C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\tools\cmake\3.23.1\bin;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\tools\ninja\1.10.2\;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\python_env\idf4.4_py3.10_env\Scripts;C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  cargo:rerun-if-changed=C:\Users\slice\Desktop\code\esp-idf-bare\sdkconfig.defaults
  CMAKE_PREFIX_PATH_riscv32imc-esp-espidf = None
  CMAKE_PREFIX_PATH_riscv32imc_esp_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_riscv32imc-esp-espidf = None
  CMAKE_riscv32imc_esp_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "C:\\Users\\slice\\Desktop\\code\\esp-idf-bare\\target\\riscv32imc-esp-espidf\\release\\build\\esp-idf-sys-7875691fe9951d37\\out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=C:\\Users\\slice\\Desktop\\code\\esp-idf-bare\\.embuild\\espressif\\esp-idf\\release-v4.4\\tools\\cmake\\toolchain-esp32c3.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/python_env/idf4.4_py3.10_env/Scripts/python.exe" "-DCMAKE_INSTALL_PREFIX=C:\\Users\\slice\\Desktop\\code\\esp-idf-bare\\target\\riscv32imc-esp-espidf\\release\\build\\esp-idf-sys-7875691fe9951d37\\out" "-DCMAKE_C_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_CXX_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany"
  -- Checking Python dependencies...
  Python requirements from C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\esp-idf\release-v4.4\requirements.txt are satisfied.
  -- Project sdkconfig file C:/Users/slice/Desktop/code/esp-idf-bare/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-7875691fe9951d37/out/sdkconfig
  Loading defaults file C:/Users/slice/Desktop/code/esp-idf-bare/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-7875691fe9951d37/out/gen-sdkconfig.defaults...
  Loading defaults file C:/Users/slice/Desktop/code/esp-idf-bare/sdkconfig.defaults...
  -- App "libespidf" version: 1
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-7875691fe9951d37/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/ld/esp32c3/sections.ld.in
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib-time.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
  -- Adding linker script C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/soc/esp32c3/ld/esp32c3.peripherals.ld
  -- Configuring done
  -- Generating done
  -- Build files have been written to: C:/Users/slice/Desktop/code/esp-idf-bare/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-7875691fe9951d37/out/build
  running: "cmake" "--build" "." "--config" "MinSizeRel" "--parallel" "8"
  [1/925] Building CXX object esp-idf/nvs_flash/CMakeFiles/__idf_nvs_flash.dir/src/nvs_pagemanager.cpp.obj
  FAILED: esp-idf/nvs_flash/CMakeFiles/__idf_nvs_flash.dir/src/nvs_pagemanager.cpp.obj
  C:\Users\slice\Desktop\code\esp-idf-bare\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -IC:/Users/slice/Desktop/code/esp-idf-bare/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-7875691fe9951d37/out/build/config -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/nvs_flash/include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/nvs_flash/private_include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/newlib/platform_include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/freertos/include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/freertos/include/esp_additions/freertos -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/freertos/port/riscv/include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/freertos/include/esp_additions -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/include/soc -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/include/soc/esp32c3 -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/port/esp32c3/. -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/esp_hw_support/port/esp32c3/private_include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/heap/include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/log/include -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/lwip/include/apps -IC:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/lw [... cut ...]
                   from C:/Users/slice/Desktop/code/esp-idf-bare/.embuild/espressif/esp-idf/release-v4.4/components/nvs_flash/src/nvs_pagemanager.cpp:6:
  c:\users\slice\desktop\code\esp-idf-bare\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\riscv32-esp-elf\include\c++\8.4.0\riscv32-esp-elf\rv32imc\ilp32\no-rtti\bits\c++config.h:508:10: fatal error: bits/os_defines.h: No such file or directory
   #include <bits/os_defines.h>
            ^~~~~~~~~~~~~~~~~~~
  compilation terminated.
```

ivmarkov, 2022-10-07:
Windows has a problem with long pathnames. Shorten your project directory name and move it as closer to the drive root as possible.

closed 2022-10-07 by ivmarkov

georgik, 2022-10-07:
@Brad-Hesson I would like to confirm @ivmarkov statement.
This is a limitation for Windows. Please, consider moving the project to shorter path, idealy somewhere neer root of filesystem.
Other option is to use mapping: https://github.com/esp-rs/rust/discussions/111#discussioncomment-2363015

Recommendation for Windows users to avoid long path problem
use `subst r:\ c:\my-rust-project`
https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/subst
The command maps directory to new drive.
The command avoids all problems with short paths problem which is pretty complex and deeply rooted in all tools like git, CMake, CCache, CMD or OS itself

note: Solution with UNC Paths does not work, because CMake is generating CMD based scripts :-/

georgik, 2022-10-07:
@SergioGasquez Would it be possible to add the check to `cargo generate`?