esp-rs/esp-idf-template #71: Can the `esp_idf_sys::link_patches();` call be removed with ESP-IDF 4.4?

faern, 2022-12-20:
The template always generates the following code into `main`:
```rust
// Temporary. Will disappear once ESP-IDF 4.4 is released, but for now it is necessary to call this function once,
// or else some patches to the runtime implemented by esp-idf-sys might not link properly.
esp_idf_sys::link_patches();
```
... Even when the project uses ESP-IDF 4.4 (which is the default now!). Can this code be safely removed from any project using 4.4? Would it make sense to update the template to only conditionally contain this code when an older ESP-IDF is used? Currently the fresh template is a bit confusing since it's contradicting itself a bit.

https://github.com/esp-rs/esp-idf-template/blob/79a7345b69541409db1fea2ee792a4256b1c16d4/cargo/src/main.rs#L10-L12

MabezDev, 2022-12-20:
No, unfortunately, it can't. We should really update the comment generated from the template :laughing:. The idea was that we wouldn't need any more patches after 4.4, but we still have a few - one of which I imagine we'll keep forever (lstat).

faern, 2022-12-20:
Ok! Good to know. But yeah, I agree the comment should be upgraded. Preferably with a link to where the developer can read more about this. I suppose there is some issue or PR or similar where there is more info about the used patches and why they are needed?

closed 2022-12-22 by MabezDev

ActuallyHappening, 2023-04-05:
> 

LOL
This is where that comment links to!

PR #72 (2022-12-22): Update link_patches() comment to reflect current status
Fixes #71 

I'm not very satisfied with the provided URL. It currently only tells the user *that* it's needed, not exactly why. Do you have a better link we can use that helps the user understand the reason for `link_patches`?