esp-rs/esp-idf-template #75: sdkconfig.defaults ignored

br0kenpixel, 2022-12-31:
Hi. I'm having issues changing the SDK configuration.
I'd like to disable logging in the bootloader and HAL as well as a few other things, which can be done by adding the following lines to `sdkconfig.defaults`:
```
CONFIG_BOOTLOADER_LOG_LEVEL=0
CONFIG_BOOTLOADER_WDT_ENABLE=n
CONFIG_HAL_LOG_LEVEL=0
CONFIG_LOG_DEFAULT_LEVEL=0
CONFIG_ESP_SYSTEM_PANIC=ESP_SYSTEM_PANIC_PRINT_HALT
```
After adding these lines to `sdkconfig.defaults` and running `cargo build` (with or without `--release`) I noticed that none of these changes were applied. My ESP32 was still displaying boot logs and calling `panic!()` would make the chip reboot instead of halting as specified above.

I tried:
- Adding `ESP_IDF_SDKCONFIG_DEFAULTS = "sdkconfig.defaults"` to the `[env]` section in `.cargo/config.toml`
- Adding `ESP_IDF_SDKCONFIG = "sdkconfig.defaults"` to the `[env]` section in `.cargo/config.toml`

None of these helped.
Is there extra configuration needed to make this work?

> ESP-IDF version: release/v4.4
> Board: Generic ESP32 dev board
> Project generated using: `cargo generate https://github.com/esp-rs/esp-idf-template cargo`
> STD support: yes, Dev container support: no

Talebna, 2023-07-01:
I'm having this issue. Did you found any workaround ? @br0kenpixel

ivmarkov, 2023-07-01:
The bootloader settings affect the **bootloader**. Meaning - you need to flash the bootloader as well  (which is generated as a by-product from the build). 

Now, the xtal settings are not bootloader settings so it should just work. I've however noticed that sometimes when using a cargo workspace, the sdkconfig is not always picked up. Not sure yet what is the root cause.

As a workaround, use the `$ESP_IDF_SDKCONFIG_DEFAULTS` env variable, as described [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD_OPTIONS.md).

ivmarkov, 2023-10-19:
There is no such thing as `sdkconfig.defaults` being ignored - I've confirmed it. Watch out for:
* Using a workspace and NOT placing the file in the workspace root. It MUST be there;
* Trying flags which affect the bootloader. Bootloader WILL be rebuilt as a result, but `espflash` is flashing **its own** bootloader, unless you pull the bootloader built as a side effect from your project and instruct `espflash` to flash that one instead.

closed 2023-10-19 by ivmarkov

Vollbrecht, 2023-10-19:
> There is no such thing as `sdkconfig.defaults` being ignored - I've confirmed it. Watch out for:
>
>     * Using a workspace and NOT placing the file in the workspace root. It MUST be there;
>
>     * [...]

Its true that a sdkconfing.defaults file itself will not be ignored - but a wrongly typed config inside for example will be ignored. Overall you are correct that this is a espflash/bootloader problem

ivmarkov, 2023-10-19:
> Its true that a sdkconfing.defaults file itself will not be ignored - but a wrongly typed config inside for example will be ignored. 

Sure. But this is on the ESP-IDF side to eventually address/fix.