esp-rs/esp-idf-template #76: Error `psram: Not a valid or known package id: 6` on ESP32-PICO-MINI-02

l0calh05t, 2023-01-03:
I have an Adafruit ESP32 Feather V2 with an ESP32-PICO-MINI-02 and am getting an error / boot loop when trying to run the example:

```
ets Jul 29 2019 12:21:46
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
0x40078000 - __udivmoddi4
    at ??:??
load:0x4007c428,len:4840
0x4007c428 - __udivmoddi4
    at ??:??
entry 0x4007c6a0
0x4007c6a0 - __udivmoddi4
    at ??:??
E (201) psram: Not a valid or known package id: 6
abort() was called at PC 0x400791d7
0x400791d7 - __udivmoddi4
    at ??:??
```

So far I have

- Set the target to `xtensa-esp32-espidf` to match the μC on the module
- Added `--use-stub` to the `runner` setting to get it to upload at all
- Updated the `esp-idf-sys` and `embuild` versions to their current version
- Set `ESP_IDF_VERSION` to `release/v5.0` (incl. setting the `rustflags` and doing a `cargo clean`)
- Added the following to `sdkconfig.defaults`
  ```
  CONFIG_SPIRAM=y
  CONFIG_SPIRAM_TYPE=SPIRAM_TYPE_AUTO
  CONFIG_SPIRAM_CACHE_WORKAROUND=y
  ```

All to no avail, as I keep getting that error. The board itself (still) works fine with Arduino

ryankurte, 2023-01-14:
hey i had the same problem, resolved by running the latest `espflash` from git (seems like there's a flash stub update or something)

l0calh05t, 2023-01-20:
> hey i had the same problem, resolved by running the latest `espflash` from git (seems like there's a flash stub update or something)

Thanks for the hint! Installing espflash 2.0.0-rc.3 instead of the stable 1.7.0 and replacing `espflash --monitor` in `.cargo/cargo.toml` by `espflash flash --monitor` solved the issue for me

closed 2023-01-20 by l0calh05t