esp-rs/esp-idf-template #77: Error connecting to ws://localhost:9012. Please reload the page.

wlopezob, 2023-01-16:
Hi, when execute  **./scripts/run-wokwi.sh debug**  in devcontainer is result open the follow url: https://wokwi.com/_alpha/wembed/xxxxxxxx?partner=espressif&port=9012&data=demo but the page not work.

The error is: Error connecting to ws://localhost:9012. Please reload the page.

Please help me!.
[image]

Thank you.

SergioGasquez, 2023-01-16:
Hi!
Just gave a quick try to simulate the default project template with both the dockerhub image (had to [install `wokwi-server`](https://github.com/MabezDev/wokwi-server#installation)) and building it from the Dockerfile, and had no issues. Could you elaborate on a few ends?
- Are you running the devcontainer using the dockerhub image or building it from the Dockerfile?
- What are you using as [`WOKWI_PROJECT_ID`](https://github.com/esp-rs/esp-idf-template/blob/master/cargo/scripts/run-wokwi.sh#L36)?
- Have you tried to simulate the template project with no changes?
- I assume you are using VsCode, can you check the `PORTS` tab of VsCode? 
[image]

wlopezob, 2023-01-16:
Hi @SergioGasquez .

I was again run the project and it is ok.

Thank you.

closed 2023-01-17 by wlopezob