esp-rs/esp-idf-template #79: missing cfg flags with cmake build

simonchatts, 2023-02-18:
Thanks for all your work on esp-rs!

esp-idf-svc has a bunch of cfg flags, eg [esp_idf_comp_esp_netif_enabled](https://github.com/esp-rs/esp-idf-svc/blob/master/src/lib.rs#L69), that normally get set by some automatic process I don't understand.

This doesn't work with cmake projects from this template. If you:

 - create a std project called "test" with `cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cmake`
 - add `esp-idf-svc = "0.42.5"` to  `components/rust-test/Cargo.toml`
 - add `use esp_idf_svc::netif;` to `components/rust-test/src/lib.rs`

then you get this error:

```
error[E0432]: unresolved import `esp_idf_svc::netif`
 --> src/lib.rs:2:5
  |
2 | use esp_idf_svc::netif;
  |     ^^^^^^^^^^^^^^^^^^ no `netif` in the root
```

If you brute-force it, by adding `set(ESP_RUSTFLAGS "--cfg esp_idf_comp_esp_netif_enabled")` to `components/rust-test/CMakeLists.txt`, then this shifts the error onto other missing flags (such as `esp_idf_comp_esp_event_enabled`).

Any ideas?

ivmarkov, 2023-02-18:
`esp_idf_comp_*` flags get auto-defined by the build system *when you enable the corresponding ESP IDF component [as part of the CMake build](https://github.com/esp-rs/esp-idf-template/blob/master/cmake/components/rust-%7B%7Bproject-name%7D%7D/CMakeLists.txt#L5)*.

closed 2023-02-18 by ivmarkov

simonchatts, 2023-02-18:
Aha! Thank you!