esp-rs/esp-idf-template #83: Python required for building esp-idf-sys

m1212e, 2023-03-11:
After hours of searching for the cause of my build fail with only `error: failed to run custom build command for esp-idf-sys v0.31.12` as an output, I ran `cargo build -vv` to find that python is required to build the `esp-idf-sys`  crate (embuild uses this?).
I would suggest to mention this somewhere. Maybe in the README or the book, I don't know where it would be best. But this took me sooo much time to find out.

N3xed, 2023-03-12:
> `error: failed to run custom build command for esp-idf-sys v0.31.12` as an output

As far as I know, when a build script fails the whole output of it is printed, this wasn't the case? But yeah the error log of `esp-idf-sys` can get a bit unwieldy if you don't know what to look for.

Basically, `esp-idf-sys` uses `git` to clone the `esp-idf` (Espressif's IoT Development Kit). The `esp-idf` makes heavy use of python for tools used during compilation and for installing toolchains and so on (though the installing part can be (mostly) avoided by using `espup`).

> I would suggest to mention this somewhere. Maybe in the README or the book, I don't know where it would be best.

PRs are always welcome.

ivmarkov, 2023-03-12:
In all fairness, in the readme, in the how to build section, the first bullet point is a link to the build prerequisites of the `esp-idf-template` project. There, under the wi dows section, python3 is listed as a clear prerequisite.

What else shall we do? Type "you need python3" in big bold red letters? And then same for the xtensa toolchain?

ivmarkov, 2023-03-12:
Hold on. This IS the `esp-idf-template` project. And you are building for Windows I assume (as python is I stalled by default on linux). And haven't noticed the python prerequisites in the readme?

m1212e, 2023-03-13:
> Hold on. This IS the `esp-idf-template` project. And you are building for Windows I assume (as python is I stalled by default on linux). And haven't noticed the python prerequisites in the readme?

You are so very right, I missed this. Thats on me, sorry.
I did not go through the full README (which I should have) but came from the book which I thought would get me going and just didn't think about it. I got a little confused with the plethora of resources and repositories. Sorry again, esp-rs is a bit overwhelming for newcommers.

closed 2023-03-13 by m1212e