esp-rs/esp-idf-template #86: WOKWI PROJECT ID

GabrielAlonsoCabral, 2023-03-21:
Hey guys, i've started to use esp-idf-template and i enjoy so much.
Also, i've notice that inside ./scripts/run-wokwi.sh we must to set WOKWI_PROJECT_ID (hard coded).

Could i open an PR to get dynamically from .env ?

SergioGasquez, 2023-03-22:
Hi! This could be a great improvement, and, of course, your PR will be more than welcomed. That being said, we will probably update the template to use [Wokwi VS Code extension](https://docs.wokwi.com/vscode/getting-started) instead of using the [wokwi-server crate](https://github.com/MabezDev/wokwi-server). See https://github.com/esp-rs/esp-template/pull/56 for more details.

Vollbrecht, 2023-04-03:
should we add conditional option for adding wokwi support independent of an container or not? It should also generate an diagram.json and wokwi.toml. I would like that

SergioGasquez, 2023-04-03:
> should we add conditional option for adding wokwi support independent of an container or not?

Yes, today I started working on updating the template (https://github.com/SergioGasquez/esp-idf-template/tree/feature/devcontainer-update) and this is something that I would definitely add. I already added it for `esp-template` today: https://github.com/esp-rs/esp-template/commit/c2f792ffbbfecdddd4ee9da6ad6c6d6c146bad42

closed 2023-04-17 by SergioGasquez