esp-rs/esp-idf-template #91: Add `v5.0` choice for esp-idf version

SergioGasquez, 2023-03-31:
The `v5.0` option should follow [`release/v5.0`](https://github.com/espressif/esp-idf/tree/release/v5.0) branch as we do with `v4.4`

Vollbrecht, 2023-04-04:
should this be added, or add and remove the oldstable v4.3.2 ?

Vollbrecht, 2023-04-04:
in that light what do you think of  releasing versions of the cargo-generate files?  This can justify an remove of oldstable v4.3.2. People than just can use the old git release of it ? Just an idea

MabezDev, 2023-04-05:
In the last community meeting, I mentioned that 4.3.x is still supported until December this year, and I didn't see any point in removing it. I've had a slight perspective change since then and am thinking it might be a good idea to remove it as 4.3.x is not recommended for new projects, and this template is all about new projects.

SergioGasquez, 2023-04-14:
> in that light what do you think of releasing versions of the cargo-generate files? This can justify an remove of oldstable v4.3.2 ( as with gitpod support) . People than just can use the old git release of it ? Just an idea

I don't think many people would use it, but it does not hurt, so I don't really mind

SergioGasquez, 2023-04-14:
Also, I added `v5.0` choice [in the branch that Im working](https://github.com/SergioGasquez/esp-idf-template/tree/feature/devcontainer-update), and [CI build is failing](https://github.com/SergioGasquez/esp-idf-template/actions/runs/4698298953/jobs/8330413373) due to https://github.com/esp-rs/esp-idf-svc/issues/196, which was solved in  https://github.com/esp-rs/esp-idf-svc/pull/200 but has not been released yet, hence if we point to the git repo it works, but atm we are using the latest release (`0.45.0`)

closed 2023-04-17 by SergioGasquez