esp-rs/esp-idf-template #97: VSCode setup as an option for non devcontainer setup

Vollbrecht, 2023-04-07:
Many people like vscode, many dont. I personally would like an option to include some vscode setup like in the container base, mainly for the debugging options. 
Setup debugging is hard and we should make it as simple as possible for people. 
Though in light of not overbloading #95  the generating process what you guys think?

SergioGasquez, 2023-04-14:
Not a full solution, but since we are adding a Wokwi standalone prompt that adds support for Wokwi Vs Code extension, it will include a [`.vscode/launch.json`](https://github.com/SergioGasquez/esp-idf-template/blob/feature/devcontainer-update/cargo/.vscode/launch.json) with the required settings to debug

Vollbrecht, 2023-04-14:
one potential problem, depending on what debugging extension we are using, it need to be installed by the user. So using cppdbg would need the microsoft C/C++ extension, using corged-m-debug would Cortex-Debug extension etc. And iirc VSCode is not smart enough to figoure it out. Have to look into it though, maybe we can find something. We dont have this problem in the container where we can setup the vscode extension by ourself.