esp-rs/esp-idf-template #98: Remove/move cmake/pio versions of the template

SergioGasquez, 2023-04-14:
Since cargo-first is the recommended approach, what if we remove the cmake version from this template. It would simplify the repo, and/or it can be moved to another repository like `esp-idf-cmake-template`. 

It is just an idea, I would like to hear any other opinions about this @ivmarkov @MabezDev @JurajSadel @georgik @Vollbrecht @bjoernQ

ivmarkov, 2023-04-14:
I'm strongly against this. What would it solve? Save one prompt? If you are anyway doing the two-prompts approach, that you mentioned elsewhere, what's the point?

As for cargo-pio - it is just a readme file.

I think we might be over-rotating on the simplification idea.

SergioGasquez, 2023-04-14:
As you mentions, it does not save much from user perspective, the main advantage would be cleaning the repository, but if you are strongly against this I am more than happy to close it.

closed 2023-04-14 by SergioGasquez

ivmarkov, 2023-04-14:
Thank you! I think the cmake code is not really that much in terms of code size or comexity. And more importantly - users do not see the repo code size anyway. Just the generated project, which is either cargo-first, or cmake, but not both.