esp-rs/esp-idf-template #106: Generation fails with certain input combination

Vollbrecht, 2023-04-22:
## Bug description
.cargo/config.toml dont get generated correctly 
```
[build]
target = ""

[target.]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x

[unstable]
build-std = ["std", "panic_abort"]

[env]
# Note: these variables are not used when using pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "release/v5.0"
```

## To Reproduce
genreate it template with the following
```
cargo generate --vcs git --git https://github.com/esp-rs/esp-idf-template
✔  Which template should be expanded? · cargo
 Project Name : test-5-sat
 Destination: /home/.../Documents/esp_rust/test-5-sat ...
 Generating template ...
✔  Use the default template values? · false
✔  Which MCU to target? · esp32c3
✔  Enable STD support? · true
✔  ESP-IDF version (mainline = UNSTABLE) · v5.0
✔  Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · false
✔  Configure project to support Wokwi simulation with Wokwi VS Code extension? · false
✔  Include esp-idf-hal/esp-idf-svc? · Yes (default features)
[ 1/11]   Done: .cargo/config.toml
[ 2/11]   Done: .cargo
[ 3/11]   Done: .gitignore
[ 4/11]   Done: .vscode
[ 5/11]   Done: Cargo.toml
[ 6/11]   Done: build.rs
[ 7/11]   Ignored: pre-script.rhai
[ 8/11]   Done: rust-toolchain.toml
[ 9/11]   Done: sdkconfig.defaults
[10/11]   Done: src/main.rs
[11/11]   Done: src
 Moving generated files into: `/home/.../Documents/esp_rust/test-5-sat`...
 Initializing a fresh Git repository
 Done! New project created /home/.../Documents/esp_rust/test-5-sat
 ```

## Expected behavior

Correctly generate the [target.riscv32imc-esp-espidf] section

Vollbrecht, 2023-04-22:
also fails for v4.4 option

SergioGasquez, 2023-04-22:
Just tried reproducing the issue and my `.cargo/config.toml` was properly generated:
```
Documents/Espressif/tests
❯ cargo generate --vcs git --git https://github.com/esp-rs/esp-idf-template

✔ 🤷   Which template should be expanded? · cargo
🤷   Project Name: test-5
🔧   Destination: /home/sergio/Documents/Espressif/tests/test-5 ...
🔧   project-name: test-5 ...
🔧   Generating template ...
✔ 🤷   Use the default template values? · false
✔ 🤷   Which MCU to target? · esp32c3
✔ 🤷   ESP-IDF version (mainline = UNSTABLE) · v5.0
? 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · false
✔ 🤷   Enable STD support? · true
? 🤷   Configure project to support Wokwi simulation with Wokwi VS Code exten✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · false
✔ 🤷   Include esp-idf-hal/esp-idf-svc? · Yes (default features)
🔧   Moving generated files into: `/home/sergio/Documents/Espressif/tests/test-5`...
Initializing a fresh Git repository
✨   Done! New project created /home/sergio/Documents/Espressif/tests/test-5

Documents/Espressif/tests took 30s
❯ cd test-5/

test-5 on  master [?] is 📦 v0.1.0 via 🦀 v1.71.0-nightly
❯ cat .cargo/config.toml
───────┬─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
       │ File: .cargo/config.toml
───────┼─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ [build]
   2   │ target = "riscv32imc-esp-espidf"
   3   │
   4   │ [target.riscv32imc-esp-espidf]
   5   │ linker = "ldproxy"
   6   │ # runner = "espflash --monitor" # Select this runner for espflash v1.x.x
   7   │ runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
   8   │ rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]
   9   │
  10   │ [unstable]
  11   │ build-std = ["std", "panic_abort"]
  12   │
  13   │ [env]
  14   │ # Note: these variables are not used when using pio builder (`cargo build --features pio`)
  15   │ ESP_IDF_VERSION = "release/v5.0"
  16   │
───────┴───────────────────────────────────────────────────────────────────────
```

SergioGasquez, 2023-04-22:
> also fails for v4.4 option

Which `cargo-generate` version are you using?

Vollbrecht, 2023-04-22:
> > also fails for v4.4 option
> 
> Which `cargo-generate` version are you using?

yeah i had v0.16 installed -> updated to 0.18.2 and it works

closed 2023-04-22 by Vollbrecht