esp-rs/esp-idf-template #118: Build fails for cmake template with esp-idf v5.0

mrsch, 2023-05-10:
The cmake template with esp-idf version 5.0 fails to build local and in the CI with the following issue:

```
% idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/mario/tmp/rust-component/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/mario/.espressif/python_env/idf5.1_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/mario/tmp/rust-component"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/home/mario/tmp/rust-component/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.40.1")
-- Component directory /home/mario/tmp/esp-idf/components/asio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/mario/tmp/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/mario/tmp/esp-idf/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/mario/tmp/esp-idf/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/mario/tmp/esp-idf/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/mario/tmp/esp-idf/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/mario/tmp/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mario/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mario/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mario/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/mario/tmp/rust-component/sdkconfig
Loading defaults file /home/mario/tmp/rust-component/sdkconfig.defaults...
Compiler supported targets: xtensa-esp32-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/mario/.espressif/python_env/idf5.1_py3.11_env/bin/python (found version "3.11.3") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "rust-component" version: 1
-- Adding linker script /home/mario/tmp/rust-component/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/mario/tmp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/mario/tmp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/mario/tmp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/mario/tmp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/mario/tmp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/mario/tmp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/mario/tmp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/mario/tmp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread rust-rust-component sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/mario/tmp/esp-idf/components/app_trace /home/mario/tmp/esp-idf/components/app_update /home/mario/tmp/esp-idf/components/bootloader /home/mario/tmp/esp-idf/components/bootloader_support /home/mario/tmp/esp-idf/components/bt /home/mario/tmp/esp-idf/components/cmock /home/mario/tmp/esp-idf/components/console /home/mario/tmp/esp-idf/components/cxx /home/mario/tmp/esp-idf/components/driver /home/mario/tmp/esp-idf/components/efuse /home/mario/tmp/esp-idf/components/esp-tls /home/mario/tmp/esp-idf/components/esp_adc /home/mario/tmp/esp-idf/components/esp_app_format /home/mario/tmp/esp-idf/components/esp_coex /home/mario/tmp/esp-idf/components/esp_common /home/mario/tmp/esp-idf/components/esp_eth /home/mario/tmp/esp-idf/components/esp_event /home/mario/tmp/esp-idf/components/esp_gdbstub /home/mario/tmp/esp-idf/components/esp_hid /home/mario/tmp/esp-idf/components/esp_http_client /home/mario/tmp/esp-idf/components/esp_http_server /home/mario/tmp/esp-idf/components/esp_https_ota /home/mario/tmp/esp-idf/components/esp_https_server /home/mario/tmp/esp-idf/components/esp_hw_support /home/mario/tmp/esp-idf/components/esp_lcd /home/mario/tmp/esp-idf/components/esp_local_ctrl /home/mario/tmp/esp-idf/components/esp_mm /home/mario/tmp/esp-idf/components/esp_netif /home/mario/tmp/esp-idf/components/esp_netif_stack /home/mario/tmp/esp-idf/components/esp_partition /home/mario/tmp/esp-idf/components/esp_phy /home/mario/tmp/esp-idf/components/esp_pm /home/mario/tmp/esp-idf/components/esp_psram /home/mario/tmp/esp-idf/components/esp_ringbuf /home/mario/tmp/esp-idf/components/esp_rom /home/mario/tmp/esp-idf/components/esp_system /home/mario/tmp/esp-idf/components/esp_timer /home/mario/tmp/esp-idf/components/esp_wifi /home/mario/tmp/esp-idf/components/espcoredump /home/mario/tmp/esp-idf/components/esptool_py /home/mario/tmp/esp-idf/components/fatfs /home/mario/tmp/esp-idf/components/freertos /home/mario/tmp/esp-idf/components/hal /home/mario/tmp/esp-idf/components/heap /home/mario/tmp/esp-idf/components/http_parser /home/mario/tmp/esp-idf/components/idf_test /home/mario/tmp/esp-idf/components/ieee802154 /home/mario/tmp/esp-idf/components/json /home/mario/tmp/esp-idf/components/log /home/mario/tmp/esp-idf/components/lwip /home/mario/tmp/rust-component/main /home/mario/tmp/esp-idf/components/mbedtls /home/mario/tmp/esp-idf/components/mqtt /home/mario/tmp/esp-idf/components/newlib /home/mario/tmp/esp-idf/components/nvs_flash /home/mario/tmp/esp-idf/components/openthread /home/mario/tmp/esp-idf/components/partition_table /home/mario/tmp/esp-idf/components/perfmon /home/mario/tmp/esp-idf/components/protobuf-c /home/mario/tmp/esp-idf/components/protocomm /home/mario/tmp/esp-idf/components/pthread /home/mario/tmp/rust-component/components/rust-rust-component /home/mario/tmp/esp-idf/components/sdmmc /home/mario/tmp/esp-idf/components/soc /home/mario/tmp/esp-idf/components/spi_flash /home/mario/tmp/esp-idf/components/spiffs /home/mario/tmp/esp-idf/components/tcp_transport /home/mario/tmp/esp-idf/components/ulp /home/mario/tmp/esp-idf/components/unity /home/mario/tmp/esp-idf/components/usb /home/mario/tmp/esp-idf/components/vfs /home/mario/tmp/esp-idf/components/wear_levelling /home/mario/tmp/esp-idf/components/wifi_provisioning /home/mario/tmp/esp-idf/components/wpa_supplicant /home/mario/tmp/esp-idf/components/xtensa
-- Configuring done (3.3s)
-- Generating done (0.2s)
-- Build files have been written to: /home/mario/tmp/rust-component/build
Running ninja in directory /home/mario/tmp/rust-component/build
Executing "ninja all"...
[... cut ...] p32-elf-gcc "RUSTFLAGS=--cfg espidf_time64" cargo build --target xtensa-esp32-espidf --target-dir /home/mario/tmp/rust-component/build/esp-idf/rust-rust-component/target --release -Zbuild-std=std,panic_abort -Zbuild-std-features=panic_immediate_abort
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/mario/tmp/rust-component/build/log/idf_py_stderr_output_125756 and /home/mario/tmp/rust-component/build/log/idf_py_stdout_output_125756
```

mrsch, 2023-05-10:
I was able to fix this by adding the following to `src/include/esp-idf/bindings.h` in `esp-idf-sys`:

`#include "sys/stat.h"`

Not sure if this is the right way to fix this.

ivmarkov, 2023-05-12:
When you say "esp idf 5.0" which 5.0 version do you mean precisely? The `release/v5.0` branch, or some tag? If a tag, which one?

SergioGasquez, 2023-05-12:
> When you say "esp idf 5.0" which 5.0 version do you mean precisely? The `release/v5.0` branch, or some tag? If a tag, which one?

I suppose it's `release/v5.0` as he mentioned seeing the same issue in the CI

ivmarkov, 2023-05-12:
@mrsch I think I just fixed it. Please issue `cargo update` in the root of your rust crate to pull the latest `esp-idf-sys` version.

Reopen if it still does not work for you.

closed 2023-05-12 by ivmarkov

mrsch, 2023-05-15:
Everything is working now, thanks a lot.