esp-rs/esp-idf-template #121: Support for ESP-IDF 5.1

ivmarkov, 2023-05-13:
@SergioGasquez Given that the upcoming `esp-idf-*` release actually does support ESP IDF 5.1, we should probably be having an option for that release as well?

Vollbrecht, 2023-05-13:
my personal opinion:  maybe just update the 5.0 option -> use 5.1 instead and leave the two other options? as the template is mostly for generating new projects. This way we focus on support for v4.4 and the current v5 release branch.

SergioGasquez, 2023-05-25:
@ivmarkov we discussed this topic in today's community meeting, and there was no hard opinion, but most of us agreed with removing `5.0` option and keeping only the `5.1` option for ESP-IDF 5, do you have any input on this?

closed 2023-06-13 by SergioGasquez