esp-rs/esp-idf-template #127: How to configure esp-idf-template without cargo-generate

zdl361, 2023-06-18:
Ubuntu 22.04.1 x86_64 GNU/Linux
I'm getting a segmentation fault when I running "cargo-generate generate esp-rs/esp-idf-template cargo" .
Both the "cargo install cargo-generate" and the "cargo binstall cargo-generate" have segment fault and I can't fix it.
How to configure esp-idf-template without cargo-generate?

shell:

```bash
!32 ~/src % cargo generate esp-rs/esp-idf-template cargo         
⚠️   Favorite `esp-rs/esp-idf-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-idf-template.git
zsh: segmentation fault  cargo generate esp-rs/esp-idf-template cargo
```

GDB:
```
Core was generated by `/home/zdl361/.cargo/bin/cargo-generate generate esp-rs/esp-idf-template cargo'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:74
```

zdl361, 2023-06-18:
The bug comes from the code that handles git. The solution is to manually download the esp-idf-template to the local path. Use cargo generate to read templates from a local path
```bash
git clone https://github.com/esp-rs/esp-idf-template
cargo generate --vcs none ./esp-idf-template
```

SergioGasquez, 2023-06-18:
Which version of cargo-generate are you using (`cargo generate --version`)?

zdl361, 2023-06-18:
> Which version of cargo-generate are you using (`cargo generate --version`)?

cargo generate-generate 0.18.3

SergioGasquez, 2023-06-18:
> cargo generate-generate 0.18.3

That's very weird, I'm using the same version but no issues, have you tried using the whole git url: `cargo generate https://github.com/esp-rs/esp-idf-template cargo`? And also, to test some other thing, can you try generating esp-template `cargo generate -a esp-rs/esp-template`

SergioGasquez, 2023-07-24:
Hi @zdl361, have you tried any of the suggested solutions?

zdl361, 2023-08-03:
[image]
[image]

ivmarkov, 2023-10-19:
Closing as this is the only ever reported crash with `cargo-generate` and it is anyway an upstream problem.

Enhancing `esp-idf-template` to work also without `cargo-generate` is not an option, as an isolated crash on one developer machine with it is not enough of an evidence that we need to do such a big shift.

closed 2023-10-19 by ivmarkov