esp-rs/esp-idf-template #129: Building a freshly created template fails

flyaruu, 2023-06-21:
Hi, I'm having issues building a vanilla cargo template. I do see some issues in the CI (but not exactly the same)

## Bug description
Build of a vanilla template project fails.

## To Reproduce

Create a templated project:
```
cargo generate --git https://github.com/esp-rs/esp-idf-template cargo --name test54 --vcs none --silent -d mcu=esp32c3 -d defaults=false -d espidfver=v5.1 -d std=true -d hal="Yes (all features)" -d devcontainer=false -d wokwi=false
```
Then build:
```
cargo build -Zbuild-std=std,panic_abort --release
```
yields:
```
...
Compiling embedded-svc v0.25.1
error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /Users/f.lyaruu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.1/src/uart.rs:1091:25
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                         ^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /Users/f.lyaruu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.1/src/uart.rs:1091:60
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                                                            ^^^^^^^^^^^^ not found in this scope

error[E0560]: struct `esp_idf_sys::uart_config_t` has no field named `source_clk`
   --> /Users/f.lyaruu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.1/src/uart.rs:932:9
    |
932 |         source_clk: config.source_clock.into(),
    |         ^^^^^^^^^^ `esp_idf_sys::uart_config_t` does not have this field
    |
    = note: available fields are: `__bindgen_anon_1`
```
- Would you like to work on a fix? [y/n]

## Expected behavior

Build completing successfully

## Environment

- OS: MacOS, latest espup

Vollbrecht, 2023-06-21:
related/duplicate to https://github.com/esp-rs/esp-idf-template/issues/128 can you please use 
```toml
[patch.crates-io]
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
``` 
building against that works for me with the template. This is only needed till the next release of esp-idf-hal hit's

flyaruu, 2023-06-21:
I get the exact same error, except that I now see it using the patched version 0.41.2:
```
error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /Users/f.lyaruu/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/f235969/src/uart.rs:1091:25
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                         ^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /Users/f.lyaruu/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/f235969/src/uart.rs:1091:60
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                                                            ^^^^^^^^^^^^ not found in this scope

error[E0560]: struct `esp_idf_sys::uart_config_t` has no field named `source_clk`
   --> /Users/f.lyaruu/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/f235969/src/uart.rs:932:9
    |
932 |         source_clk: config.source_clock.into(),
    |         ^^^^^^^^^^ `esp_idf_sys::uart_config_t` does not have this field
    |
    = note: available fields are: `__bindgen_anon_1`
```

ivmarkov, 2023-06-21:
@flyaruu did you happen to build against `master`? You know, the most unstable branch of them all? Which is not the default suggestion?

ivmarkov, 2023-06-21:
@flyaruu OK; so I took your exact `cargo-generate` command line from above, generated the project and then fired `cargo build` inside.
Builds without a problem. Granted, this is with the newly released `esp-idf-hal`, however this new release **does not** take care of the `cannot find value UART_NUM_MAX in this scope` error, as it is very recent and seems to be caused by a regression in ESP-IDF `master` (they've - likely by accident - removed this `#define`).

So it still seems to me as if you are building against ESP-IDF `master`. Can you please:
* Tell me what you have on [this line](https://github.com/esp-rs/esp-idf-template/blob/master/cargo/.cargo/config.toml#L25) in your `.cargo/config.toml`?
* What you see when you `ll` `test54/.embuild/espressif/esp-idf`?

Thanks.

flyaruu, 2023-06-22:
Sorry for the delay, thanks for loolking into this.
My .cargo/config.toml:
```
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
# Note: these variables are not used when using pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "release/v5.1"
```

As for my .embuild folder, _I don't seem to have one_ (?). Om my older projects it's there, but for more recent ones there isn't. I'm pretty confused right now.

I do have an esp installation in ~/esp/esp-idf it's on branch release/v5.1

ivmarkov, 2023-06-22:
> As for my .embuild folder, _I don't seem to have one_ (?). Om my older projects it's there, but for more recent ones there isn't. I'm pretty confused right now.

Most likely you have one of these defined:
* `ESP_IDF_TOOLS_INSTALL_DIR`
* `IDF_PATH `
* `ESP_IDF_REPOSITORY`

(Check the README in the [esp-idf-sys repo](https://github.com/esp-rs/esp-idf-sys).)
Remove all of these, and then rebuild, so that the 5.1 ESP IDF release can be re-downloaded and placed in the default `.embuild` folder.

flyaruu, 2023-06-23:
That was it! I got tangled up in all the ENV variables. Thank you so much for your time! 🙏

closed 2023-06-23 by flyaruu