esp-rs/esp-idf-template #130: error: failed to run custom build command for `esp-idf-sys v0.33.1`

Geophyte, 2023-06-26:
## Bug description

I tried building unmodified esp-idf-template using cargo. I believe all prerequisites listed in README.md are installed correctly.

```
error: failed to run custom build command for `esp-idf-sys v0.33.1`

Caused by:
  process didn't exit successfully: `C:\_prog\vm_shared\esp\target\debug\build\esp-idf-sys-201d91811fe1e259\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  Submodule path 'components/asio/asio': checked out 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
  Submodule path 'components/bt/controller/lib_esp32': checked out '8ce6d27e0981c09e1c5a982976f182d0a37662f6'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out '08e289633f823191a6c526377bfd68f31fb392e0'
  Submodule path 'components/bt/host/nimble/nimble': checked out '4ed953346e55320ddcd09e0633fd2c4a8aef6b5b'
  Submodule path 'components/cbor/tinycbor': checked out '7c349dbb6b8d76db39383b226d3ebdf59b8ab37d'
  Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
  Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
  Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
  Submodule path 'components/coap/libcoap': checked out '3aa11612c143c9734d72022720f33e12506f7a2c'
  Submodule path 'components/coap/libcoap/ext/tinydtls': checked out '59055b8a935bc53bf69d002fc089ad4bd08851b2'
  Submodule path 'components/esp_phy/lib': checked out '7255b24596a11f67bada61ef169b7a7a00791cbd'
  Submodule path 'components/esp_wifi/lib': checked out '8ac58d5636c1218577583705a76d765a385b2f6d'
  Submodule path 'components/esptool_py/esptool': checked out 'ed3af0f60b04ea7d02f995abad9bfb0c8f9b1e1a'
  Submodule path 'components/expat/expat': checked out '454c6105bc2d0ea2521b8f8f7a5161c2abd8c386'
  Submodule path 'components/ieee802154/lib': checked out 'f7b5e8059a3bb6f321e79ac3bf2aa4d2a9b93326'
  Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
  Submodule path 'components/libsodium/libsodium': checked out '4f5e89fa84ce1d178a6765b8b46f2b6f91216677'
  Submodule path 'components/lwip/lwip': checked out '8290c3b8f2adaf82aa45ec992b87f16205f2689b'
  Submodule path 'components/mbedtls/mbedtls': checked out '15b55d406db3918bac88aaf5ef2c6e036d1e0f0e'
  Submodule path 'components/mqtt/esp-mqtt': checked out 'bb9c8af9d552b608dd3aabf9617bde757a538ebe'
  Submodule path 'components/nghttp/nghttp2': checked out '8f7b008b158e12de0e58247afd170f127dbb6456'
  Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
  Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'b967ca054f48a36f82d8fcdd32e54ec5144f2751'
  Submodule path 'components/openthread/lib': checked out '9a8d34d8f698cad2c9468468b473e26a3dda51b9'
  Submodule path 'components/openthread/openthread': checked out 'c36c0e77a2465355bcf13bd7dc718d8c9aa6ff64'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'f224ab2eeb648a818eb20687d7150a285442c907'
  Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
  Submodule path 'components/tinyusb/tinyusb': checked out 'c4badd394eda18199c0196ed0be1e2d635f0a5f6'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
  Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out '36d0642e66ff5b1c7a291873f24c498ca6ffedef'
  PATH=C:\_prog\vm_shared\esp\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  pip 23.1.2 from C:\msys64\mingw64\lib\python3.10\site-packages\pip (python 3.10)
  created virtual environment CPython3.10.11.final.0-64 in 4642ms
    creator CPython3Windows(dest=C:\_prog\vm_shared\esp\.embuild\espressif\python_env\idf4.4_py3.10_env, clear=False, no_vcs_ignore=False, global=False)
    seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
      added seed packages: pip==23.1.2, setuptools==67.8.0, wheel==0.40.0
    activators BashActivator,BatchActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
  Creating a new Python environment in C:\_prog\vm_shared\esp\.embuild\espressif\python_env\idf4.4_py3.10_env
  Installing Python packages from C:\_prog\vm_shared\esp\.embuild\espressif\esp-idf\release-v4.4\requirements.txt
```

## Environment

- OS: Windows 10
- espup 0.4.1
- Python: 3.10.11, 3.11.4

ivmarkov, 2023-06-26:
Hmm, the output does not contain the `stderr` stream contents? In any case, please try to shorten the path of the project as much as possible. As in moving `esp` directly to `C:\` And try again. Please report if successful or not - there are issues on Windows with path length limit that are difficult to overcome.

If that turns out to be the culprit, I strongly recommend moving to WSL2 (or somehow living with short project name in the drive root).

Geophyte, 2023-06-26:
I got the same error so I guess that's not it, although the stdout is shorter

```
error: failed to run custom build command for `esp-idf-sys v0.33.1`

Caused by:
  process didn't exit successfully: `C:\esp\target\debug\build\esp-idf-sys-201d91811fe1e259\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  PATH=C:\esp\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  pip 23.1.2 from C:\msys64\mingw64\lib\python3.10\site-packages\pip (python 3.10)
  created virtual environment CPython3.10.11.final.0-64 in 1649ms
    creator CPython3Windows(dest=C:\esp\.embuild\espressif\python_env\idf4.4_py3.10_env, clear=False, no_vcs_ignore=False, global=False)
    seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
      added seed packages: pip==23.1.2, setuptools==67.8.0, wheel==0.40.0
    activators BashActivator,BatchActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
  Creating a new Python environment in C:\esp\.embuild\espressif\python_env\idf4.4_py3.10_env
  Installing Python packages from C:\esp\.embuild\espressif\esp-idf\release-v4.4\requirements.txt
warning: build failed, waiting for other jobs to finish...
```

ivmarkov, 2023-06-27:
The output from the failed build is too short. Are you sure this is all of it?

Geophyte, 2023-06-27:
Well, I cleaned it and built it again. If I don't need to enable something to see more of the output then I guess that's it. I also tried to manually install `requirements.txt`, but it still crashes with the same error when I continue building.

```
PS C:\esp> cargo build
   Compiling compiler_builtins v0.1.91
   Compiling core v0.0.0 (C:\Users\idab\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core)
   Compiling proc-macro2 v1.0.63
   Compiling unicode-ident v1.0.9
   Compiling quote v1.0.28
   Compiling libc v0.2.140
   Compiling cc v1.0.77
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (C:\Users\idab\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std)
   Compiling serde v1.0.164
   Compiling winapi v0.3.9
   Compiling windows_x86_64_msvc v0.48.0
   Compiling fnv v1.0.7
   Compiling cfg-if v1.0.0
   Compiling autocfg v1.1.0
   Compiling libc v0.2.147
   Compiling windows-targets v0.48.0
   Compiling once_cell v1.18.0
   Compiling syn v1.0.109
   Compiling windows-sys v0.48.0
   Compiling glob v0.3.1
   Compiling windows_x86_64_msvc v0.42.2
   Compiling regex-syntax v0.7.2
   Compiling semver v1.0.17
   Compiling aho-corasick v1.0.2
   Compiling syn v2.0.22
   Compiling lazy_static v1.4.0
   Compiling clang-sys v1.6.1
   Compiling rustversion v1.0.12
   Compiling stable_deref_trait v1.2.0
   Compiling log v0.4.19
   Compiling as-slice v0.2.1
   Compiling aho-corasick v0.7.20
   Compiling bstr v1.5.0
   Compiling serde_json v1.0.99
   Compiling thiserror v1.0.40
   Compiling minimal-lexical v0.2.1
   Compiling windows-targets v0.42.2
   Compiling aligned v0.4.1
   Compiling nom v7.1.3
   Compiling tempfile v3.6.0
   Compiling thread_local v1.1.7
   Compiling unwind v0.0.0 (C:\Users\idab\.rustup\toolchains\esp\lib\rustlib\src\rust\library\unwind)
   Compiling regex v1.8.4
   Compiling winapi-util v0.1.5
   Compiling libloading v0.7.4
   Compiling same-file v1.0.6
   Compiling cvt v0.1.2
   Compiling itoa v1.0.6
   Compiling walkdir v2.3.3
   Compiling bitflags v1.3.2
   Compiling anyhow v1.0.71
   Compiling heck v0.4.1
   Compiling either v1.8.1
   Compiling bindgen v0.63.0
   Compiling ryu v1.0.13
   Compiling globset v0.4.10
   Compiling fs_at v0.1.8
   Compiling normpath v1.1.1
   Compiling cexpr v0.6.0
   Compiling which v4.4.0
   Compiling ignore v0.4.20
   Compiling windows-sys v0.45.0
   Compiling lazycell v1.3.0
   Compiling shlex v1.1.0
   Compiling rustc-hash v1.1.0
[... 98 lines cut ...]
  Submodule path 'components/libsodium/libsodium': checked out '4f5e89fa84ce1d178a6765b8b46f2b6f91216677'
  Submodule path 'components/lwip/lwip': checked out '8290c3b8f2adaf82aa45ec992b87f16205f2689b'
  Submodule path 'components/mbedtls/mbedtls': checked out '15b55d406db3918bac88aaf5ef2c6e036d1e0f0e'
  Submodule path 'components/mqtt/esp-mqtt': checked out 'bb9c8af9d552b608dd3aabf9617bde757a538ebe'
  Submodule path 'components/nghttp/nghttp2': checked out '8f7b008b158e12de0e58247afd170f127dbb6456'
  Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
  Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'b967ca054f48a36f82d8fcdd32e54ec5144f2751'
  Submodule path 'components/openthread/lib': checked out '9a8d34d8f698cad2c9468468b473e26a3dda51b9'
  Submodule path 'components/openthread/openthread': checked out 'c36c0e77a2465355bcf13bd7dc718d8c9aa6ff64'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'f224ab2eeb648a818eb20687d7150a285442c907'
  Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
  Submodule path 'components/tinyusb/tinyusb': checked out 'c4badd394eda18199c0196ed0be1e2d635f0a5f6'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
  Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out '36d0642e66ff5b1c7a291873f24c498ca6ffedef'
  PATH=C:\esp\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  pip 23.1.2 from C:\msys64\mingw64\lib\python3.10\site-packages\pip (python 3.10)
  created virtual environment CPython3.10.11.final.0-64 in 4456ms
    creator CPython3Windows(dest=C:\esp\.embuild\espressif\python_env\idf4.4_py3.10_env, clear=False, no_vcs_ignore=False, global=False)
    seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
      added seed packages: pip==23.1.2, setuptools==67.8.0, wheel==0.40.0
    activators BashActivator,BatchActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
  Creating a new Python environment in C:\esp\.embuild\espressif\python_env\idf4.4_py3.10_env
  Installing Python packages from C:\esp\.embuild\espressif\esp-idf\release-v4.4\requirements.txt
```

ivmarkov, 2023-06-27:
OK, one last try:
* Please do `export RUST_BACKTRACE=1` prior to running `cargo build` and paste the build output again
* Are you sure that you have python3 and git installed? Both should be installed and on your `%PATH%`

Geophyte, 2023-06-27:
I have Python 3.10.11 (default) and 3.11.4. Apparently, Python 3.10.11 is not very descriptive and setting `RUST_BACKTRACE=1` or `RUST_BACKTRACE=full` didn't really change anything. After I forced it to use Python 3.11.4 and setting `RUST_BACKTRACE=1` I got the following output:
```
error: failed to run custom build command for `esp-idf-sys v0.33.1`

Caused by:
  process didn't exit successfully: `C:\esp\target\debug\build\esp-idf-sys-201d91811fe1e259\build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  IDF_PYTHON_ENV_PATH=C:\esp\.embuild\espressif\python_env\idf4.4_py3.11_env
  PATH=C:\esp\.embuild\espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:\esp\.embuild\espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\esp\.embuild\espressif\tools\cmake\3.23.1\bin;C:\esp\.embuild\espressif\tools\ninja\1.10.2\;C:\esp\.embuild\espressif\python_env\idf4.4_py3.11_env\Scripts;C:\esp\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  Current system platform: win64
  Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=C:\esp\.embuild\espressif\python_env\idf4.4_py3.11_env
  PATH=C:\esp\.embuild\espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:\esp\.embuild\espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\esp\.embuild\espressif\tools\cmake\3.23.1\bin;C:\esp\.embuild\espressif\tools\ninja\1.10.2\;C:\esp\.embuild\espressif\python_env\idf4.4_py3.11_env\Scripts;C:\esp\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  cargo:rerun-if-changed=C:\esp\sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "C:\\esp\\target\\xtensa-esp32-espidf\\debug\\build\\esp-idf-sys-b5ae659f1c10d311\\out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=C:\\esp\\.embuild\\espressif\\esp-idf\\release-v4.4\\tools\\cmake\\toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=C:/esp/.embuild/espressif/python_env/idf4.4_py3.11_env/Scripts/python.exe" "-DCMAKE_INSTALL_PREFIX=C:\\esp\\target\\xtensa-esp32-espidf\\debug\\build\\esp-idf-sys-b5ae659f1c10d311\\out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from C:\esp\.embuild\espressif\esp-idf\release-v4.4\requirements.txt are satisfied.
  -- Project sdkconfig file C:/esp/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b5ae659f1c10d311/out/sdkconfig
  Loading defaults file C:/esp/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b5ae659f1c10d311/out/gen-sdkconfig.defaults...
  Loading defaults file C:/esp/sdkconfig.defaults...
  -- App "libespidf" version: 1
  -- Adding linker script C:/esp/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b5ae659f1c10d311/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
  -- Adding linker script C:/esp/.embuild/espressif/esp-idf/release-v4.4/components/soc/esp32/ld/esp32.peripherals.ld
  -- Configuring done
  -- Generating done
  -- Build files have been written to: C:/esp/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b5ae659f1c10d311/out/build
  running: "cmake" "--build" "." "--config" "MinSizeRel" "--parallel" "12"
  [1/7] Generating ld/sections.ld
  [2/7] Linking C executable libespidf.elf
  [3/7] Generating binary image from built executable
  esptool.py v3.3.4-dev
  Creating esp32 image...
  Merged 2 ELF sections
  Successfully created esp32 image.
[... cut ...]  usb, tinyusb, ulp, wifi_provisioning
  thread 'main' panicked at 'Unable to find libclang: "the `libclang` shared library at C:\\Users\\idab\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-16.0.0-20230516\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"', C:\Users\idab\.cargo\registry\src\index.crates.io-6f17d22bba15001f\bindgen-0.63.0\./lib.rs:2338:31
  stack backtrace:
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

ivmarkov, 2023-06-27:
Let me start with the TL;DR: The problem is that the `clang` library - for some reason - cannot be loaded on your system:
```
thread 'main' panicked at 'Unable to find libclang: "the `libclang` shared library at C:\\Users\\idab\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-16.0.0-20230516\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"', C:\Users\idab\.cargo\registry\src\index.crates.io-6f17d22bba15001f\bindgen-0.63.0\./lib.rs:2338:31
```

That's weird. It might be that you need some additional dependency that is not on your system, hence the lib fails to load. Another reason might be that you are running a 32 bit system while the lib is 64 bit, or the other way around. Guessing here.

You might try to down-grade the ESP toolchain installed by `espup`. @SergioGasquez  - any other ideas?

As for the other things you mentioned: 
* `RUST_BACKTRACE` is for cargo/Rust; not for Python, so that I can check where exactly it is failing, but it is now clear.
* You now **do** have an error log as well. Where it says `--- stderr`. The display of this is **not** affected by `RUST_BACKTRACE` at all. So are you really sure you did not skip it the previous time?

Geophyte, 2023-06-27:
Okay, I managed to build it. Turns out I needed to run `%USERPROFILE%\export-esp.ps1` before `cargo build`. I guess The Rust on ESP Book is a bit misleading with this:
> There is no need to execute the file for Windows users. It is only created to show the modified environment variables.

As for the error log, I assure you that when I was using Python 3.10.11 there was no `--- stderr` to be found. Also, it seems to have been crashing earlier in the building process than Python 3.11.4.

SergioGasquez, 2023-06-27:
> > There is no need to execute the file for Windows users. It is only created to show the modified environment variables.

Which `espup` version did you use to install the environment?

Geophyte, 2023-06-27:
I used `espup 0.4.1`

SergioGasquez, 2023-06-27:
> I used `espup 0.4.1`

`espup v0.4.1` should be able to inject the environment variables to your system without having to source the export file. Would you mind trying to reproduce the issue again:
```
espup uninstall
espup install 
```
Then check the environment variables

Geophyte, 2023-06-27:
After reinstalling `espup` and making a clean build I got a different error message. The previous one was resolved by setting the `%PATH%` variable (I forgot about it so I edited my previous comment to reflect that). This one gets resolved by running `%USERPROFILE%\export-esp.ps1` before `cargo run`
```
thread 'main' panicked at 'Unable to find libclang: "couldn't find any valid shared libraries h
matching: ['clang.dll', 'libclang.dll'], set the `LIBCLANG_PATH` environment variable to a path t
where one of these files can be found (invalid: [])"',
```

SergioGasquez, 2023-06-27:
> After reinstalling `espup` and making a clean build I got a different error message. The previous one was resolved by setting the `%PATH%` variable (I forgot about it so I edited my previous comment [...]

Have you tried restarting the terminal between installation and project compilation?

SergioGasquez, 2023-06-28:
> Yes, I have restarted the terminal after installation. The strange thing about it is that after echoing the environmental variables are set correctly according to `export-esp.ps1`, but for some reason I still have to execute it manually from within the `Powershell` in VS Code. I also tried running it from `cmd`, but I encountered the same problem.

So, if I understood correctly:
- You use a terminal for installation
  - If you echo any variable of the `export-esp.ps1` on this terminal, its properly printed
- When using VsCode terminal
  - Those environment variables are not set, and you need to source the export file
  
 Is my understanding correct? Which VsCode terminal are you using?

SergioGasquez, 2023-06-29:
> So before I run/build the template and before I run export-esp.ps1 on the terminal I tried to echo environmental variables which should have been set during espup installation. The variables echo correctly, but building the template still fails because LIBCLANG_PATH is apparently not set. To build the project I need to run export-esp.ps1 on the terminal I'm currently using for it to build without crashing.

Just to confirm, you are echoing the variables in the same terminal as your are running `cargo build`, right?

> I checked out environmental variables through the GUI app provided by Windows and by the looks of it the variables are set for the current user(me), but aren't set for all users. I think in order for the environmental variables to be visible during building they need to be set globally for all users.

Can you check what happens when you source the export file? I'd guess that they are only set for current user.

I will try to reproduce the error

SergioGasquez, 2023-06-29:
> I will try to reproduce the error

I did some investigations, after running `espup install` variables are shown under `Control Panel > Edit the system environment variables > Environment variables` for your user. But if I try to echo the variable, e.g ` $env:libclang_path` its not printed. I tried reloading the terminal with a few methods (`. $profile` and `Invoke-Command { & "pwsh.exe"       } -NoNewScope`) but none of them worked, then I tried restarting the computer, and that did work, after rebooting I was able to build a project without having to source the export file

ivmarkov, 2023-06-29:
@SergioGasquez I'm not sure whether the above visual location in the latest Windows incarnation you are mentioning is for "system" or "user" environment variables, as somehow the wording mentions both. In any case:
* We should absolutely not register / change system environment variables, because `espup` installs the toolchain in the user's folder; we should be altering the *user* environment variables
*  Up until Windows 10 at least, changing user environment variables only required restarting the user command shell, and not the whole PC; not really sure how that relates to VSCode and its "terminal"
* Altering system environment variables (which - again - we should not be doing) always required restart of the PC 

This:
> I checked out environmental variables through the GUI app provided by Windows and by the looks of it the variables are set for the current user(me), but aren't set for all users. I think in order [...]

... is not correct. And we should - again - not do any system-wide changes.

ivmarkov, 2023-06-29:
@SergioGasquez Some more suggestions:
- Either we are OK with having the user's PATH (and potentially other ENV vars) automatically changed for the user on **both** Windows and *nix, or we are not ok with doing that on either of the two (i.e. the user should always use the `export` script); or - we might ask during installation
- Note that `rustup` itself is not shying away from modifying the user's `PATH` upon installation, in that it is appending `. $HOME/.cargo/env` to your `.profile`. Perhaps we should name our `export.sh` `env` and we should do exactly what `rustup` does on *nix?

ivmarkov, 2023-06-29:
... and we should shuffle our `env` inside the `.rustup` directory, where the toolchain is installed as well.

SergioGasquez, 2023-08-02:
> - Either we are OK with having the user's PATH (and potentially other ENV vars) automatically changed for the user on both Windows and *nix, or we are not ok with doing that on either of the two [...]

I don't have a hard opinion on this regard, but this was discussed in the past, and we decided that on Windows we would modify the `PATH` while in Unix we will require the file sourcing.

Ideally, once we can use LLD as linker, this problem will disappear (we would only require setting `LIBCLANG_PATH` if I'm not mistaken). Sadly [LLD might take longer than expected ](https://github.com/esp-rs/esp-hal/pull/357#issuecomment-1654647690)

> On what terminal did you try to execute it? In cmd you use echo %libclang_path% and in PowerShell echo $env:libclang_path should work

I was using powershell.

> Well, today (after restarting my PC) when I tried building the template I was greeted with this error:
thread 'main' panicked at 'Unable to find libclang: "the `libclang` shared library at C:\\Users\\idab\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-16.0.0-20230516\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"', C:\Users\idab\.cargo\registry\src\index.crates.io-6f17d22bba15001f\bindgen-0.63.0\./lib.rs:2338:31
which was previously resolved by adding clang to %PATH%. The workaround didn't change and all I needed to do to build it was execute export-esp.ps1 in the terminal.

Not sure why your environment changes this way. I was able to build projects in Windows with no issue

ivmarkov, 2023-08-02:
> I don't have a hard opinion on this regard, but this was discussed in the past, and we decided that on Windows we would modify the `PATH` while in Unix we will require the file sourcing.

I don't remember the discussion, but regardless, I keep noticing folks routinely forget to call the script. And was showing evidence, that other tooling on Unix bravely touches `.bashrc` and friends.

Also - and hopefully - even if the path is modified by sourcing the `export` script from `.bashrc`, I hope that would not preclude folks from doing `rustup default stable` and `rustup default nightly`, so the change does not have a high risk.

ivmarkov, 2023-10-19:
Closing in favor of https://github.com/esp-rs/esp-idf-hal/issues/319