esp-rs/esp-idf-template #139: missing field `home_chan_dwell_time` in initializer of `esp_idf_sys::wifi_scan_config_t`

faern, 2023-07-22:
## Bug description

Upon generating a project from this template and configure it to use ESP-IDF 5.1 (latest stable) and include svc-support, I'm greeted with the following error upon `cargo build`:

```
   Compiling esp-idf-svc v0.46.0
error[E0063]: missing field `home_chan_dwell_time` in initializer of `esp_idf_sys::wifi_scan_config_t`
  --> /home/faern/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.46.0/src/wifi.rs:93:13
   |
93 |             Self {
   |             ^^^^ missing `home_chan_dwell_time`

For more information about this error, try `rustc --explain E0063`.
error: could not compile `esp-idf-svc` (lib) due to previous error
```

This field seem to have been added to ESP-IDF in `master` ([link to docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv418wifi_scan_config_t)). But it's not there in the docs for 5.1 stable ([link to docs](https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32/api-reference/network/esp_wifi.html#_CPPv418wifi_scan_config_t)). This confuses me, since I have configured the template to use 5.1 stable. Is the bug that it pulls in ESP-IDF master when it should not?

- Would you like to work on a fix? [y/n]: Maybe. Right now I just wanted this reported to see if someone with more insight into the crate easily figures out the issue.

## Expected behavior

I expect the template to build and work for all the versions of ESP-IDF it allows me to select.

## Environment

- OS: Fedora 37
- How did you install the environment: Since I only target the C3 chip, I did not do anything special except getting latest nightly rustc from rustup.

faern, 2023-07-22:
The CI for this repository failed with the same error two days ago: https://github.com/esp-rs/esp-idf-template/actions/runs/5629581355/job/15254691621

timleslie, 2023-07-23:
First time working in this ecosystem, so I can't provide insight into how/why this issue turned up, but I was able to find an upstream commit that addresses the issue: https://github.com/esp-rs/esp-idf-svc/commit/42e60e8c3fcca670632b9653c790fd8efe475aac

Updating my `Cargo.toml` with the following dependency allows me to compile.

```
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc.git", branch = "master", optional = true, default-features = false }
```

Good luck!

faern, 2023-07-23:
Awesome. I checked their issue tracker and PR history but did not find anything. I also searched their git history. But they managed to patch it in a way where this missing field name did not show up anywhere 😅

This still does not answer why the issue is present when selecting 5.1 stable however?

faern, 2023-07-23:
This is being discussed more in https://github.com/espressif/esp-idf/issues/11921. I'm going to close this as it has been patched and this template repo is not at fault. Just hoping for a patch release of `esp-idf-svc` soon :)

closed 2023-07-23 by faern

SergioGasquez, 2023-07-24:
I added a patch in the Cargo.toml (https://github.com/esp-rs/esp-idf-template/pull/142) until the new release of esp-idf-svc is released