esp-rs/esp-idf-template #148: use of undeclared crate or module `esp_idf_svc`

javoerro, 2023-08-09:
Compile fails with the error: use of undeclared crate or module `esp_idf_svc`

Steps to reproduce the behavior:

1. followed prerequisites of https://github.com/esp-rs/esp-idf-template
2. generated project with "cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cmake"
3. open created folder with vscode
4. use esp-idf vscode extension to change target to esp32s3
5. use esp-idf vscode extension to build (hit the build button)

Windows 10, vscode, espup 0.4.1

ivmarkov, 2023-08-10:
Can you provide some extra details:
- Excerpt from the build log where the error appears
- does it work with `./idf.py build` and only fail with the vscode extension?
- what template parameters you used when generating the project

javoerro, 2023-08-10:
@ivmarkov your request was added in first post, thank you!

ivmarkov, 2023-08-23:
Sorry for the late reply. I'll try to get to it in the next days.weeks. Seems to be a bug in the generated project in that the `esp-idf-svc` dependency is not explicitly listed in `Cargo.toml` in some cases.

In the meantime, you can add it manually.

I'm not in a rush to fix it, as anyway the upcoming versionf os `esp-idf-svc` / `esp-idf-hal` / `esp-idf-sys` there is a change that would allow users to only depend on `esp-idf-svc` as it will re-export the other crates.
The templates recently grew way too complex, and are now difficult to maintain - I would like to simplify the options the user can choose from. For one "all HAL", "no HAL" etc. need to go. They only serve very niche use cases, and the purpose of these templates is to bootstrap the user. She can always adjust the `Cargo.toml`manually after that.

Maldus512, 2023-10-19:
I'm encountering the same issue on Void Linux with the latest master; the only difference is `espup` version 0.7.0

ivmarkov, 2023-10-19:
Should no longer happen with the PR which just got merged. Please reopen if you are still experiencing the issue.