esp-rs/esp-idf-template #149: v4 build error while linking

kons-9, 2023-08-13:
## Bug description
_error occurs in link process.
here is a part of output of the error. this error doesn't occur in stable v5.

```zsh
= note: [ldproxy] Running ldproxy
   6   │           Error: Linker /home/wslmtl/Documents/rust/std-template/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-e
       │ lf-gcc failed: exit status: 1
   7   │           STDERR OUTPUT:
   8   │           /home/wslmtl/Documents/rust/std-template/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/
       │ 8.4.0/../../../../riscv32-esp-elf/bin/ld: error: esp-idf/esp_ringbuf/libesp_ringbuf.a(ringbuf.c.obj): Mis-matched ISA version for 'i' extension. 2.0 vs 2.1
   9   │           /home/wslmtl/Documents/rust/std-template/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/
       │ 8.4.0/../../../../riscv32-esp-elf/bin/ld: failed to merge target specific data of file esp-idf/esp_ringbuf/libesp_ringbuf.a(ringbuf.c.obj)
  10   │           /home/wslmtl/Documents/rust/std-template/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/
       │ 8.4.0/../../../../riscv32-esp-elf/bin/ld: -march=: ISA string must begin with rv32 or rv64
```

- Would you like to work on a fix? [y/n]
n

## To Reproduce
just build by default mode

## Expected behavior
work correctly

## Screenshots

## Environment
I tried macOS and WSL2 using cargo-generate

## Additional context

Vollbrecht, 2023-08-13:
you are using the latest nightly compiler? can you try using one that is not newer than 8th of August. You can install a specific version using `rustup install nightly-2023-08-07` . Than change the rust-toolchain.toml to that version or use the `+nightly-2023-08-07` syntax added to your `cargo b` command.

kons-9, 2023-08-14:
Yes, my version was latest nightly(2023-8-12)
And using nightly-2023-08-07, it works.

LarryMerino, 2023-08-17:
I try updating rust version and now I'm using `1.73.0-nightly (7c3904d6c 2023-08-14)` but the same problem 
```rust
   Compiling embedded-svc v0.25.3
error: linking with `ldproxy` failed: exit status: 1
```
If I use custom configuration and select ESP-IDF version v5.1 it works but v4.4 have this error

ivmarkov, 2023-08-23:
Should be fixed now.

closed 2023-08-23 by ivmarkov