esp-rs/esp-idf-template #152: Template for esp32h2 fails to run and enters an infinite loop

RaynalHugo, 2023-09-08:
Hey there 👋 
First of all, I'd like to thank you all for the work you provide here. I am relatively new to embedded rust. The various books, tools and examples you provide made it easier for me to start in this field! ❤️ 

## Bug description

Template using std doesn't work on esp32-h2. Whereas the no-std template work

- Would you like to work on a fix? No, I am far from being able to.

## To Reproduce

Steps to reproduce the behavior:

1. Have a [ESP32-H2-DevKitM-1](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32h2/esp32-h2-devkitm-1/user_guide.html#esp32-h2-devkitm-1) 
2. Plug it to your computer on the USB type C to UART port
3. On a terminal, run `cargo generate esp-rs/esp-idf-template cargo`
4. provide a project name
5. choose `esp32h2` as a MCU to target
6. Do not use the advanced options (answer `false`)
7. `cd` to the newly created folder
8. run `cargo run`
9. select the device port ( in my case: `/dev/tty.usbserial-10 - CP2102N USB to UART Bridge Controller`, or `/dev/cu.usbserial-10 - CP2102N USB to UART Bridge Controller`, the result seems to be the same)

The process now enters a loop, this is the first output 
```
    Finished dev [optimized + debuginfo] target(s) in 0.09s
     Running `espflash flash --monitor target/riscv32imac-esp-espidf/debug/repro`
[2023-09-08T10:37:29Z INFO ] Detected 6 serial ports
[2023-09-08T10:37:29Z INFO ] Ports which match a known common dev board are highlighted
[2023-09-08T10:37:29Z INFO ] Please select a port
[2023-09-08T10:37:32Z INFO ] Serial port: '/dev/tty.usbserial-10'
[2023-09-08T10:37:32Z INFO ] Connecting...
[2023-09-08T10:37:33Z INFO ] Using flash stub
Chip type:         esp32h2 (revision v0.1)
Crystal frequency: 32MHz
Flash size:        4MB
Features:          BLE
MAC address:       48:31:b7:c0:1d:de
App/part. size:    435,424/4,128,768 bytes, 10.55%
[00:00:01] [========================================]      13/13      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:21] [========================================]     215/215     0x10000                                       [2023-09-08T10:37:57Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit
```

Then, this part gets repeated every 7-8s, indefinitely.

```
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4083cfd0,len:0xc10
0x4083cfd0 - _heap_start
    at ??:??
load:0x4083efd0,len:0x2d2c
0x4083efd0 - _heap_start
    at ??:??
load:0x408460e0,len:0x1790
0x408460e0 - _heap_start
    at ??:??
SHA-256 comparison failed:
Calculated: b38e5fddcd28d8a0e99239e4848bd0641aa8f152d07cfb6eada3c2e7a0255f35
Expected: f787a66e6b4ad8e710dc9b693f706abf29015428c909f15614cbf44be2ce0b3c
Attempting to boot anyway...
entry 0x4083cfd0
0x4083cfd0 - _heap_start
    at ??:??
I (42) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:03:28
I (44) boot: chip revision: v0.1
I (47) boot.esp32h2: SPI Speed      : 32MHz
I (52) boot.esp32h2: SPI Mode       : DIO
I (57) boot.esp32h2: SPI Flash Size : 4MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 003f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=43ce0h (277728) map
I (190) esp_image: segment 1: paddr=00053d08 vaddr=40800000 size=04310h ( 17168) load
I (196) esp_image: segment 2: paddr=00058020 vaddr=42048020 size=1d230h (119344) map
I (229) esp_image: segment 3: paddr=00075258 vaddr=40804310 size=042a4h ( 17060) load
I (235) esp_image: segment 4: paddr=00079504 vaddr=408085c0 size=00fach (  4012) load
I (239) boot: Loaded app from partition at offset 0x10000
I (241) boot: Disabling RNG early entropy source...
```

## Expected behavior

When I use the no std version, using `cargo generate esp-rs/esp-template`, this is the log I receive:
```
    Finished dev [unoptimized + debuginfo] target(s) in 2.13s
     Running `espflash flash --monitor target/riscv32imac-unknown-none-elf/debug/esp_32_test_no_std`
[2023-09-08T10:15:17Z INFO ] Detected 6 serial ports
[2023-09-08T10:15:17Z INFO ] Ports which match a known common dev board are highlighted
[2023-09-08T10:15:17Z INFO ] Please select a port
[2023-09-08T10:15:22Z INFO ] Serial port: '/dev/cu.usbserial-10'
[2023-09-08T10:15:22Z INFO ] Connecting...
[2023-09-08T10:15:22Z INFO ] Using flash stub
Chip type:         esp32h2 (revision v0.1)
Crystal frequency: 32MHz
Flash size:        4MB
Features:          BLE
MAC address:       48:31:b7:c0:1d:de
App/part. size:    232,864/4,128,768 bytes, 5.64%
[00:00:01] [========================================]      13/13      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:07] [========================================]      68/68      0x10000                                       [2023-09-08T10:15:32Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4083cfd0,len:0xc10
0x4083cfd0 - __global_pointer$
    at ??:??
load:0x4083efd0,len:0x2d2c
0x4083efd0 - __global_pointer$
    at ??:??
load:0x408460e0,len:0x1790
0x408460e0 - __global_pointer$
    at ??:??
SHA-256 comparison failed:
Calculated: b38e5fddcd28d8a0e99239e4848bd0641aa8f152d07cfb6eada3c2e7a0255f35
Expected: f787a66e6b4ad8e710dc9b693f706abf29015428c909f15614cbf44be2ce0b3c
Attempting to boot anyway...
entry 0x4083cfd0
0x4083cfd0 - __global_pointer$
    at ??:??
I (42) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:03:28
I (44) boot: chip revision: v0.1
I (47) boot.esp32h2: SPI Speed      : 32MHz
I (52) boot.esp32h2: SPI Mode       : DIO
I (57) boot.esp32h2: SPI Flash Size : 4MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  2 factory          factory app      00 00 00010000 003f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=40804000 size=01048h (  4168) load
I (114) esp_image: segment 1: paddr=00011070 vaddr=00000000 size=0ef88h ( 61320)
I (138) esp_image: segment 2: paddr=00020000 vaddr=42000000 size=23328h (144168) map
I (178) esp_image: segment 3: paddr=00043330 vaddr=00000000 size=00018h (    24)
I (179) esp_image: segment 4: paddr=00043350 vaddr=42023350 size=05a20h ( 23072) map
I (190) boot: Loaded app from partition at offset 0x10000
I (190) boot: Disabling RNG early entropy source...
Hello world!
```
(And , it doesn't loop)

## Environment

I installed all dependencies following the documentation.

- OS: Macos Ventura 13.3 on a Apple M1 pro
- cargo-generate: v0.18.4
- ldproxy: (don't know how to log the version)
- espup: v0.5.0
- espflash: v2.0.1
- cargo-espflash 2.0.1 
- rustc:  1.74.0-nightly (1e746d774 2023-09-07) (overwritten by `rust-toolchain.toml`)

## Additional context

- I bought multiple (identical) boards, I tried with 2 of them and got the same result
- I tried plugging the board in either the USB and UART plugs, got the same result.

Thank you for looking into this! Let me know if there is anything I can do on my side to help out.

ivmarkov, 2023-09-13:
I don't have an `esp32h2` board yet, so I can only guess what is going on here.
There's a slim chance that i[t could be caused by us using the wrong Rust target](https://github.com/esp-rs/esp-idf-sys/blob/master/build/native/cargo_driver/chip.rs#L71) (but the chance is really slim!).

What you can do, is to enable the `riscv32imac-esp-espidf` target in your `.cargo/config/toml` file (pay attention - the currently enabled target there is most likely `riscv32imc-esp-espidf`, without the "a") and then try again.

If it works for you with this change (but the chance is really slim), I'll update `esp-idf-sys` to default to the imAc target for the h2.

SergioGasquez, 2023-09-13:
> I don't have an `esp32h2` board yet, so I can only guess what is going on here. There's a slim chance that i[t could be caused by us using the wrong Rust [...]

The template already uses the IMAC target: https://github.com/esp-rs/esp-idf-template/blob/master/cargo/pre-script.rhai#L28

ivmarkov, 2023-09-13:
I'm out of ideas then. In any case we need to update `esp-idf-sys` as it is not deriving the IMAC target.

RaynalHugo, 2023-09-13:
I can confirm that the template I used had the `riscv32imac-esp-espidf` target.

> I don't have an `esp32h2` board yet, so I can only guess what is going on here.

@ivmarkov let me know, if there is anything you want me to try on my end.
Thank you for having a look at this anyway!

Syphixs, 2023-10-08:
@RaynalHugo Did you manage to get it to work? 
I got my h2-devkit yesterday and sadly have the exact same problem.

RaynalHugo, 2023-10-09:
@Syphixs 
Unfortunately, I haven't. I paused this project for a moment. But please let me know if you find any workaround.

SergioGasquez, 2023-10-19:
@ivmarkov just revamped the template project and bumped dependencies, please try generating a new project and trying it again, I did some testing on C6 which also had a similar issue and it was now wokring fine

ivmarkov, 2023-10-19:
> @ivmarkov just revamped the template project and bumped dependencies, please try generating a new project and trying it again, I did some testing on C6 which also had a similar issue and it was now working fine

I confirm that re-generating your project might fix it.
The previous version of the template did not set the `MCU=esp32h2` variable, so it is likely that you got something compiled for a different MCU altogether.

SergioGasquez, 2023-10-19:
Just tried with an ESP32-H2 and I can confirm that the problem is now fixed!