esp-rs/esp-idf-template #158: Failed to compile due to esp-idf-hal

adrien3d, 2023-10-16:
## Bug description
When trying to compile a freshly generated project for `esp32`, I have the following error:

```
   Compiling fline v0.1.0 (/Users/adrien/Dev/rust/cosail/fline)
error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.2/src/uart.rs:1091:25
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                         ^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.2/src/uart.rs:1091:60
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                                                            ^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find function `xTaskGetIdleTaskHandleForCPU` in this scope
     --> /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.2/src/task.rs:152:9
      |
152   |         xTaskGetIdleTaskHandleForCPU(core as u32)
      |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ help: a function with a similar name exists: `xTaskGetIdleTaskHandleForCore`
      |
     ::: /Users/adrien/Dev/rust/cosail/fline/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-26796a8b83637180/out/bindings.rs:17086:5
      |
17086 |     pub fn xTaskGetIdleTaskHandleForCore(xCoreID: BaseType_t) -> TaskHandle_t;
      |     ------------------------------------------------------------------------- similarly named function `xTaskGetIdleTaskHandleForCore` defined here

error[E0560]: struct `esp_idf_sys::uart_config_t` has no field named `source_clk`
   --> /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.2/src/uart.rs:932:9
    |
932 |         source_clk: config.source_clock.into(),
    |         ^^^^^^^^^^ `esp_idf_sys::uart_config_t` does not have this field
    |
    = note: available fields are: `__bindgen_anon_1`

Some errors have detailed explanations: E0425, E0560.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `esp-idf-hal` (lib) due to 4 previous errors
warning: build failed, waiting for other jobs to finish...
```

ivmarkov, 2023-10-16:
Esp idf version?

adrien3d, 2023-10-17:
Sorry for the inconvenience, but might be that my clone esp-idf was on master while the version selected while generating the project might have bin `v5.1`.
Could you maybe precise on the `Readme.md` what is the default version, I think it is master because it is in bold, but you can may be add (default) after.
Thank you.

closed 2023-10-17 by adrien3d