esp-rs/esp-idf-template #165: build issue

yashwanthsinghm, 2023-10-27:
To generate the issue :
Note : All the prerequisites were installed before setting up esp-isf -template.

``` ~/projects/esp32$ cargo generate esp-rs/esp-idf-template cargo
⚠️   Favorite `esp-rs/esp-idf-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-idf-template.git
🤷   Project Name: esp2
🔧   Destination: /home/iyg1kor/projects/esp32/esp2 ...
🔧   project-name: esp2 ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32
✔ 🤷   Configure advanced template options? · true
✔ 🤷   ESP-IDF version (master = UNSTABLE) · v5.1
✔ 🤷   Enable STD support? · true
✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · false
✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · false
✔ 🤷   Add CI files for GitHub Action? · false
🔧   Moving generated files into: `/home/iyg1kor/projects/esp32/esp2`...
🔧   Initializing a fresh Git repository
✨   Done! New project created /home/iyg1kor/projects/esp32/esp2
```

Building the project : cargo build

``` 
~/projects/esp32/esp2$ cargo build
   Compiling esp-idf-sys v0.33.4
   Compiling embedded-svc v0.26.1
error: failed to run custom build command for `esp-idf-sys v0.33.4`

Caused by:
  process didn't exit successfully: `/home/iyg1kor/projects/esp32/esp2/target/debug/build/esp-idf-sys-e59d96eabaaaa50e/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  IDF_PYTHON_ENV_PATH=/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/python_env/idf5.1_py3.8_env
  ESP_IDF_VERSION=5.1
  PATH=/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/cmake/3.24.0/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/ninja/1.10.2/:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/python_env/idf5.1_py3.8_env/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/tmp/tmp81919_icidf_67045
  Current system platform: linux-amd64
  Skipping xtensa-esp32-elf@esp-12.2.0_20230208 (already installed)
  Skipping cmake@3.24.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/python_env/idf5.1_py3.8_env
  ESP_IDF_VERSION=5.1
  PATH=/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/cmake/3.24.0/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/ninja/1.10.2/:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/python_env/idf5.1_py3.8_env/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/tmp/tmpjic9hbrkidf_67045
  cargo:rerun-if-changed=/home/iyg1kor/projects/esp32/esp2/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/iyg1kor/projects/esp32/esp2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f447ffa9fbedba9f/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1" IDF_TARGET="esp32" IDF_TOOLS_PATH="/home/iyg1kor/projects/esp32/esp2/.embuild/espressif" PATH="/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1/tools:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/ninja/1.10.2/:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/tools/cmake/3.24.0/bin:/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/python_env/idf5.1_py3.8_env/bin:/home/iyg1kor/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/iyg1kor/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/iyg1kor/.cargo/bin:/opt/osd/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PROJECT_DIR="/home/iyg1kor/projects/esp32/esp2" SDKCONFIG_DEFAULTS="/home/iyg1kor/projects/esp32/esp2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f447ffa9fbedba9f/out/gen-sdkconfig.defaults;/home/iyg1kor/projects/esp32/esp2/sdkconfig.defaults" "cmake" "/home/iyg1kor/projects/esp32/esp2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f447ffa9fbedba9f/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/python_env/idf5.1_py3.8_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/home/iyg1kor/projects/esp32/esp2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-f447ffa9fbedba9f/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- git describe returned 'fatal: bad revision 'HEAD''
  -- Could not use 'git describe' to determine PROJECT_VER.
  -- Building ESP-IDF components for target esp32
[... 20 lines cut ...]
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.1.1") }
  Using esp-idf v5.1.1 at '/home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1'
  ERROR: /home/iyg1kor/projects/esp32/esp2/.embuild/espressif/espidf.constraints.v5.1.txt doesn't exist. Perhaps you've forgotten to run the install scripts. Please check the installation guide for more information.
  CMake Error at /home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/build.cmake:363 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/build.cmake:498 (__build_check_python)
    /home/iyg1kor/projects/esp32/esp2/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/project.cmake:547 (idf_build_process)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /home/iyg1kor/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

Tederean, 2023-10-27:
Same error message using fresh created project template and dev-container from 1h ago...

SergioGasquez, 2023-10-27:
I just tried reproducing the issue, and I was able to build a template project with the exact same prompts. 

@yashwanthsinghm are you sourcing the export file `. $HOME/export-esp.sh`?
@Tederean trying with devcontainers, will update once I tested it

SergioGasquez, 2023-10-27:
@Tederean I was also able to build the project inside the provided devcontainer.
1. `cargo generate esp-rs/esp-idf-template cargo --name issue165 -d mcu=esp32 -d espidfver=v5.1 -d advanced=true -d std=true  -d devcontainer=true -d wokwi=false -d ci=false`
2. `cd issue165`
3. `code .`
4. Inside VsCode "Reopen in container"
5. `container$ cargo build`

SergioGasquez, 2023-10-27:
> Yes, I have sourced export file @SergioGasquez, I have tired both on windows and Linux machine I’m getting same build issue. How to solve this ??

A few ideas that you can try:
- `cargo clean && rm -rf .embuild && cargo b`
- If you have docker, I will suggest trying the devcontainer approach with the instructions in https://github.com/esp-rs/esp-idf-template/issues/165#issuecomment-1783202630

Tederean, 2023-10-27:
@SergioGasquez 

I executed `./scripts/build.sh` as suggested by readme file (https://github.com/esp-rs/esp-idf-template/blob/master/cargo/docs/README.md) instead of `cargo build`. 

Using `./scripts/build.sh` in first place caused the error i encountered.  When i use `cargo build` in first place it works after 45min compile time 😲

Now i executed `./scripts/build.sh debug` after `cargo build` and this works just fine.

(Are this compile times normal? 45mins in release and 7min in debug...?)

N3xed, 2023-10-27:
> (Are this compile times normal? 45mins in release and 7min in debug...?)

45min seems a bit excessive, but most of this is probably the clone of the esp-idf repo. If you run the initial build with `cargo build -vv` (very verbose) it will show all build script output (including esp-idf-sys's) which displays the progress of the clone and build of the esp-idf. See the [build options](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md) for configurations of where the esp-idf + tools are stored, how it is built, etc.

yashwanthsinghm, 2023-10-28:
Hi @SergioGasquez i tried both the options, even i followed the step to run on dev container, it didn't work for me here are the logs.

```
esp@420b80357d12:~/issue165$ cargo build
    Blocking waiting for file lock on build directory
   Compiling libc v0.2.147
.......
error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/esp/.rustup/toolchains/esp/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/esp/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/vscode/vscode-server/bin/linux-x64/e7e037083ff4455cf320e344325dacb480062c3c/bin/remote-cli:/home/esp/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/esp/.cargo/bin" VSLANG="1033" "cc" "-m64" "/tmp/rustcj87mBW/symbols.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.00.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.01.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.02.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.03.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.04.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.05.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.06.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.07.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.08.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.09.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.10.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.11.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.12.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.13.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.14.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.build_script_build.c06dcb89121d25e-cgu.15.rcgu.o" "/home/esp/issue165/target/debug/build/esp-idf-sys-6eb62c15753fd0fd/build_script_build-6eb62c15753fd0fd.4t0wnlmclvjjivtv.rcgu.o" "-Wl,--as-needed" "-L" "/home/esp/issue165/target/debug/deps" "-L" "/home/esp/.rustup/toolchains/esp/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/esp/issue165/target/debug/deps/libenvy-53e8e3a18d82f2cd.rlib" "/home/esp/issue165/target/debug/deps/libcargo_metadata-ba3e4a2c4f2ea60e.rlib" "/home/esp/issue165/target/debug/deps/libcargo_platform-eb18bef61b69201d.rlib" "/home/esp/issue165/target/debug/deps/libsemver-52a2d3dc37fd4489.rlib" "/home/esp/issue165/target/debug/deps/libcamino-08943a4f334c65eb.rlib" "/home/esp/issue165/target/debug/deps/libembuild-f2ba18c73796f86e.rlib" "/home/esp/issue165/target/debug/deps/libfiletime-8a7ca627eef8c21d.rlib" "/home/esp/issue165/target/debug/deps/libglobwalk-3c265d2bf1fa52a9.rlib" "/home/esp/issue165/target/debug/deps/libignore-acb6444778b03b2e.rlib" "/home/esp/issue165/target/debug/deps/libwalkdir-e392bc4dd33db064.rlib" "/home/esp/issue165/target/debug/deps/libsame_file-4d02d4760fb117bf.rlib" "/home/esp/issue165/target/debug/deps/libthread_local-d0a9f0251a3e062b.rlib" "/home/esp/issue165/target/debug/deps/libonce_cell-7302fb7e29d0287e.rlib" "/home/esp/issue165/target/debug/deps/libglobset-e3809edd914be9fa.rlib" "/home/esp/issue165/target/debug/deps/libfnv-d76296c6599a31fb.rlib" "/home/esp/issue165/target/debug/deps/libbstr-e4de0c79cf2678d1.rlib" "/home/esp/issue165/target/debug/deps/libremove_dir_all-e92b4d1d550119a7.rlib" "/home/esp/issue165/target/debug/deps/libfs_at-a58a7a9ee4260e11.rlib" "/home/esp/issue165/target/debug/deps/libnix-b6a90b8a6184e5a5.rlib" "/home/esp/issue165/target/debug/deps/libcvt-bef9bdbe9cddf487.rlib" "/home/esp/issue165/target/debug/deps/libnormpath-449490608d639a30.rlib" "/home/esp/issue165/target/debug/deps/libtempfile-b965a3e9bd42b7b8.rlib" "/home/esp/issue165/target/debug/deps/libfastrand-b76ce8c10b4a479b.rlib" "/home/esp/issue165/target/debug/deps/libthiserror-e06eee6f30e5d754.rlib" "/home/esp/issue165/target/debug/deps/libcmake-c5e56400805f4de0.rlib" "/home/esp/issue165/target/debug/deps/libcc-3af3a451b3e6eaab.rlib" "/home/esp/issue165/target/debug/deps/libse [... cut ...]
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `esp-idf-sys` (build script) due to previous error
```

rustup show

```
esp@420b80357d12:~/issue165$ rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/esp/.rustup

esp (overridden by '/home/esp/issue165/rust-toolchain.toml')
rustc 1.73.0-nightly (2fba83302 2023-10-18) (1.73.0.1)

```

SergioGasquez, 2023-10-31:
> Hi @SergioGasquez i tried both the options, even i followed the step to run on dev container, it didn't work for me here are the logs.

This is very strange, if you generated the example with my same command (using the same arguments), it should work, at least using devcontainers, as the environment should be exactly the same. Only thing that I can think of, is that you publish into a repo the generated project and I try to build it.

bingliu221, 2023-11-02:
I ran into the same issue on archlinux.
Then I tried the dev container method @SergioGasquez provided above, `cargo build` succeeded.

I still wonder what's causing the issue though.

Vollbrecht, 2023-11-02:
make sure that on your machine the equivalent packages of this debian packages are installed " git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0 clang libclang-dev"

bingliu221, 2023-11-02:
I also tried sourcing `~/export-esp.sh`, it didn't work before.
Somehow it works now. This is weird. Perhaps because I updated some packages?

ivmarkov, 2023-11-02:
> make sure that on your machine the equivalent packages of this debian packages are installed " git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0 clang libclang-dev"

Do we need all of these?
`ccache` - why?
The `clang` stuff might not be useful at all for `xtensa` chips

Vollbrecht, 2023-11-02:
its the shotgun approach yeah, but that deps are the one official mentioned in the esp-idf install guide minus the clang dep

ivmarkov, 2023-11-02:
> its the shotgun approach yeah, but that deps are the one official mentioned in the esp-idf install guide minus the clang dep

Shall we then mention these in the "Prerequisites" section?

Vollbrecht, 2023-11-02:
Yeah it might be a good idea - we could directly link them to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#for-linux-users
There is also some for macOS mentiond under it and for M1 users specifically.
And personally i would add the clang dep's as i needed it on a freshly debian install last week. So updating it in the template to mentiond it there  and maybe in the book on this [page](https://esp-rs.github.io/book/installation/std-requirements.html)?

strahe, 2023-11-29:
> I also tried sourcing `~/export-esp.sh`, it didn't work before. Somehow it works now. This is weird. Perhaps because I updated some packages?

Is it solved?  same issue on my archlinux too

closed 2024-01-26 by SergioGasquez