esp-rs/esp-idf-template #174: Can't build new project

ysak-y, 2023-11-14:
## Bug description

Build fail when build new project from `esp-idf-template`.

- Would you like to work on a fix? n

## To Reproduce

Steps to reproduce the behavior:

1. Run `cargo generate esp-rs/esp-idf-template cargo` and generate project without advanced configurations
2. Run `cargo build` on project directory

## Expected behavior

Build success.

## Screenshots

Below is full build log. `xtensa-esp32-elf-gcc: error: unrecognized command-line option '--target=xtensa-esp32-espidf'` is the actual reason of error. I'm not sure whether it should be reported on this repository, please let me know if there is proper place to report.
I did [prerequisites](https://github.com/esp-rs/esp-idf-template#prerequisites) already and I can build the project until sometime yesterday.

```
error: failed to run custom build command for `esp-idf-sys v0.33.7`

Caused by:
  process didn't exit successfully: `/Users/yoshiakiyamada/git/rust-on-m5stack/test2/target/debug/build/esp-idf-sys-19b83de14286a6a1/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
  Submodule path 'components/bt/controller/lib_esp32': checked out 'e1b3a29c58a452a6c017113686262094afd87042'
  Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out '5fcfd0e457a76ebb234086ccd6dc391690d69e90'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out 'b438f60a295183e7c67eb42ae05f4580f4b1ced0'
  Submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib': checked out 'd785de0a7c46d9badcd73bc83c2e5cb78f7054b2'
  Submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib': checked out '35bd3cd7352014d303a96c46d8ea8446ea0a9a54'
  Submodule path 'components/bt/host/nimble/nimble': checked out '9478bc510fa9fc792d98efbc80dc547b97a13b99'
  Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
  Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
  Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
  Submodule path 'components/esp_coex/lib': checked out '37698e17a4c7fd2e0872317fc8befaac3bb10cb4'
  Submodule path 'components/esp_phy/lib': checked out '998e05636344949ba219e8ac56feb733e27bf5d2'
  Submodule path 'components/esp_wifi/lib': checked out '0a89d5ffd2c452407940c2e617434e54f3c34576'
  Submodule path 'components/heap/tlsf': checked out '8c9cd0517adf99e363812e9a295dfe3898fdd345'
  Submodule path 'components/ieee802154/lib': checked out '102b03c8095de8a337c293f79dce189be63186f3'
  Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
  Submodule path 'components/lwip/lwip': checked out '269a58f18597aedec31c439e34cdaea47229dda3'
  Submodule path 'components/mbedtls/mbedtls': checked out '65b3c0857c9f207dcb0b646f2b4ebd8e5edf83c1'
  Submodule path 'components/mqtt/esp-mqtt': checked out 'dffabb067fb3c39f486033d2e47eb4b1416f0c82'
  Submodule path 'components/openthread/lib': checked out 'c9c17a0b6634af016111c9817f0be1fe0e7112f6'
  Submodule path 'components/openthread/openthread': checked out '5beae143700db54c6e9bd4b15a568abe2f305723'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'abc67a11c6db271bedbb9f58be85d6f4e2ea8389'
  Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  ESP_IDF_VERSION=5.1
  PATH=/Users/yoshiakiyamada/git/rust-on-m5stack/test2/.embuild/espressif/esp-idf/v5.1.1/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/var/folders/_p/r21b9lq54713qfqnxhpsmpvr0000gn/T/tmpitdv2oanidf_37438
  Creating a new Python environment in /Users/yoshiakiyamada/git/rust-on-m5stack/test2/.embuild/espressif/python_env/idf5.1_py3.11_env
  Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt
  Destination: /Users/yoshiakiyamada/git/rust-on-m5stack/test2/.embuild/espressif/espidf.constraints.v5.1.txt.tmp
[... 299 lines cut ...]
      /Users/yoshiakiyamada/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections --target=xtensa-esp32-espidf -o CMakeFiles/cmTC_93ccf.dir/testCCompiler.c.obj -c /Users/yoshiakiyamada/git/rust-on-m5stack/test2/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-8fb7717dc012a26e/out/build/CMakeFiles/CMakeTmp/testCCompiler.c
      xtensa-esp32-elf-gcc: error: unrecognized command-line option '--target=xtensa-esp32-espidf'
      ninja: build stopped: subcommand failed.
      
      

    

    CMake will not be able to correctly generate this project.
  Call Stack (most recent call first):
    /Users/yoshiakiyamada/git/rust-on-m5stack/test2/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/project.cmake:448 (__project)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /Users/yoshiakiyamada/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

## Environment

- OS: macOS 14.0
- How did you install the environment: espup 0.5.0

ysak-y, 2023-11-14:
I realized `espup` would too old (current is 0.9 but I use 0.5). I'll update `espup` and try again.

SergioGasquez, 2023-11-14:
Hi! Let me know how it goes after updating installation with recent `espup`, also, which device are you targeting ESP32?

ysak-y, 2023-11-14:
Hi, thank you for your comment, still failing even if update espup to `0.9`. It seems same error happening.

> /Users/yoshiakiyamada/git/rust-on-m5stack/play-wav/.embuild/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections --target=xtensa-esp32-espidf -o CMakeFiles/cmTC_cf89f.dir/testCCompiler.c.obj -c /Users/yoshiakiyamada/git/rust-on-m5stack/play-wav/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-ab3bcb30f5ec8ba0/out/build/CMakeFiles/CMakeTmp/testCCompiler.c
      xtensa-esp32-elf-gcc: error: unrecognized command-line option '--target=xtensa-esp32-espidf'

I noticed export-esp.sh now indicates `esp-13.2.0` as PATH but it seems build script still using `esp-12.2.0` that is older version as above. I wonder why... There is no `esp-12.2.0` when `echo $PATH` but still use it when build...
`export PATH="/Users/yoshiakiyamada/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:$PATH"`

Let me know if you need more information.

LarryMerino, 2023-11-14:
Same problem with ESP32C3

```Toml
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.1"
```

```
aused by:
  process didn't exit successfully: `/Volumes/T7/EOS/torrot-project/sand-box/build-test/target/debug/build/esp-idf-sys-c44dda218b03b259/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
  Submodule path 'components/bt/controller/lib_esp32': checked out 'e1b3a29c58a452a6c017113686262094afd87042'
  Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out '5fcfd0e457a76ebb234086ccd6dc391690d69e90'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out 'b438f60a295183e7c67eb42ae05f4580f4b1ced0'
  Submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib': checked out 'd785de0a7c46d9badcd73bc83c2e5cb78f7054b2'
  Submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib': checked out '35bd3cd7352014d303a96c46d8ea8446ea0a9a54'
  Submodule path 'components/bt/host/nimble/nimble': checked out '9478bc510fa9fc792d98efbc80dc547b97a13b99'
  Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
  Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
  Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
  Submodule path 'components/esp_coex/lib': checked out '37698e17a4c7fd2e0872317fc8befaac3bb10cb4'
  Submodule path 'components/esp_phy/lib': checked out '998e05636344949ba219e8ac56feb733e27bf5d2'
  Submodule path 'components/esp_wifi/lib': checked out '0a89d5ffd2c452407940c2e617434e54f3c34576'
  Submodule path 'components/heap/tlsf': checked out '8c9cd0517adf99e363812e9a295dfe3898fdd345'
  Submodule path 'components/ieee802154/lib': checked out '102b03c8095de8a337c293f79dce189be63186f3'
  Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
  Submodule path 'components/lwip/lwip': checked out '269a58f18597aedec31c439e34cdaea47229dda3'
  Submodule path 'components/mbedtls/mbedtls': checked out '65b3c0857c9f207dcb0b646f2b4ebd8e5edf83c1'
  Submodule path 'components/mqtt/esp-mqtt': checked out 'dffabb067fb3c39f486033d2e47eb4b1416f0c82'
  Submodule path 'components/openthread/lib': checked out 'c9c17a0b6634af016111c9817f0be1fe0e7112f6'
  Submodule path 'components/openthread/openthread': checked out '5beae143700db54c6e9bd4b15a568abe2f305723'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'abc67a11c6db271bedbb9f58be85d6f4e2ea8389'
  Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  ESP_IDF_VERSION=5.1
  PATH=/Volumes/T7/EOS/torrot-project/sand-box/build-test/.embuild/espressif/esp-idf/v5.1.1/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/var/folders/45/_dlnfy713_3g0hfrjfq6syz00000gn/T/tmp31feygavidf_23205
  Creating a new Python environment in /Volumes/T7/EOS/torrot-project/sand-box/build-test/.embuild/espressif/python_env/idf5.1_py3.11_env
  Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt
  Destination: /Volumes/T7/EOS/torrot-project/sand-box/build-test/.embuild/espressif/espidf.constraints.v5.1.txt.tmp
Done
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
[... 286 lines cut ...]

      Change Dir: /Volumes/T7/EOS/torrot-project/sand-box/build-test/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-1b1ac1cba6d99804/out/build/CMakeFiles/CMakeTmp

      Run Build Command(s):/Volumes/T7/EOS/torrot-project/sand-box/build-test/.embuild/espressif/tools/ninja/1.10.2/ninja cmTC_2df0c && [1/2] Building C object CMakeFiles/cmTC_2df0c.dir/testCCompiler.c.obj
      FAILED: CMakeFiles/cmTC_2df0c.dir/testCCompiler.c.obj
      /Volumes/T7/EOS/torrot-project/sand-box/build-test/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc   -march=rv32imc_zicsr_zifencei  -ffunction-sections -fdata-sections --target=riscv32imc_zicsr_zifencei-esp-espidf -o CMakeFiles/cmTC_2df0c.dir/testCCompiler.c.obj -c /Volumes/T7/EOS/torrot-project/sand-box/build-test/target/riscv32imc-esp-espidf/debug [... cut ...]
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

My Cmake version is `cmake version 3.27.7` 

I created other projects before for the same target and didn't get this error. Any ideas what could be going on?

ysak-y, 2023-11-14:
> which device are you targeting ESP32?

I'll use M5Stack Gray. target in `.cargo/config.toml` indicates `xtensa-esp32-espidf`.

Vollbrecht, 2023-11-14:
The gcc version used to build esp-idf is closely tied to the esp-idf release itself, when esp-idf is installed in .embuild it will come with its own gcc version. The version that is used is essentially the one declared in the respective [tools.json](https://github.com/espressif/esp-idf/blob/master/tools/tools.json) file. So for std development you can pass the --std flag to espup to don't install the potential newer gcc versions. But they should not lead to any problems its just dead weight in this context.

LarryMerino, 2023-11-14:
Yes, I'm using MacOs @Vollbrecht 
I have tried the command you suggest and it works

```
build-test on  master [?] is 📦 v0.1.0 via 🦀 v1.74.0-nightly on ☁️  (eu-west-3)
❯ CRATE_CC_NO_DEFAULTS=1 cargo build
   Compiling esp-idf-sys v0.33.7
   Compiling esp-idf-hal v0.42.5
   Compiling esp-idf-svc v0.47.3
   Compiling build-test v0.1.0 (/Volumes/T7/EOS/torrot-project/sand-box/build-test)
    Finished dev [optimized + debuginfo] target(s) in 1m 08s
```

Vollbrecht, 2023-11-14:
It seams that the new released version of the cc crate version 1.0.84 brakes MacOS.  The following PR maybe introduced the behavior https://github.com/rust-lang/cc-rs/pull/709

ysak-y, 2023-11-14:
I'm using macOS and it works too. Thanks @Vollbrecht !!

ivmarkov, 2023-11-14:
> I'm using macOS and it works too. Thanks @Vollbrecht !!

Can you explain why you closed the bug, and how the *hypothesis* that the issue at hand is caused by latest `CC` (btw, these failures seem to only happen on Mac - CI runs fine on Linux?) **fixed the problem for you**?

ivmarkov, 2023-11-14:
Still - let's keep this issue open. Ideally, we should not be asking users to use `CRATE_CC_NO_DEFAULTS` and we have to understand how exactly `CC` was regressed in such a way, that it does not even find the correct GCC anymore (even though it is the first one on the path?)

Vollbrecht, 2023-11-14:
> Still - let's keep this issue open. Ideally, we should not be asking users to use `CRATE_CC_NO_DEFAULTS` and we have to understand how exactly `CC` was regressed in such a way, that it does not even find the correct GCC anymore (even though it is the first one on the path?)

Can you rename the issue to make it clear that it is scoped to MacOS user?

Vollbrecht, 2023-11-14:
cc version 1.0.84 was yanked - so if you get back to version 1.0.83 you should not need to compile with `CRATE_CC_NO_DEFAULTS` here are related PR's https://github.com/rust-lang/cc-rs/pull/901

cKhoff, 2024-03-07:
This helped me as well.
`~/git/esp/humidity master ?8 ❯ CRATE_CC_NO_DEFAULTS=1 cargo build                                                                          19:04:31
   Compiling esp-idf-sys v0.34.1
   Compiling esp-idf-hal v0.43.1
   Compiling esp-idf-svc v0.48.1
   Compiling humidity v0.1.0 (/Users/xxxxx/git/esp/humidity)
error[E0063]: missing field `hys_ctrl_mode` in initializer of `esp_idf_sys::gpio_config_t`
    --> /Users/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.43.1/src/gpio.rs:1442:15
     |
1442 |     let cfg = gpio_config_t {
     |               ^^^^^^^^^^^^^ missing `hys_ctrl_mode`

For more information about this error, try `rustc --explain E0063`.
error: could not compile `esp-idf-hal` (lib) due to 1 previous error`

in my case i also had to monkey patch this file by adding hys_ctrl_mode: 0 to make it successfully compile.

ivmarkov, 2024-11-26:
Closing now that:
a) The current `esp-idf-template` has a `CC_NO_DEFAULTS=1` in `.cargo/config.toml`
b) The programmatic invocation of `cmake-rs` without default flags being added got merged and released yesterday.