esp-rs/esp-idf-template #178: `-C default-linker-libraries` breaks the build at link time (pthread multiple symbols) with recent ESP IDF toolchains

ivmarkov, 2023-12-28:
Also see [this](https://github.com/esp-rs/esp-idf-hal/issues/289#issuecomment-1870878887).

Need to get to the bottom of this but the most likely hypothesis is as per the comment above.

WORKAROUND: Remove `-C default-linker-libraries` from `.cargo/config.toml`.

ivmarkov, 2023-12-28:
Seems to me that linking to [`libnosys.a`](https://stackoverflow.com/questions/76774814/what-exactly-does-a-c-toolchain-expect-to-be-linked-with-for-bare-metal-develo) is the culprit.

ivmarkov, 2023-12-28:
Digging a bit further: seems that the problem is with ESP IDF's **own** fork of newlib. Specifically, this thing: https://github.com/espressif/newlib-esp32/commit/ff0b7d937917fd4847d9eb1a12d709a1cb651938

Still need to understand 
(a) why this code was necessary in the first place - perhaps - to fill in the gaps in ESP IDF's own pthread implementation?
(b) why it does not link with weak syms
(c) why it does link only when `default-linker-libraries` is used

mrchantey, 2024-02-13:
fyi in my case after removing `-C default-linker-libraries` I got all kinds of errors when trying to use `esp_idf_svc` features like wifi. All good after doing a full clean including removing `.embuild`, then updating dependencies.

My current dependencies are working and look like this:
```toml
#[dependencies]
embedded-svc = "0.27"
embedded-hal = "1.0.0"
esp-idf-hal = "0.43"
esp-idf-svc = { version = "0.48", default-features = false,features = ["std"] }

```

ivmarkov, 2025-01-02:
This can finally be closed. `-C default-linker-libraries` should simply **not** be used, period.
We should/are good enough with the libraries emitted by the ESP-IDF linker script. No more, no less.

closed 2025-01-02 by ivmarkov