esp-rs/esp-idf-template #204: Project generated do not work with esp-idf v5.2.1

PocketPi, 2024-04-02:
## Bug description

I have esp-idf v5.2.1 installed as im using that for my main project. I have now tried to make a rust project off the template but that do not build. This is what i get from the build process: 

```
peter:~/data/code/esprs-test$ cargo build
   Compiling esp-idf-sys v0.34.1
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.34.1: Compiler version doesn't include clang or GCC: "cc" "--version"
warning: esp-idf-sys@0.34.1: Compiler version doesn't include clang or GCC: "c++" "--version"

error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `/home/peter/data/code/esprs-test/target/debug/build/esp-idf-sys-be56d55db34120ca/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:rerun-if-changed=/home/peter/data/code/esprs-test/sdkconfig.defaults
  cargo:warning=Compiler version doesn't include clang or GCC: "cc" "--version"
  cargo:warning=Compiler version doesn't include clang or GCC: "c++" "--version"
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/peter/data/tools/esp-idf" IDF_TARGET="esp32" PATH="/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/data/tools/esp-idf/components/espcoredump:/home/peter/data/tools/esp-idf/components/partition_table:/home/peter/data/tools/esp-idf/components/app_update:/home/peter/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/peter/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/peter/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/peter/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/peter/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/peter/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/peter/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/peter/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/peter/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/peter/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/data/tools/esp-idf/components/espcoredump:/home/peter/data/tools/esp-idf/components/partition_table:/home/peter/data/tools/esp-idf/components/app_update:/home/peter/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/peter/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/peter/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/peter/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/peter/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/peter/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/peter/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/peter/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/peter/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/peter/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/peter/.espressif/python_env/idf5.2_py3.11_env/bin:/home/peter/data/tools/esp-idf/tools:/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/data/tools:/home/peter/data/tools/sdcc/bin:/home/peter/.yarn/bin:/home/peter/.config/yarn/global/node_modules/.bin:/home/peter/.nvm/versions/node/v18.18.0/bin:/home/peter/.rvm/gems/ruby-2.6.5/bin:/home/peter/.rvm/gems/ruby-2.6.5@global/bin:/home/peter/.rvm/rubies/ruby-2.6.5/bin:/home/peter/.local/bin:/home/peter/.cargo/bin:/home/peter/.local/bin:/home/peter/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/peter/Android/Sdk/emulator:/home/peter/Android/Sdk/platform-tools:/home/peter/Android/Sk/emulator:/home/peter/Android/Sk/platform-tools:/home/peter/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin:/home/peter/Dotfiles:/home/peter/.rvm/bin:/home/peter/.rvm/bin" PROJECT_DIR="/home/peter/data/code/esprs-test" SDKCONFIG_DEFAULTS="/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out/gen-sdkconfig.defaults;/home/peter/data/code/esprs-test/sdkconfig.defaults" "cmake" "/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/peter/data/tools/esp-idf/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/peter/.espressif/python_env/idf5.2_py3.11_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Configuring incomplete, errors occurred!

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/home/peter/data/tools/esp-idf",
          ),
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v5.2.1 environment at '/home/peter/data/tools/esp-idf'
  CMake Error at /home/peter/data/tools/esp-idf/tools/cmake/project.cmake:506 (__project):
    Running

     '/home/peter/.local/bin/ninja' '--version'

    failed with:

     Traceback (most recent call last):
      File "/home/peter/.local/bin/ninja", line 5, in <module>
        from ninja import ninja

    ModuleNotFoundError: No module named 'ninja'

  Call Stack (most recent call first):
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

- Would you like to work on a fix? y

## To Reproduce

Steps to reproduce the behavior:

1. cargo generate esp-rs/esp-idf-template cargo
2. giv project name 
3. select esp32
4. select no
5. cd project
6. cargo build

## Environment

- OS: ubuntu 23.10
- How did you install the environment: [e.g. espup 0.3.2]
 setup as described in this guide: https://github.com/esp-rs/esp-idf-template?tab=readme-ov-file#prerequisites

PocketPi, 2024-04-02:
I Just installed v5.1 and it still doesnt work. 

```
peter:~/data/code/esprs-test$ cargo build
   Compiling bindgen v0.63.0
   Compiling embuild v0.31.4
   Compiling esp-idf-sys v0.34.1
   Compiling esp-idf-hal v0.43.1
   Compiling esp-idf-svc v0.48.1
   Compiling esprs-test v0.1.0 (/home/peter/data/code/esprs-test)
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.34.1: Compiler version doesn't include clang or GCC: "cc" "--version"
warning: esp-idf-sys@0.34.1: Compiler version doesn't include clang or GCC: "c++" "--version"

error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `/home/peter/data/code/esprs-test/target/debug/build/esp-idf-sys-be56d55db34120ca/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:rerun-if-changed=/home/peter/data/code/esprs-test/sdkconfig.defaults
  cargo:warning=Compiler version doesn't include clang or GCC: "cc" "--version"
  cargo:warning=Compiler version doesn't include clang or GCC: "c++" "--version"
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/peter/data/tools/esp-idf-v5.1" IDF_TARGET="esp32" PATH="/home/peter/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/peter/data/tools/esp-idf-v5.1/components/espcoredump:/home/peter/data/tools/esp-idf-v5.1/components/partition_table:/home/peter/data/tools/esp-idf-v5.1/components/app_update:/home/peter/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/home/peter/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:/home/peter/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/peter/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/peter/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/peter/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/peter/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/peter/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:/home/peter/.espressif/python_env/idf5.1_py3.11_env/bin:/home/peter/data/tools/esp-idf-v5.1/tools:/home/peter/data/tools:/home/peter/data/tools/sdcc/bin:/home/peter/.yarn/bin:/home/peter/.config/yarn/global/node_modules/.bin:/home/peter/.nvm/versions/node/v18.18.0/bin:/home/peter/.rvm/gems/ruby-2.6.5/bin:/home/peter/.rvm/gems/ruby-2.6.5@global/bin:/home/peter/.rvm/rubies/ruby-2.6.5/bin:/home/peter/.local/bin:/home/peter/.cargo/bin:/home/peter/.local/bin:/home/peter/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/peter/Android/Sdk/emulator:/home/peter/Android/Sdk/platform-tools:/home/peter/Android/Sk/emulator:/home/peter/Android/Sk/platform-tools:/home/peter/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin:/home/peter/Dotfiles:/home/peter/.rvm/bin:/home/peter/.rvm/bin" PROJECT_DIR="/home/peter/data/code/esprs-test" SDKCONFIG_DEFAULTS="/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out/gen-sdkconfig.defaults;/home/peter/data/code/esprs-test/sdkconfig.defaults" "cmake" "/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/peter/data/tools/esp-idf-v5.1/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/peter/.espressif/python_env/idf5.1_py3.11_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/home/peter/data/code/esprs-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-95e72fc75916bcbf/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Configuring incomplete, errors occurred!

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/home/peter/data/tools/esp-idf-v5.1",
          ),
          extra_components: [],
[... cut ...]
      File "/home/peter/.local/bin/ninja", line 5, in <module>
        from ninja import ninja

    ModuleNotFoundError: No module named 'ninja'

  Call Stack (most recent call first):
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
``

Vollbrecht, 2024-04-02:
esp-idf uses a specifc set of tools -> cmake, ninja, riscv-gcc , xtensa-gcc. When you install a specifc version of esp-idf via the python install scripts it also should install all this tools. 

In a fully managed esp-idf repo ( auto installed by us if no activated esp-idf env found) we install this tools for you inside the .embuild dir for the respected esp-idf version. 

It looks like your install is missing ninja so that's why it seams to fail.

 You don't have [this env variable](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_tools_install_dir-esp_idf_tools_install_dir) set right?

Vollbrecht, 2024-04-02:
Technically it should be able to use the "other" sourced environment if it is setup with all the tools needed. The build script is activity trying to do so, but again in this case seams to stumble onto a maybe not fully installed ninja ?

ivmarkov, 2024-04-02:
@PocketPi What is the OS and hardware of the host where you are compiling?

PocketPi, 2024-04-17:
ubuntu 23.10 and its an intel cpu

Vollbrecht, 2024-09-07:
closing this for now as i outlined what the problem was. In general we recommend to either only use exactly one external activated env, or use our managed installs. 

Our managed installs don't permanently change env vars and so can't create this issues. And if you are using a single external activated env, through the official script the impact also should be limted. 

In any case keep your env vars clean.

closed 2024-09-07 by Vollbrecht