esp-rs/esp-idf-template #206: Unable to flash "Broken Pipe" m1 mac

toddmcintire, 2024-04-12:
## Bug description

- Would you like to work on a fix? [y/n]n

## To Reproduce

Steps to reproduce the behavior:

1. ... build with ```CRATE_CC_NO_DEFAULTS=1 cargo build```
2. ... run with ```cargo run```

## Expected behavior

output along the lines of 

```I (268) cpu_start: Starting scheduler. Hello, world```

## Screenshots

[image]

## Environment

- OS: Sonomia 14.4.1
- How did you install the environment: [e.g. espup 0.3.2]
both rustup and espup but attempted to mostly follow https://docs.esp-rs.org/std-training/
## Additional context

I can get the std-training/intro/hardware-check to work but attempting to run the esp-idf-template results in a ```Error:   × Broken pipe```

Vollbrecht, 2024-04-12:
What board are you using? What version of `espflash` are you using` ? What happens if you manually connect after flashing with `espflash monitor`?

toddmcintire, 2024-04-15:
> What board are you using? What version of `espflash` are you using`? What happens if you manually connect after flashing with`espflash monitor`?

Im using the [ESP32-C3-DevKit-RUST-1 ](https://www.adafruit.com/product/5787?utm_source=Adafruit+Products+Newsletter&utm_campaign=816eb9f914-EMAIL_CAMPAIGN_2023_08_15_05_12&utm_medium=email&utm_term=0_-816eb9f914-%5BLIST_EMAIL_ID%5D) and my `espflash` is version `3.0.0`.
If I plug the board in and run `espflash monitor` I get the following output 
[image]
If I press `CTRL + R` I get the broken pipe.
[image]

Vollbrecht, 2024-04-15:
espflash has a additional optional argument `--no-stub`. Can you try it out and see if it makes a difference in your case?

toddmcintire, 2024-04-15:
> espflash has a additional optional argument `--no-stub`. Can you try it out and see if it makes a difference in your case?

Sorry it seems this bug report may have been in error. 
Before running `espflash` with `--no-stub` I tried with just monitor and it spat out an error about `Invalid chip id`. To which I just deleted the entire folder and re-generated everything with cargo-generate. 
Upon flashing I now get the correct output as seen below.
[image]
I do apologize for all confusion created by this issue.

closed 2024-04-15 by toddmcintire