esp-rs/esp-idf-template #213: Awesome, list of projects based on this template

stappersg, 2024-06-06:
## Motivations

We learn much from examples.

## Solution

Have in this **template** a link, an URL, to a list of projects based on this **template**.

## Alternatives

Inspiration:
* https://github.com/rust-unofficial/awesome-rust
* https://github.com/esp-rs/awesome-esp-rust#no_std 
* https://github.com/jetli/awesome-yew
*  

## Additional context

I don't know much https://github.com/esp-rs/awesome-esp-rust#no_std projects are based on this template project.

SergioGasquez, 2024-06-07:
I would say that most of (or all) the projects of https://github.com/esp-rs/awesome-esp-rust were created using templates, and if they were not, it would have been easier to do so. So, I don't really see the value of having a list of projects based on this template, as it's just a tool that make things simpler, but you can achieve the same by doing everything manually.

stappersg, 2024-06-07:
It was https://www.reddit.com/r/rust/comments/1d9f9nu/bare_metal_on_esp32/ that triggered me to open this issue.  The goal is to have a collection of _esp-rs/esp-idf-template success stories_.

SergioGasquez, 2024-06-11:
I think that list would be a duplication of https://github.com/esp-rs/awesome-esp-rust?tab=readme-ov-file#std, and would require more maintenance. I will be closing the issue as I don't see advantages, but thanks for suggesting it!

closed 2024-06-11 by SergioGasquez