esp-rs/esp-idf-template #214: Cannot set PROJECT_NAME and PROJECT_VER in IDF project

elosev, 2024-06-09:
## Bug description

Cannot set PROJECT_NAME and PROJECT_VER in project generated by the template. For an empty project PROJECT_NAME and PROJECT_VER are coming from esp-idf-sys and are  hardcoded to 'libespidf' and version defined by 'git describe'. I found that as a work-around both can be changed by patching esp-idf-sys:

```
diff --git a/build/native/cargo_driver.rs b/build/native/cargo_driver.rs
index c263ba76..bd2f4234 100644
--- a/build/native/cargo_driver.rs
+++ b/build/native/cargo_driver.rs
@@ -531,9 +531,9 @@ pub fn build() -> Result<EspIdfBuildOutput> {
     let target = replies
         .get_codemodel()?
         .into_first_conf()
-        .get_target("libespidf.elf")
+        .get_target("unicorn.elf")
         .unwrap_or_else(|| {
-            bail!("Could not read build information from cmake: Target 'libespidf.elf' not found")
+            bail!("Could not read build information from cmake: Target 'unicorn.elf' not found")
         })?;
 
     let compiler = replies
diff --git a/resources/cmake_project/CMakeLists.txt b/resources/cmake_project/CMakeLists.txt
index 63da9f05..8890afda 100644
--- a/resources/cmake_project/CMakeLists.txt
+++ b/resources/cmake_project/CMakeLists.txt
@@ -25,7 +25,9 @@ else()
     endif()
 endif()
 
-project(libespidf)
+git_describe(_PROJECT_VER_GIT "${CMAKE_CURRENT_SOURCE_DIR}")
+set(PROJECT_VER "42.43.0-${_PROJECT_VER_GIT}")
+project(unicorn)
 
 # Copy the components.lock to the cargo project dir
 if(EXISTS ${dependencies_lock})
```

- Would you like to work on a fix? [n]

## To Reproduce

Steps to reproduce the behavior:

1. Create a project cargo generate esp-rs/esp-idf-template cargo
2. Build the project and check PROJECT_NAME and PROJECT_VER in the binary 

## Expected behavior

Configurable PROJECT_NAME and PROJECT_VER

## Screenshots

## Environment

- OS: Ubuntu 20.04
- How did you install the environment: espup

## Additional context

ivmarkov, 2024-06-09:
We do this using the [app_desc! macro](https://github.com/esp-rs/esp-idf-sys/blob/master/src/app_desc.rs#L2).

Please confirm if calling `app_desc!()` solves the problem for you, or else elaborate why you want these ESP IDF vars configured to something else in the first place.

elosev, 2024-06-09:
Oh, it works perfectly! Thanks!

closed 2024-06-09 by elosev