esp-rs/esp-idf-template #215: Modifying default log level does not work

rdeterre, 2024-06-17:
## Bug description

Setting the values of `CONFIG_LOG_DEFAULT_LEVEL` and `CONFIG_LOG_MAXIMUM_LEVEL` in sdkconfig.defaults appears to have no effect on the compiled project.

## To Reproduce

Steps to reproduce the behavior:

1. Generate a project with this template `cargo generate esp-rs/esp-idf-template cargo`
   ```shell
   🤷   Project Name: test2
   🔧   Destination: /Users/romain/Documents/test2 ...
   🔧   project-name: test2 ...
   🔧   Generating template ...
   ✔ 🤷   Which MCU to target? · esp32
   ✔ 🤷   Configure advanced template options? · false

   ```
2. Add the following lines to `sdkconfig.defaults` (based on [log config documentation here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-log-default-level)) and add a `log::debug!("Debug log!")` line in `main()` to test if debug logs are printed
   ```
   CONFIG_LOG_DEFAULT_LEVEL=4
   CONFIG_LOG_MAXIMUM_LEVEL=4
   ```
3. Compile the project: `cargo build`
4. Flash and run `cargo espflash flash --monitor`

## Expected behavior

Expect to see the "Debug log!" message appear in the log file

## Screenshots

## Environment

- OS: macOS
- How did you install the environment: espup 0.11.0

Vollbrecht, 2024-06-17:
your values that you put into sdkconfig are wrong. If you look at the [offical](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-log-default-level) docu, you should use the values like they are printed out in the available options sections.

```kconfig
CONFIG_LOG_DEFAULT_LEVEL_WARN=y
CONFIG_LOG_MAXIMUM_LEVEL_WARN=y
```

rdeterre, 2024-06-18:
Ah, that was it! Thanks and sorry for the noise!

closed 2024-06-18 by rdeterre